MichaIng/DietPi #4101: [Odroid HC1] [Question] OpenCL

RicardoCst, 2021-02-11:
Is there a way to install OpenCL on dietpi? clinfo doesnt show the drivers themselves.  only clinfo number of devices 0

ive tried installing malit628-odroid but this does nothing.

MichaIng, 2021-05-27:
Sorry for the late reply.

The `malit628-odroid` package indeed ships the `libOpenCL.so` required to make it work with that Mali GPU + development headers. I'm not sure how well Debian's DEB package replacement works. Please try to purge `ocl-icd-libopencl1` (which was installed as dependency for `clinfo`), the headers package and then reinstall `malit628-odroid`:
```
apt purge --autoremove ocl-icd-libopencl1 opencl-headers
apt install --reinstall malit628-odroid
```

RicardoCst, 2021-06-08:
clinfo:

clinfo: /usr/lib/arm-linux-gnueabihf/libOpenCL.so.1: version `OPENCL_1.0' not found (required by clinfo)

MichaIng, 2021-09-09:
@meveric
Could you have a look here? While the library and development headers are shipped with `malit628-odroid`, `clinfo` does not find the `OPENCL_1.0` definitions within or so? Not sure which versions the `libMali.so` includes, but based on the symlinks naming it should match.

meveric, 2021-09-12:
yes, that is due to the fact that back then when I made the packages for the old XU3/XU4 did not know how OpenCL worked, and thought I need to do the same as with GLESv1 and GLESv2 and link it all against libMali.so
But that's in fact not true for OpenCL.
OpenCL only needs a config file telling it what is the library to use for OpenCL (as it supports multiple platform and drivers).

I'll take a look at it, but it's gonna take a while, it's been a while since I did Mali drivers.

MichaIng, 2021-09-12:
Many thanks, good to know that it is hence expected. Do you know if this may work with Mesa (v20+): https://packages.debian.org/bullseye/mesa-opencl-icd

meveric, 2021-09-12:
mesa should not be important.. anyway in the meantime here some manual steps you can do:
1. delete wrongly created libOpenCL symlinks
```
rm /usr/lib/arm-linux-gnueabihf/libOpenCL.so*
ldconfig
```
2. download and extract regular libOpenCL drivers:
```
apt download ocl-icd-libopencl1
dpkg -x ocl-icd-libopencl1*.deb extract
```
3. copy over libOpenCL drivers:
```
sudo cp -av extract/usr/lib/arm-linux-gnueabihf/libOpenCL.so.1* /usr/lib/arm-linux-gnueabihf/
ldconfig
```
4. install MESA OpenCL (cause why not):
```
apt install mesa-opencl-icd
```
5. create config file for Mali GPU:
```
echo "libMali.so" > /etc/OpenCL/vendors/mali.icd
```
6. test config:
```
root@odroid-buster:~# clinfo
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
Number of platforms                               2
  Platform Name                                   ARM Platform
  Platform Vendor                                 ARM
  Platform Version                                OpenCL 1.2 v1.r17p0-01rel0.d17cf84c10daf186c21e3efcfae21a65
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_byte_addressable_store cl_khr_3d_image_writes cl_khr_fp64 cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_khr_fp16 cl_khr_icd cl_khr_egl_image cl_khr_image2d_from_buffer cl_arm_core_id cl_arm_printf cl_arm_thread_limit_hint cl_arm_non_uniform_work_group_size cl_arm_import_memory
  Platform Extensions function suffix             ARM

  Platform Name                                   Clover
  Platform Vendor                                 Mesa
  Platform Version                                OpenCL 1.1 Mesa 18.3.6
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_icd
  Platform Extensions function suffix             MESA

  Platform Name                                   ARM Platform
Number of devices                                 2
  Device Name                                     Mali-T628
  Device Vendor                                   ARM
  Device Vendor ID                                0x6200010
  Device Version                                  OpenCL 1.2 v1.r17p0-01rel0.d17cf84c10daf186c21e3efcfae21a65
  Driver Version                                  1.2
  Device OpenCL C Version                         OpenCL C 1.2 v1.r17p0-01rel0.d17cf84c10daf186c21e3efcfae21a65
  Device Type                                     GPU
  Device Profile                                  FULL_PROFILE
  Device Available                                Yes
  Compiler Available                              Yes
  Linker Available                                Yes
  Max compute units                               4
  Max clock frequency                             600MHz
  Device Partition                                (core)
    Max number of sub-devices                     0
    Supported partition types                     None
    Supported affinity domains                    (n/a)
  Max work item dimensions                        3
  Max work item sizes                             256x256x256
  Max work group size                             256
  Preferred work group size multiple              4
  Preferred / native vector sizes                 
[... 165 lines cut ...]
  clGetDeviceIDs(NULL, CL_DEVICE_TYPE_ALL, ...)   Success [ARM]
  clCreateContext(NULL, ...) [default]            Success [ARM]
  clCreateContext(NULL, ...) [other]              `
�Pp
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_DEFAULT)  Success (1)
    Platform Name                                 ARM Platform
    Device Name                                   Mali-T628
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CPU)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_GPU)  Success (2)
    Platform Name                                 ARM Platform
    Device Name                                   Mali-T628
    Device Name                                   Mali-T628
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ACCELERATOR)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CUSTOM)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ALL)  Success (2)
    Platform Name                                 ARM Platform
    Device Name                                   Mali-T628
    Device Name                                   Mali-T628

ICD loader properties
  ICD loader Name                                 OpenCL ICD Loader
  ICD loader Vendor                               OCL Icd free software
  ICD loader Version                              2.2.12
  ICD loader Profile                              OpenCL 2.2
```

MichaIng, 2021-09-12:
Great, many thanks for providing the workaround!

`mesa-opencl-icd` pulls in `ocl-icd-libopencl1` as dependency, so the manual download and extraction can be skipped. But as it contains `/usr/lib/arm-linux-gnueabihf/libOpenCL.so.1`, which `malit628-odroid` contains as well, doesn't it deny the install as of file list overlap?

So probably this would do it:
```sh
rm /usr/lib/arm-linux-gnueabihf/libOpenCL.so*
apt install mesa-opencl-icd
echo 'libMali.so' > /etc/OpenCL/vendors/mali.icd
```

MichaIng, 2021-09-12:
I repacked the package, so you can import it into the repo when you find time: https://dietpi.com/downloads/binaries/malit628-odroid_20210912-r17p0-big-1_armhf.deb
- `usr/lib/arm-linux-gnueabihf/libOpenCL.so*` symlinks removed
- `/etc/OpenCL/vendors/mali.icd` added
- "Replaces" and "Provides" aligned, so that `malit628-odroid` also provides `libgles1`, `libgles2` and `libegl1`
- Added `ocl-icd-libopencl1` as suggestion
- Updated checksum and installed size

Only left question is what do to about the OpenCL headers which are included as well. I think it is correct to keep those and the `ocl-icd-opencl-dev` provides/replaces accordingly as those headers need to be used to work with the Mali driver?

closed 2022-08-07 by MichaIng