MichaIng/DietPi #4108: Error: could not find a distribution template for Raspbian/buster

tcurdt, 2021-02-14:
### Creating a bug report/issue

#### Required Information
- DietPi version |
```
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=34
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version | `buster`
- Kernel version | `Linux DietPi 5.4.83-v7l+ #1379 SMP Mon Dec 14 13:11:54 GMT 2020 armv7l GNU/Linux`
- SBC model | `RPi 4 Model B (armv7l)`
- Power supply used | Official RPi 4
- SDcard used | SanDisk ultra

#### Additional Information (if applicable)
- Was the software title installed freshly or updated/migrated? **fresh install**
- Can this issue be replicated on a fresh installation of DietPi? **yes**

#### Steps to reproduce

Fresh download and install (32-bit) and then:

```
    $ apt-get install \
      apt-transport-https \
      ca-certificates \
      curl \
      gnupg-agent \
      software-properties-common
    $ curl -fsSL https://download.docker.com/linux/debian/gpg | apt-key add -
    $ add-apt-repository \
      "deb [arch=armhf] https://download.docker.com/linux/raspbian \
      $(lsb_release -cs) \
      stable"
```

#### Expected behaviour

There should be a file `/usr/share/python-apt/templates/Raspbian.info` looking like this

```
Suite: buster
RepositoryType: deb
BaseURI: http://raspbian.raspberrypi.org/raspbian/
MatchURI: http://raspbian.raspberrypi.org/raspbian/
```

#### Actual behaviour

`add-apt-repository` fails with the error `Error: could not find a distribution template for Raspbian/buster`

#### Extra details

see https://github.com/RPi-Distro/repo/issues/160

MichaIng, 2021-02-14:
Many thanks for your report.

I'm however not sure whether it's addressed correctly. Those templates are part of the [python-apt-common](https://packages.debian.org/buster/python-apt-common) package, so the maintainers of that package could be asked to add Raspbian. Note that Raspbian (archive.raspbian.org) and the Raspberry Pi repository (archive.raspberrypi.org) are two different things, so https://github.com/RPi-Distro/repo is the wrong place to hope for help with this. Either the package maintainers or the Raspbian maintainers would help with it: https://bugs.launchpad.net/raspbian

I'm wondering about whether it's worth it. You literally pass all relevant parameters to that `add-apt-repository` command that you can create a list file directly with:
```
echo "deb [arch=armhf] https://download.docker.com/linux/raspbian/ $(lsb_release -cs) stable" > /etc/apt/sources.list.d/docker.list
```
That command can only mess things up when adding it to the main `/etc/apt/sources.list` (don't do that!) or trying some magic that would be out of your control.

Docker can btw be installed via `dietpi-software`, which uses the Docker installer that creates the list file as above: https://github.com/docker/docker-install

tcurdt, 2021-02-14:
> I'm wondering about whether it's worth it. You literally pass all relevant parameters to that add-apt-repository command that you can create a list file directly with

That's actually a very fair point :)

> Docker can btw be installed via dietpi-software, which uses the Docker installer that creates the list file as above

Yeah, but I want to keep the install process as close to the normal Buster as possible.
So I am not really using the `dietpi-*` tools for that kind of things.

PS: Thx for dietpi!

MichaIng, 2021-02-14:
> Yeah, but I want to keep the install process as close to the normal Buster as possible.
> So I am not really using the `dietpi-*` tools for that kind of things.

I know what you mean: When you know what you're doing and how you want it, it's best to do it yourself 🙂.

However, closest to normal Buster would be `apt install docker.io`, though the version is naturally outdated 😉. Closest to how Docker devs intend it is probably: `sh -c "$(curl -sSfL https://get.docker.com)"`

But indeed, that installer does a lot of distribution detection that is not required when you know how to add an APT list to your Debian. And the installer e.g. does not allow Debian Bullseye until it's released and implemented into the installer, while likely their Buster packages would run fine on Bullseye. Looks like we should bypass that installer with `dietpi-software` as well and add the repo manually 😄. Then we can unlock Docker (+Portainer + Docker Compose) for Bullseye for testing purpose, anyway.

tcurdt, 2021-02-14:
> However, closest to normal Buster would be apt install docker.io, though the version is naturally outdated

That's not what the [docker instructions](https://docs.docker.com/engine/install/debian/) say.

> "Older versions of Docker were called docker, docker.io, or docker-engine. If these are installed, uninstall them:"

I guess docker provides a dedicated repo for a reason :)

It's just the official non-curl-script instructions just don't work so well because of said problem.
I guess that's also why they don't officially support it for Raspbian yet.

> For Raspbian, installing using the repository is not yet supported

But this turned into a non-issue for me. After manually adding the repo it works just fine.
Leave open or close? I am not sure. It still is a potential issue (for others).

Thanks for the help!

MichaIng, 2021-02-14:
> That's not what the docker instructions say.

That's why I differentiate between "close to" distro and close to software developers 😉. The `docker.io` is not an older or obsolete version, but actively updated and maintained by the Debian package maintainers, available as version 20.10.3 on Bullseye. It's just not the package that the Docker developers maintain and recommend, reasonably IMO (not in Debian's opinion obviously).

I'll do a PR to the Docker website to have `add-apt-repository` removed/replaced, especially since directly below
> Note: This procedure works for Debian on x86_64 / amd64, Debian ARM, or Raspbian.

is stated, which is and always was obviously wrong.

The info that Raspbian is not supported by the installer is outdated/wrong as well, since the repo has a separate Raspbian branch for a long time (https://download.docker.com/linux/raspbian/) and the installer detects and correctly handles and adds this repo for Raspbian systems: https://github.com/docker/docker-install/blob/master/install.sh#L316

Let me keep this issue open to track the Docker website updates 🙂.

MichaIng, 2021-02-14:
Lol I made: https://github.com/docker/docker.github.io/pull/12331
Before I found: https://github.com/docker/docker.github.io/pull/11990

However, I hope it brings some life in the topic at best get the older one merged first and mine afterwards which still clarifies about Raspbian support and contains a few other updates.

MichaIng, 2021-03-13:
The underlying issues has not been solved, which will likely never happen, but the steps provided by the Docker docs do not contain `add-apt-repository` anymore: https://github.com/docker/docker.github.io/pull/11990

The docs btw clearly state(d) that this manual steps are not supported on Raspbian and only the installer would work on Raspbian: https://docs.docker.com/engine/install/debian/#install-using-the-repository
> **Raspbian users cannot use this method!**

But actually Raspbian users can use is pretty fine, especially now that `add-apt-repository` is not contained anymore, only a single word needs to be changed in the repo URL. I address this with: https://github.com/docker/docker.github.io/pull/12331

I mark this issue hence as closed.

closed 2021-03-13 by MichaIng