MichaIng/DietPi #4109: RAMlog #2 - where is the syslog?

Gill-Bates, 2021-02-14:
A great feature of Dietpi is to write the logs to RAM to save the SD card.

For this purpose the menu `dietpi-software` offers three different options to choose from:

[image]

- `RAMlog #1` Seems to be clear: here nothing is written to the SD card at all.
- `Full` I also understand: here Dietpi behaves like an ordinary Linux distribution. All logs taking place inside `/var/log`

But what about `RAMlog #2`? Where can I find the `syslog` on the file system?

Joulinar, 2021-02-14:
Hi,

As stated on the screen short you posted, log files are saved to `/root/logfile_storage/*` before clean up

MichaIng, 2021-02-14:
`/root/logfile_storage/` contains the accumulated `/var/log/` file content, but not the "syslog", if you mean that literally. E.g. `/var/log/syslog` is created by `rsyslog` from what it gets from `systemd-journald`, hence requires full logging option.

However, `systemd-journald` runs and logs anyway (to RAM) and instead of having another daemon that writes those logs to a plain text file, you can access them directly via: `journalctl`
Use `journalctl -u dietpi-ramlog` to see logs for the dietpi-ramlog service, likewise for any other particular service/unit/target/mount, and you can sort and filter and format the output in many ways.

closed 2021-05-07 by MichaIng