MichaIng/DietPi #4111: net-tools

cocoflan, 2021-02-14:
https://packages.debian.org/stretch/net-tools

Package: net-tools (1.60+git20161116.90da8a0-1)
Links for net-tools
Screenshot
Debian Resources:

    Bug Reports
    Developer Information
    Debian Changelog
    Copyright File
    Debian Patch Tracker

Download Source Package net-tools:

    [net-tools_1.60+git20161116.90da8a0-1.dsc]
    [net-tools_1.60+git20161116.90da8a0.orig.tar.gz]
    [net-tools_1.60+git20161116.90da8a0-1.debian.tar.xz]

Maintainers:

    net-tools Team (QA Page, Mail Archive)
    Martín Ferrari (QA Page) 

External Resources:

    Homepage [sourceforge.net]

Similar packages:

    iproute2
    inetutils-tools
    nftables
    apf-firewall
    can-utils
    linux-wlan-ng
    lxc-tests
    netsniff-ng
    python3-nftables
    libnftables0

NET-3 networking toolkit

This package includes the important tools for controlling the network subsystem of the Linux kernel. This includes arp, ifconfig, netstat, rarp, nameif and route. Additionally, this package contains utilities relating to particular network hardware types (plipconfig, slattach, mii-tool) and advanced aspects of IP configuration (iptunnel, ipmaddr).

In the upstream package 'hostname' and friends are included. Those are not installed by this package, since there is a special "hostname*.deb".

MichaIng, 2021-02-15:
Many thanks for your suggestion.

Note that `net-tools` is deprecated and superseded by [iproute2](https://packages.debian.org/buster/iproute2), which is the reason why we intentionally removed it from pre-installed packages and replaced all calls with appropriate `ip` command calls.

However, if you are used to use the old net-tools commands, you can simply install them from the Debian repo via `apt install net-tools`, which guarantees that it does not conflict with other packages commands (like hostname command of the hostname package you mentioned). But especially for Linux newcomers I strongly recommend to not start getting used to it anymore.

cocoflan, 2021-02-15:
Ok good to know.

MichaIng, 2021-02-21:
I mark this issue as closed. Feel free to reopen if required.

closed 2021-02-21 by MichaIng