MichaIng/DietPi #4115: Wifi Hotspot 5GHZ

retro24, 2021-02-16:
Hello i installed Hotspot on my Raspberry pi 4 using the latest dietpi but the wifi works in 2,4Ghz and i wanted 5Ghz, i dont see any options in the network adapters to change it to 5Ghz,how to i change it?

thanks

Joulinar, 2021-02-16:
Hi,

many thanks for your message. I found a similar topic on the RPi Github. https://github.com/raspberrypi/linux/issues/2619#issuecomment-410703338

Solutions seems to be setting a different country code like US

retro24, 2021-02-17:
Can you in a future release to add a setting to select 2,4ghz or 5ghz ??

Thanks

MichaIng, 2021-02-17:
Jep, for devices with onboard WiFi where we know that they support 5 GHz WiFi, or if there is a different way to check it, we can add `ieee80211ac=1` to `wpa_supplicant.conf`. Or we simply add a switch to enable/disable with the hint that it may or may not work depending on the WiFi adapter, so admins need to verify themselves?

retro24, 2021-02-19:
it will be great if you can add this

thanks

nicoolaro, 2022-03-25:
@MichaIng  is this still in progress? I search everywhere and couldn't find additional info. I did change CC to US but still no luck.
Should i manually add `ieee80211ac=1 to wpa_supplicant.conf` ?

MichaIng, 2022-03-25:
If `ieee80211ac=1` and using correct WiFi country code (reboot afterwards) does not work, then I'm currently not sure either how to enable it.

> still not seeing 5g option

You mean using WiFi clients you see that there is no 5G SSID provided by the hotspot?

nicoolaro, 2022-03-25:
> If `ieee80211ac=1` and using correct WiFi country code (reboot afterwards) does not work, then I'm currently not sure either how to enable it.
> 
> > still not seeing 5g option
> 
> You mean using WiFi clients you see that there is no 5G SSID provided by the hotspot?

Correct no 5G WiFi Hotspot also no options within Diepti Config.

I understand if this is still not fixed. 
Here are logs if you need them anyway. And as soon as i type this command WiFi 2.4g disables 

```
root@DietPi:~# hostapd -dd /etc/hostapd/hostapd.conf
random: getrandom() support available
Configuration file: /etc/hostapd/hostapd.conf
nl80211: Using driver-based roaming
nl80211: TDLS supported
nl80211: Supported cipher 00-0f-ac:1
nl80211: Supported cipher 00-0f-ac:5
nl80211: Supported cipher 00-0f-ac:2
nl80211: Supported cipher 00-0f-ac:4
nl80211: Supported cipher 00-0f-ac:6
nl80211: Using driver-based off-channel TX
nl80211: Supported vendor command: vendor_id=0x1018 subcmd=1
nl80211: Use separate P2P group interface (driver advertised support)
nl80211: Enable multi-channel concurrent (driver advertised support)
nl80211: use P2P_DEVICE support
nl80211: interface wlan0 in phy phy0
nl80211: Set mode ifindex 3 iftype 3 (AP)
nl80211: Setup AP(wlan0) - device_ap_sme=1 use_monitor=0
nl80211: Subscribe to mgmt frames with AP handle 0x55a7dd1ca0 (device SME)
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a7dd1ca0 match=04
nl80211: kernel reports: Match already configured
nl80211: Register frame command failed (type=208): ret=-114 (Operation already in progress)
nl80211: Register frame match - hexdump(len=1): 04
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a7dd1ca0 match=0501
nl80211: kernel reports: Match already configured
nl80211: Register frame command failed (type=208): ret=-114 (Operation already in progress)
nl80211: Register frame match - hexdump(len=2): 05 01
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a7dd1ca0 match=0503
nl80211: kernel reports: Match already configured
nl80211: Register frame command failed (type=208): ret=-114 (Operation already in progress)
nl80211: Register frame match - hexdump(len=2): 05 03
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a7dd1ca0 match=0504
nl80211: kernel reports: Match already configured
nl80211: Register frame command failed (type=208): ret=-114 (Operation already in progress)
nl80211: Register frame match - hexdump(len=2): 05 04
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a7dd1ca0 match=06
nl80211: kernel reports: Match already configured
nl80211: Register frame command failed (type=208): ret=-114 (Operation already in progress)
nl80211: Register frame match - hexdump(len=1): 06
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a7dd1ca0 match=08
nl80211: kernel reports: Match already configured
nl80211: Register frame command failed (type=208): ret=-114 (Operation already in progress)
nl80211: Register frame match - hexdump(len=1): 08
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a7dd1ca0 match=09
nl80211: kernel reports: Match already configured
nl80211: Register frame command failed (type=208): ret=-114 (Operation already in progress)
nl80211: Register frame match - hexdump(len=1): 09
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a7dd1ca0 match=0a
nl80211: kernel reports: Match already configured
nl80211: Register frame command failed (type=208): ret=-114 (Operation already in progress)
nl80211: Register frame match - hexdump(len=1): 0a
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a7dd1ca0 match=11
nl80211: kernel reports: Match already configured
nl80211: Register frame command failed (type=208): ret=-114 (Operation already in progress)
nl80211: Register frame match - hexdump(len=1): 11
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a7dd1ca0 match=12
nl80211: kernel reports: Match already configured
nl80211: Register frame command failed (type=208): ret=-114 (Operation already in progress)
[... 80 lines cut ...]
wlan0: AP-DISABLED
wlan0: Unable to setup interface.
hostapd_interface_deinit_free(0x55a7dce9a0)
hostapd_interface_deinit_free: num_bss=1 conf->num_bss=1
hostapd_interface_deinit(0x55a7dce9a0)
wlan0: interface state DISABLED->DISABLED
hostapd_bss_deinit: deinit bss wlan0
wlan0: Deauthenticate all stations
nl80211: sta_remove -> DEL_STATION wlan0 ff:ff:ff:ff:ff:ff --> 0 (Success)
wlan0: AP-DISABLED
hostapd_cleanup(hapd=0x55a7dcfff0 (wlan0))
wlan0: CTRL-EVENT-TERMINATING
hostapd_free_hapd_data: Interface wlan0 wasn't started
hostapd_interface_deinit_free: driver=0x558fdcd670 drv_priv=0x55a7dd1460 -> hapd_deinit
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211: Remove monitor interface: refcount=0
nl80211: Remove beacon (ifindex=3)
netlink: Operstate: ifindex=3 linkmode=0 (kernel-control), operstate=6 (IF_OPER_UP)
hostapd_interface_free(0x55a7dce9a0)
hostapd_interface_free: free hapd 0x55a7dcfff0
hostapd_cleanup_iface(0x55a7dce9a0)
hostapd_cleanup_iface_partial(0x55a7dce9a0)
hostapd_cleanup_iface: free iface=0x55a7dce9a0
root@DietPi:~#
```

MichaIng, 2022-03-25:
> also no options within Diepti Config.

Of course, otherwise the issue would have a `solution available` tag and be closed.

> I understand if this is still not fixed.

This is not a fix bug an additional feature, and as of your results we still don't know how to implement it and whether it is supported by current `hostapd` at all.

> Here are logs if you need them anyway. And as soon as i type this command WiFi 2.4g disables

There is a service running already, so this is supposed to fail. After `ieee80211ac=1`, to apply the change and see the logs:
```sh
systemctl restart hostapd
sleep 3
journalctl -u hostapd
```

symbios24, 2023-07-02:
Hi same problem here raspberry pi 4 has 5ghz wifi but uses the 2.4ghz only in hotspot mode and there is no option to change it from the network adapters

MichaIng, 2023-08-20:
This has been added for next DietPi release: #6558
Beta starts within the next hours.

closed 2023-08-26 by MichaIng