MichaIng/DietPi #4116: Medusa broken after update

mrgreaper, 2021-02-16:
Did a sudo apt upgrade today as advised on the main screen of dietpi but Medusa wont load.

if I do "journalctl -u medusa" I see a lot of this error...no idea where region.py is (its not in /mnt/dietpi_userdata/medusa/medusa/ where I would of expected to see it)

`ERROR:stevedore.extension:Could not load 't
vdb': invalid syntax (region.py, line 404)`

"sudo systemctl status medusa.service" shows :
 ` medusa.service - Medusa Daemon
   Loaded: loaded (/etc/systemd/system/medusa.service; disabled; vendor preset: 
enabled)
   Active: active (running) since Tue 2021-02-16 19:39:29 GMT; 2s ago
 Main PID: 1905 (python3)
    Tasks: 1 (limit: 4915)
   CGroup: /system.slice/medusa.service
           └─1905 /usr/bin/python3 /mnt/dietpi_userdata/medusa/start.py -q --nol
aunch --datadir=/mnt/dietpi_userdata/medusa

Feb 16 19:39:29 DietPi systemd[1]: Started Medusa Daemon.
`

but this is a lie..... i can not access the webpage for it at all.

I have tried uninstalling and reinstalling it but to no avail.

Google searches have shown some tips ...hence the cat and status above but noting seems to have worked...completly out of my depth now

MichaIng, 2021-02-16:
Many thanks for your report.

I just did a test install on Raspbian Buster, where it works fine.

Which hardware/SBC, DietPi version and Debian version are you using?

And please paste the output of `cat /mnt/dietpi_userdata/medusa/Logs/application.log` as well.

MichaIng, 2021-02-16:
In my case:
```
# mawk 'NR==404' /mnt/dietpi_userdata/medusa/ext/dogpile/cache/region.py
        self.region_invalidator: RegionInvalidationStrategy = (
```
So either my version is different or your file has somehow been corrupted. In my case I cannot fine "`t vdb`" in that file.

Here is the file on GitHub to compare: https://github.com/pymedusa/Medusa/blob/master/ext/dogpile/cache/region.py

mrgreaper, 2021-02-16:
cat /mnt/dietpi_userdata/medusa/Logs/application.log resulted in:
`dietpi@DietPi:~$ cat /mnt/dietpi_userdata/medusa/Logs/application.log
cat: /mnt/dietpi_userdata/medusa/Logs/application.log: No such file or directory
`

 DietPi v6.34.3 : 20:57 - Tue 16/02/21
 Device model : PINE A64 (aarch64)

According to the main screen when you ssh in, no idea what version of medusa is installed...but since I uninstalled it via dietpi_software and then rebooted, then reinstalled (and rebooted again) as part of my trouble shooting I would say it must be on the latest version?

now trying the fix mentioned

MichaIng, 2021-02-16:
> ...but since I uninstalled it via dietpi_software and then rebooted, then reinstalled (and rebooted again) as part of my trouble shooting I would say it must be on the latest version?

Okay yes then it's the latest version.

Debian is Buster? `echo $G_DISTRO_NAME`

mrgreaper, 2021-02-16:
`dietpi@DietPi:/mnt/dietpi_userdata/medusa$ ls
CHANGELOG.md	 hooks		requirements.txt       tests
CONTRIBUTING.md  lib		runscripts	       themes
COPYING.txt	 MANIFEST.in	setup.cfg	       themes-default
Dockerfile	 medusa		setup.py	       tox.ini
dredd		 package.json	SickBeard.py	       yarn.lock
ext		 readme.md	start.py
ext3		 renovate.json	test_requirements.txt
`
^ above is the ls for the medusa folder....lack of log folder bad?

echo $G_DISTRO_NAME results in:
`dietpi@DietPi:/mnt/dietpi_userdata/medusa$ echo $G_DISTRO_NAME
stretch
`

`ietpi@DietPi:/mnt/dietpi_userdata/medusa$ journalctl -u medusa
-- Logs begin at Thu 2016-11-03 17:16:42 GMT, end at Tue 2021-02-16 21:06:55 GMT
. --
Feb 16 21:02:28 DietPi systemd[1]: Started Medusa Daemon.
Feb 16 21:02:39 DietPi python3[689]: ERROR:stevedore.extension:Could not load 'a
ddic7ed': invalid syntax (region.py, line 404)
Feb 16 21:02:39 DietPi python3[689]: ERROR:stevedore.extension:Could not load 'a
rgenteam': invalid syntax (region.py, line 404)
Feb 16 21:02:39 DietPi python3[689]: ERROR:stevedore.extension:Could not load 'l
egendastv': invalid syntax (region.py, line 404)
Feb 16 21:02:39 DietPi python3[689]: ERROR:stevedore.extension:Could not load 't
vsubtitles': invalid syntax (region.py, line 404)
Feb 16 21:02:39 DietPi python3[689]: ERROR:stevedore.extension:Could not load 'o
mdb': invalid syntax (region.py, line 404)
Feb 16 21:02:39 DietPi python3[689]: ERROR:stevedore.extension:Could not load 't
vdb': invalid syntax (region.py, line 404)
Feb 16 21:02:39 DietPi python3[689]: Traceback (most recent call last):
Feb 16 21:02:39 DietPi python3[689]:   File "/mnt/dietpi_userdata/medusa/start.p
y", line 6, in <module>
Feb 16 21:02:39 DietPi python3[689]:     from medusa.__main__ import main
Feb 16 21:02:39 DietPi python3[689]:   File "/mnt/dietpi_userdata/medusa/medusa/
__init__.py", line 24, in <module>
Feb 16 21:02:39 DietPi python3[689]:     initialize()
Feb 16 21:02:39 DietPi python3[689]:   File "/mnt/dietpi_userdata/medusa/medusa/
--More--
`

mrgreaper, 2021-02-16:
I cant help thinking t vdb should be tvdb though I have not edited any of these files.
I tried looking at the region.py file in mnt/dietpi_userdata/medusa/ext/dogpile/cache/ and went to line 404:

`self.region_invalidator: RegionInvalidationStrategy = (
            DefaultInvalidationStrategy()
`

I do not know Nano well enough to search for t vdb....looking into that now.Thing is this has been working for months with no issue...ok medusa said it was behind 300 odd commits, but it worked... today i decided to tell diet pi to update everything and ..well... blam :(

MichaIng, 2021-02-16:
Ah okay, Stretch will be the reason. The newest Medusa version requires Python 3.6 or higher: https://github.com/pymedusa/Medusa#dependencies
But Debian Stretch ships Python 3.5.3 only 😢: https://packages.debian.org/stretch/python3
I'm going to make a test install on Stretch as well, to verify that it causes those syntax errors.

So we need to remove the Medusa install option for Debian Stretch, sadly. Not the first software that started to require Python 3.6 or higher.

If you find time, I recommend the flash our latest image based on Debian Buster, which basically means two years newer APT packages 😉. But a quicker fix is to use an older Medusa version. Let's see which one is still Python 3.5 compatible: https://github.com/pymedusa/Medusa/releases

mrgreaper, 2021-02-16:
I am at work at the minute is there a way to upgrade from stretch to buster via ssh? (and more importantly keep my current settings, i remember vaguely setting everything up was a real pain)

if not, how do I tell the installer to install an older version?

mrgreaper, 2021-02-16:
would this work? https://github.com/chriskuehl/python3.6-debian-stretch#:~:text=%20Python%203.6%20Debian%20stretch%20backport%20%201,(3.5.3)?%20These%20packages%20provide%20a%20binary...%20More

Joulinar, 2021-02-16:
even if it is not officially supported, you could have a look to following link if you consider to upgrade to buster https://dietpi.com/phpbb/viewtopic.php?p=18988#p18988

MichaIng, 2021-02-16:
v0.5.0 dropped Python 2 support and at least with that release in the readme file, the min Python 3 version has been incremented as well. So it the last version for Stretch would be: https://github.com/pymedusa/Medusa/archive/v.0.4.6.tar.gz

Downgrading Medusa would hence be (since you already uninstalled it, you don't have important Medusa settings that need to be preserved, right?):
```
G_EXEC rm -Rf /mnt/dietpi_userdata/medusa
G_EXEC cd /tmp
G_EXEC curl -sSfLO https://github.com/pymedusa/Medusa/archive/v.0.4.6.tar.gz
G_EXEC tar xf v.0.4.6.tar.gz
G_EXEC rm v.0.4.6.tar.gz
G_EXEC mv Medusa-v.0.4.6 /mnt/dietpi_userdata/medusa
G_EXEC systemctl restart medusa
```
______________________
Upgrading Python 3 on Stretch is possible of course as well. However, those packages you found are three years old and possibly miss important security patches. I would then suggest instead to compile Python 3 manually: https://tecadmin.net/how-to-install-python-3-9-on-debian-9/
______________________
When instead upgrading Debian, note the following:
> # Also check other /etc/apt/sources.list.d/*.list files that might need to be updated

I.e. in your case of PINE64, there is `/etc/apt/sources.list.d/armbian.list` (as long as it's not a very old image) where this replacement needs to be done as well. Probably easiest:
```
sed -i 's/stretch/buster/' /etc/apt/sources.list{,.d/*.list}
```
If `apt update` throws an error then, probably some software-specific repo has no Buster branch, but that's very unlikely.
______________________
When flashing a new image, yes migrating data and config, respectively setting up the system again might take time. I personally write shell scripts for all files I create or alter on a new image, or other customisations I do. I started to put some of them to GitHub, so I can easily download and execute them on any more system 🙂: https://github.com/MichaIng/hacks

MichaIng, 2021-02-16:
Ah lol, we found that issue and disabled Medusa on Debian Stretch already for the upcoming DietPi release 😄: #3991
But this does not effect Medusa instance that were installed already, so either downgrading Medusa or compiling a newer Python 3 will still work fine in your case.

mrgreaper, 2021-02-16:
currently going the upgrade route, though if you have clarity on the definition of "risky" that may help lol
risky as in it may all crash and not work....i can live with
risky as in the device may catch fire.... thats a no go.. (I am assuming the former but would be amiss if i did not check it wasnt the later)

Have just been going through the sources when i noticed apt update showed some set to stretch (jellyfin and armbian)

Had an issue with emby which i no longer use wanting to be uninstalled but seemed to have flushed that. 

Good that its a quiet night at work lol

I did uninstall and re-install but i am rather hopeful that was the program not the settings i removed, the two should be seperate in any good application....guess i will find out soon *gulp*

MichaIng, 2021-02-16:
I cannot guarantee for anything 😄, but Debian distribution upgrades have become indeed smoother than it was a few versions ago. I remember Wheezy => Jessie that time and it broke quite much, took maybe longer to get that running as before than starting with a fresh Jessie image. But Jessie => Stretch was much better already.

There are some dietpi-software options that might need to be reinstalled. What else do you have installed aside of Medusa?
```
grep ' =2 ' /boot/dietpi/.installed
```

mrgreaper, 2021-02-16:
according t grep ' =2 ' /boot/dietpi/.installed ....nothing
but then apt upgrade is currently running

MichaIng, 2021-02-16:
Whoops, wrong command:
```
dietpi-software list | grep ' =2 '
```
or
```
grep '=2' /boot/dietpi/.installed
```
but the latter shows the IDs only, not the names.

mrgreaper, 2021-02-16:
id 41 | =2 | emby server: web interface media streaming server | | https://dietpi.com/phpbb/viewtopic.php?p=1789#p1789
id 50 | =2 | syncthing: backup and sync server with web interface | | https://dietpi.com/phpbb/viewtopic.php?p=2363#p2363
id 87 | =2 | sqlite: database | | https://dietpi.com/phpbb/viewtopic.php?p=1335#p1335
id 103 | =2 | dietpi-ramlog: minimal, optimised logging | | https://dietpi.com/phpbb/viewtopic.php?p=68#p68
id 104 | =2 | dropbear: lightweight ssh server | | https://dietpi.com/phpbb/viewtopic.php?p=62#p62
id 116 | =2 | medusa: automatic video library manager for tv shows | | https://dietpi.com/phpbb/viewtopic.php?p=3327#p3327
id 126 | =2 | libssl1.0.0: backwards compatibility (stretch, buster, bullseye) |
id 145 | =2 | radarr: automatically download movies | +sqlite | https://dietpi.com/phpbb/viewtopic.php?p=7457#p7457
id 149 | =2 | nzbget: nzb download manager | | https://dietpi.com/phpbb/viewtopic.php?p=7575#p7575
id 150 | =2 | mono: runtime libraries and repo |
id 170 | =2 | unrar: unarchiver for .rar files |
id 178 | =2 | jellyfin: foss web interface media streaming server | | https://dietpi.com/phpbb/viewtopic.php?p=26255#p26255

The emby one is a tad strange as that is uninstalled O.o

stil going through the update...to be clear...risky does not mean chance of device catching fire?

Joulinar, 2021-02-16:
no physical damage is expected, but logical issues on the software side could happen. Always a chance for Murphy being around the corner 😉 https://en.wikipedia.org/wiki/Murphy%27s_law

mrgreaper, 2021-02-16:
damn that murphy!
well its all installed and updated.
medusa is working although back to its default port and grinning at me completly devoid of all my hours of setting up.....mumble grumble mumble.

On the plus side I think i got away without needing to buy a sd card reader and waiting till I am home to redo the sd card (good job too as i remember it being a right pain last time)

Joulinar, 2021-02-16:
ok lucky you 😃 

I guess we can close this one as you are back online with Medusa, right?

MichaIng, 2021-02-16:
Due to the large version jumps it is not guaranteed that the new package version work with the old configs, e.g. So when some service fails to start, it might be due to invalid config entry. Worse would be incompatibility with existing database of such. And of course compiled software might depend on specific library versions from Stretch and deny to start on Buster, where libraries have been upgraded. But yes worst that can happen is a longer debug session, checking service logs etc. 😉.

Ah the long upgrade process might also fails in the middle somewhere, e.g. due to full disk or memory or as many package installs imply a service start, which might fail as above. So then service starts need to be fixed before the upgrade can be finished. Can be all nasty but the hardware won't care.

mrgreaper, 2021-02-16:
Yep this can now be clossed.

Thank you guys for the help, greatly appreciated. Was more than a little out of my element there lol.
side note, medusa runs so snappy when it doesnt have anything in its database...almost seems a shame to have it setup lol

closed 2021-02-16 by mrgreaper

MichaIng, 2021-02-16:
Your software list as well requires no reinstall, all fine. Jellyfin might have gotten a package upgrade when you changed the repository list file with the sed command mentioned above.

If you don't use Emby: `dietpi-software uninstall 41`
... ah lol it has no uninstall code. How did this happen, a good find 😄!

mrgreaper, 2021-02-16:
Reply I also just learnt that if medusa is going through your folders adding tv shows automagically..... dietpi-software uninstall 41 is a bad idea as it closes medusa, shouldnt cause any issues though *crosses fingers* I should of thought that one through tbh, i know when you install or remove something it closes all processes

Joulinar, 2021-02-16:
yep, usually `dietpi-software` is going to stop all service on install and uninstall.

MichaIng, 2021-02-17:
> dietpi-software uninstall 41 is a bad idea as it closes medusa

Hmm, actually on uninstalls this is not required. On installs it makes sense, as they may contain RAM-intense or CPU-intense operations, but uninstalls should not require a service stop. The ones that are uninstalled are stopped as first part of the uninstall step anyway 🤔.

MichaIng, 2021-02-18:
https://github.com/MichaIng/DietPi/commit/735b0d25804c320223214e7740f33b1318f8a78d
https://github.com/MichaIng/DietPi/commit/2e39c831df1b062b1362f585f42be3c5a6d444b8
🙂