MichaIng/DietPi #4117: dietpi-backup support for rclone

mikekuzak, 2021-02-17:
Hi,

I'm using rclone with gdrive and mounted a folder so I can easily copy stuff over to the cloud. The it's mounted to /mnt/cloud-backups when trying to change the location on the dietpi-config for the backups it's complaints it's not ext/2/3/4 compatible. 

I don't see why this shouldn't work. 

Thanks

MichaIng, 2021-02-17:
Many thanks for your suggestion!

> it's complaints it's not ext/2/3/4 compatible.

DietPi-Backup supports only a known list of file systems: https://github.com/MichaIng/DietPi/blob/dev/dietpi/dietpi-backup#L57
Good hint about the shown text, which is outdated, as Btrfs and F2FS are supported as well, and I'll add XFS now. This is to assure that the target file system has full UNIX permissions and symbolic links supported, otherwise when restoring the backup, you would restore a broken system. NFS is supported as well even that it is not really guaranteed that the file system server-side really supports modes and symlinks.

gdrive would be that same case as with NFS, I guess, so that it is possible to run e.g. `ln -s file link` and `chmod 600 file` on the mount/client side and have the fully functional symbolic link and changed mode on any other gdrive mount/client? But I personally am not keen to start now supporting all kind of proprietary mount types to be true, as then Dropbox and all the others will follow, while we actually provide easy options to run a self-hosted file server with DietPi 😉. For that at least we'd need to migrate to a generic test for remote mounts, i.e. creating an actual symlink and and changing a files mode and see if this succeeds. Would be no guarantee, but with a warning we could probably provide it.

What I am wondering, when you mount that gdrive on a let's say Windows system, how are symbolic links and UNIX permissions shown, and are they even preserved then or probably destroyed when the Windows client syncs files back with the limited capabilities of its NTFS?
_____________________
Wrapping all in a tarball and then logically compressing it on the other hand is a real option where modes and symlinks are not required anymore. We decided against compression in the past due to the overhead and as during compression and decompression it temporarily requires doubled space, breaks the ability to easily view and restore single files and all such. For a relatively small system backup that has no real benefit. But for a remote backup on the other hand it makes sense, not only to be independent of file system capabilities but as well to lower network traffic. Probably we can make it a two step backup: First, create a regular local rsync backup, then create a compressed tarball from it to move to remote file system. I'll think through it. **Open for suggestions, of course!**

> `sudo dd if=/dev/sda conv=sync,noerror bs=64K | gzip -c  > /PATH/TO/DRIVE/backup_image.img.gz`

This is a bad idea for several reasons:
- Doing raw dd on a mounted file system can lead to corrupted files and/or file system when a file is currently written or shifted through the file system while dd is running through raw bits and might then copy the same file or fragments of it on two times or not at all. For this, the file system must be at least read-only mounted, which is not as trivially possible from a running OS as one might think. Safest would be to add `ro` mount option to fstab and reboot.
- You copy really everything, including all empty space, respectively the non-linked inode bits that are still stored there while their related files have been removed already. So the image file will be as large as the whole drive. Of course, when free space is e.g. `zerofree`'d, it can be compressed effectively, but this again is only possible on ext4.
- You copy only the root drives content, no other mount. When users of e.g. RPi or Odroids have the boot partition on an SD card while moving the root file system to a USB drive, the boot partition would be missing, leading to a most likely unbootable system on restore as kernel modules are restored that do not match the kernel version of the boot partition.
- And most importantly, such a raw drive backup cannot be restored from the running system but only from an external system, as you're rewriting all the data that is currently loaded and used 😉.

Such drive clones are great, so one is able to restore the system when the drive/SD card broke, but they should be at best created either from an external system, or after rebooting the system into R/O mode, and it's really a fundamentally different kind of backup than what `dietpi-backup` aims to do 😉.

mikekuzak, 2021-02-17:
@MichaIng 
Hi there, 

First of all I would like to thank you for the great work on this project. I'm really impressed and it's a fantastic distro. Also thank you for the long post and explanation, I really appreciate this. It's not every day one gets that much luv on a post  :) 

Of course I agree on the dd and I know it shouldn't be used on a live system. So let's just forget about this one. 
Back to the original question. So the mount via rclone is done this way:
```
gdrive: on /mnt/cloud-backup type fuse.rclone (rw,nosuid,nodev,relatime,user_id=1000,group_id=1000)
```
but you said it already: "But I personally am not keen to start now supporting all kind of proprietary mount types to be true"
which is a fair statement, and I can work around this. 

The main problem is the size and type of the backup. I have a 8GB sdcard, 5GB are taken, so doing a local backup, and then trying to do a tarball won't work because of the size constrain. As you can see there are 2 problem, can't directly write to the cloud drive, can't  do a local backup, tarball, move to cloud due to disk space constrains on the local device.  

That's why I was suggesting a second option on the dietpi-backup config where it does a tarball directly if you choose this option. This way I would have a compressed file, and could move it of the device via a cron job, hopefully.

MichaIng, 2021-02-18:
> fuse.rclone

Hmm, we allow `fuse*` file systems (which includes sshfs) when first checking `ln -s` and `chmod` on a test file, then the regular way without compression.

A nice compilation of several remote backup/transfer methods: https://www.commandlinefu.com/commands/matching/rsync/cnN5bmM=/sort-by-votes
But sadly rsync cannot pipe file into a tarball directly. That would be best for us as dietpi-backup file selection/filter etc is currently all based on rsync.

mikekuzak, 2021-02-18:
When I try to point the backup directly to it this is the error I get: 
```
 Error:                                                
│                                                                                                                     │
│ /mnt/cloud-backup Has a filesystem of: , and is not supported.               │
│                                                                                                                     │
│ The filesystem must be EXT2/3/4 for symlink compatibility.            
```
```
su dietpi
drwxr-xr-x 1 dietpi dietpi    0 Feb 17 12:52 cloud-backup
drwxrwxr-x 7 dietpi dietpi 4096 Feb 14 00:24 dietpi_userdata
drwxrwxr-x 2 dietpi dietpi 4096 Jan 18 16:26 ftp_client
drwxrwxr-x 2 dietpi dietpi 4096 Jan 18 16:28 nfs_client
drwxrwxr-x 2 dietpi dietpi 4096 Jan 18 16:28 samba
```
It's not accessible from root user: 
Maybe that's the problem 
```
d????????? ? ?      ?         ?            ? cloud-backup
drwxrwxr-x 7 dietpi dietpi 4096 Feb 14 00:24 dietpi_userdata
drwxrwxr-x 2 dietpi dietpi 4096 Jan 18 16:26 ftp_client
drwxrwxr-x 2 dietpi dietpi 4096 Jan 18 16:28 nfs_client
drwxrwxr-x 2 dietpi dietpi 4096 Jan 18 16:28 samba
```

mikekuzak, 2021-02-18:
Reconfigured .. no banana .. 

```
/mnt/cloud-backup Has a filesystem of: fuse.rclone, and is not supported.                                            │
                                                          │                                                                                                                      │
                                                          │ The filesystem must be EXT2/3/4 for symlink compatibility.            
```

```
drwxr-xr-x 1 root   root      0 Feb 18 22:25 cloud-backup
drwxrwxr-x 7 dietpi dietpi 4096 Feb 14 00:24 dietpi_userdata
drwxrwxr-x 2 dietpi dietpi 4096 Jan 18 16:26 ftp_client
drwxrwxr-x 2 dietpi dietpi 4096 Jan 18 16:28 nfs_client
drwxrwxr-x 2 dietpi dietpi 4096 Jan 18 16:28 samba
```

MichaIng, 2021-02-18:
Ah, so Google Drive itself has no symlink support 😢.

mikekuzak, 2021-02-18:
It should work, but you would need to use rclone copy/sync -L instread of rsync. But I haven't tested it.

MichaIng, 2021-02-19:
> -L, --copy-links Follow symlinks and copy the pointed to item.

That would be wrong. It would copy the target of the symlink to the location of the symlink, which is not what we want and would break the system.

> -l, --links Translate symlinks to/from regular files with a '.rclonelink' extension

That would then need to be reversed as well on restore. But I think POSIX permissions are not preserved as well. E.g. `/mnt/dietpi_userdata` and it's content in the backup is most likely owned by `root:root`, right?

cocoflan, 2021-02-26:
Also version control on backup would be nice! like you have on déja dup!