MichaIng/DietPi #4118: [Feature request] Add to wiki a guide how to secure SSH access with public key authentication

ovz93br43v7, 2021-02-17:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
Not directly related to a problem. But I think your software guides the user very easy through installation of various (online) services, therefor I think the device itself should be secured with best practice. I'm not a security expert but during setup of my running server I used some online guides and most of them propose to secure the SSH access with public key authentication.

See here: https://www.thomas-krenn.com/de/wiki/OpenSSH_Public_Key_Authentifizierung_unter_Ubuntu
Or see here in the section "[SSH-KEY-AUTHENTISIERUNG](https://forum.kuketz-blog.de/viewtopic.php?f=42&t=3067&sid=31bb49a0a3ab33881d7aea0e72f95164)"

#### Describe the solution you'd like:

I would propose to add at least in the wiki a short guide how the SSH access should be secured. Or if it make sense guide the user during first setup of the dietpi how to secure the SSH access properly.

#### Describe alternatives you've considered:

Every user could do / decide it on their own. Because your project will be used from users which are to lazy or don't know how to setup the offered services on their own, I assume that most user just don't know or don't care about the risks of an unsecured SSH access.

ravenclaw900, 2021-02-17:
This looks more like an issue for [MichaIng/DietPi-Docs](https://github.com/MichaIng/DietPi-Docs/).  See https://github.com/MichaIng/DietPi-Docs/issues/244.

Joulinar, 2021-02-17:
Indeed DietPi docs are available on GitHub as well. Everyone is welcome to support and extend. Feel free to raise a PR and add such a section 😉

ovz93br43v7, 2021-02-17:
> Hello,
> This looks more like an issue for [MichaIng/DietPi-Docs](https://github.com/MichaIng/DietPi-Docs/), as the wiki on here is mainly development information. See [MichaIng/DietPi-Docs#244](https://github.com/MichaIng/DietPi-Docs/issues/244).

Oh sorry I just browsed through https://dietpi.com/docs/ and didn't found this above mentioned explanation.

ovz93br43v7, 2021-02-17:
> Indeed DietPi docs are available on GitHub as well. Everyone is welcome to support and extend. Feel free to raise a PR and add such a section wink

Sure you are completely right! In my case it just would be a translated copy of the mentioned guides I followed in the past. But due to the fact that this info is already existing in https://github.com/MichaIng/DietPi/wiki/Security-recommendation please forget my request.

closed 2021-02-17 by ovz93br43v7

MichaIng, 2021-02-17:
But the request is very reasonable. There is not much reason to not use strictly only key authentication for SSH access 👍.

Joulinar, 2021-02-17:
But this could complicate thinks for most users, as they need to exchange keys before. Isn't it? We quite a large amount of users within limited knowledge on all these thinks  🤔

MichaIng, 2021-02-17:
I do not mean to apply key authentication automatically on DietPi, which would mean to create keys on the server to move to the clients, where sometimes initial access is only possible/applicable with SSH. This is a chicken-egg issue, so initial setup must be password authentication.

But a good documentation about how to switch to key authentication is reasonable:
- Create client keys on the client, not on the server, e.g. PuTTY key generator for Windows client.
- Add only the public client to to `~/.ssh/authorized_keys` and enable key authentication (without disabling password authentication yet).
- Login via SSH from that client, to verify it's authenticating with the key. On success, disable password authentication.

When we find time, we could also add a simple UI for this (e.g. to `dietpi-config`), where one or more keys for a specific user can be added and password authentication toggled.

Joulinar, 2021-02-17:
Ah ok, this would be fine that way.

ovz93br43v7, 2021-02-18:
One thing I missed: I use on my server the email notification (guide from [here](https://www.thomas-krenn.com/de/wiki/Absicherung_eines_Debian_Servers#Benachrichtigung_bei_SSH_Login)) to inform me when somebody logs into the server via SSH. So probably this could be considered to add to your config-tool....