MichaIng/DietPi #4119: Every tried to apply for prototype fund?!?

ovz93br43v7, 2021-02-17:
I think your project should apply for the public sponsorship "[prototype fund](https://prototypefund.de/bewerbung/faq/)"

Joulinar, 2021-02-17:
Luckily part of us are located in Germany 🤣

ovz93br43v7, 2021-02-17:
Probably other countries have similar sponsorship's, I don't know...

MichaIng, 2021-02-17:
Very interesting and actually a great idea. We have to think through it how to handle contributions of non-Germans to not create a two-class contributors situation, at least among those which are currently regularly contributing. And a large amount of invested time is for end user support, debugging, testing and writing tasks indeed, which I would never be able to deal with along without significant impact on the project development, so it wouldn't be fair if I'd apply for this funding alone. Probably with donations and material costs funding there is a way to compensate non-German contributions during the 6 months without violating the rules of the program.

I'll have a look through the application form.

MichaIng, 2021-06-02:
Sadly this time it didn't work. There were about 300 applications, so I do not take it negative. A problem is that DietPi exists already, not a prototype, and with a team, so Jury members could see it as something that will go on even without funding. We tried to pronounce specific aspects, concentration on making it easy for users to setup various self-hosted cloud-ish servers, in the light of ongoing data scandals of common large providers, GDPR and all discussion around that topic, as the important society aspect DietPi addresses. But in the end it's a platform with many goals and use cases, compared to other projects that got funded in a the past, with a single clear aim and goal.

We'll likely keep sending applications with next round and try to better outline a clear distinct feature or aspect we want to address and develop within the funding period.

So much FYI, as the idea was a great one and it's good in general to keep an eye on public funding programs for projects like ours. I'll however mark this issue as closed. We'll inform everyone if any additional funding has been acquired, of course.

closed 2021-06-02 by MichaIng