MichaIng/DietPi #4121: Open Beta v7.0 | Please help testing and hardening the upcoming release

MichaIng, 2021-02-18:
RC version | v7.0.0
---------- | -------
v7.0.0 PR | #4110
Changelog | https://github.com/MichaIng/DietPi/blob/beta/CHANGELOG.txt
Code changes | https://github.com/MichaIng/DietPi/compare/master...beta
How to apply | https://github.com/MichaIng/DietPi/blob/master/BRANCH_SYSTEM.md
Release planned | Til end of February

#### Related/solved issues: https://github.com/MichaIng/DietPi/issues?q=is%3Aissue+milestone%3Av7.0

#### Better formatted overview 😃: https://github.com/MichaIng/DietPi/pull/4110#issue-573163760

Joulinar, 2021-02-18:
@MichaIng 

update on an empty system is working.
<details>
  <summary>Logfile attached. Click to expand!</summary>
```
root@DietPi4:~# G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=beta' /boot/dietpi.txt
[  OK  ] G_CONFIG_INJECT | Setting in /boot/dietpi.txt adjusted: DEV_GITBRANCH=beta
root@DietPi4:~# dietpi-update

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for available DietPi update

[ INFO ] DietPi-Update | Checking mirror: https://raw.githubusercontent.com/MichaIng/DietPi/beta/dietpi/server_version-6
[  OK  ] DietPi-Update | Using update server: https://raw.githubusercontent.com/MichaIng/DietPi/beta/dietpi/server_version-6
[  OK  ] DietPi-Update | Update available:
[ INFO ] DietPi-Update | Current version : v6.34.3 (branch: MichaIng/master)
[ INFO ] DietPi-Update | Latest version  : v6.35.-1 (branch: MichaIng/beta)

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for update pre-requirements

[  OK  ] DietPi-Update | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Update | Free space check: path=/ | available=13514 MiB | required=100 MiB
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Applying pre-patches

[  OK  ] DietPi-Update | Downloading pre-patch file
[  OK  ] DietPi-Update | Applying execute permission
[  OK  ] DietPi-Update | Successfully applied pre-patches

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Upgrading APT packages

[ INFO ] DietPi-Update | APT update, please wait...
Hit:1 https://deb.debian.org/debian buster InRelease
Hit:2 https://deb.debian.org/debian buster-updates InRelease
Hit:3 https://deb.debian.org/debian-security buster/updates InRelease
Hit:4 https://deb.debian.org/debian buster-backports InRelease
Hit:5 https://archive.raspberrypi.org/debian buster InRelease
Reading package lists...
[  OK  ] DietPi-Update | APT update
[ INFO ] DietPi-Update | APT upgrade, please wait...
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 16431 files and directories currently installed.)
Preparing to unpack .../libssl1.1_1.1.1d-0+deb10u5_arm64.deb ...
Unpacking libssl1.1:arm64 (1.1.1d-0+deb10u5) over (1.1.1d-0+deb10u4) ...
Preparing to unpack .../libisc-export1100_1%3a9.11.5.P4+dfsg-5.1+deb10u3_arm64.deb ...
Unpacking libisc-export1100:arm64 (1:9.11.5.P4+dfsg-5.1+deb10u3) over (1:9.11.5.P4+dfsg-5.1+deb10u2) ...
Preparing to unpack .../libdns-export1104_1%3a9.11.5.P4+dfsg-5.1+deb10u3_arm64.deb ...
Unpacking libdns-export1104 (1:9.11.5.P4+dfsg-5.1+deb10u3) over (1:9.11.5.P4+dfsg-5.1+deb10u2) ...
Preparing to unpack .../openssl_1.1.1d-0+deb10u5_arm64.deb ...
Unpacking openssl (1.1.1d-0+deb10u5) over (1.1.1d-0+deb10u4) ...
Setting up libssl1.1:arm64 (1.1.1d-0+deb10u5) ...
Setting up libisc-export1100:arm64 (1:9.11.5.P4+dfsg-5.1+deb10u3) ...
Setting up openssl (1.1.1d-0+deb10u5) ...
Setting up libdns-export1104 (1:9.11.5.P4+dfsg-5.1+deb10u3) ...
Processing triggers for libc-bin (2.28-10) ...
[  OK  ] DietPi-Update | APT upgrade

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Installing new DietPi code
[... 66 lines cut ...]
[ INFO ] DietPi-Update | APT autopurge, please wait...
[  OK  ] DietPi-Update | APT autopurge
[  OK  ] DietPi-Update | Incremental patching to v7.0.-1 completed
[ SUB2 ] DietPi-Update > Completed
[ INFO ] DietPi-Update | Current version : v7.0.-1
[ INFO ] DietPi-Update | Latest version  : v7.0.-1
[  OK  ] DietPi-Survey | Purging survey data
[ SUB3 ] DietPi-Services > restart
[  OK  ] DietPi-Services | restart : cron
[  OK  ] DietPi-Patch | Everything done! Terminating the obsolete DietPi-Update parent instance...

Terminated
root@DietPi4:~#
```
</details>

One observation. On some places it is station 6.35 insted of 7.0

`[ INFO ] DietPi-Update | Latest version  : v6.35.-1`
and
```
[ INFO ] DietPi-Update | Current version : v6.34.3
[ INFO ] DietPi-Update | Latest version  : v7.0.-1
[ INFO ] DietPi-Patch | Patching 6.34 to 6.35
```

Joulinar, 2021-02-18:
I guess we still need to delay `unbound` service start, otherwise `anchor` is failing on b oot

```
Feb 18 23:02:59 DietPi4 systemd[1]: Starting Unbound DNS server...
Feb 18 23:03:00 DietPi4 package-helper[423]: /var/lib/unbound/root.key has content
Feb 18 23:03:00 DietPi4 package-helper[423]: fail: the anchor is NOT ok and could not be fixed
Feb 18 23:03:00 DietPi4 unbound[431]: [431:0] info: start of service (unbound 1.9.0).
Feb 18 23:03:00 DietPi4 systemd[1]: Started Unbound DNS server.
```

Joulinar, 2021-02-18:
there is an issue installing Python

```
[  OK  ] DietPi-Software | APT install for: python3-dev
[  OK  ] DietPi-Software | G_THREAD: All threads finished
[  OK  ] DietPi-Software | Verifying download target: get-pip.py
[ INFO ] DietPi-Software | Updating file: get-pip.py
[FAILED] DietPi-Software | mv get-pip.py get-pip.py
```

MichaIng, 2021-02-19:
pip install fixed: https://github.com/MichaIng/DietPi/commit/bb226bc80cb5770fdf165d4bb66d57d6c977663a
__________
Yes that little confusing versioning info during the update is hard to mitigate: DietPi versions 6.33 and prior can only update to 6.x versions. If they would see a 7.x version, the update would be denied with an info that a new image must be flashed. The idea indeed was to have major/core version increments as radical update breaks, as it was with v159 to v6.0 that time, but I don't want to keep this idea. For this reason the update is done in two stages. First the system sees v6.35 and pulls the new code. Then the script is restarted to allow checking the new remote version file with v7.0 info inside. I guess we'll apply at least one incremental patch to that file during beta, then there would be a:
```
[ INFO ] DietPi-Patch | Patching 6.35 to 7.0
```
at least, so that v6.35 is an intermediate version, which is somehow correct.
__________
To fix the Unbound startup, maybe we need to dare something that I wanted to actually do a little later. Does the following help?
```
mkdir -p /etc/systemd/system/ifup@.service.d
echo -e '[Service]\nType=oneshot' > /etc/systemd/system/ifup@.service.d/dietpi.conf
```

Joulinar, 2021-02-19:
> To fix the Unbound startup, maybe we need to dare something that I wanted to actually do a little later. Does the following help?
```
mkdir -p /etc/systemd/system/ifup@.service.d
echo -e '[Service]\nType=oneshot' > /etc/systemd/system/ifup@.service.d/dietpi.conf
```

That's seems to be working 👍 

```
Feb 19 13:22:58 DietPi4 systemd[1]: Starting Unbound DNS server...
Feb 19 13:23:14 DietPi4 package-helper[450]: /var/lib/unbound/root.key has content
Feb 19 13:23:14 DietPi4 package-helper[450]: success: the anchor is ok
Feb 19 13:23:14 DietPi4 unbound[459]: [459:0] info: start of service (unbound 1.9.0).
Feb 19 13:23:14 DietPi4 systemd[1]: Started Unbound DNS server.
```

MichaIng, 2021-02-19:
Okay great so far. Let's implement this as fixed service override for all DietPi systems, which has the following advantage:
- The systemd `network-online.target` gets a much better meaning, since `ifup@.service` must now finish (instead of only start) before this is reached. Especially in case of DHCP, it was pretty common that `network-online.target` was reached even that no IP address was assigned yet. Adding `After=dietpi-boot.service` to network services was kinda workaround since it (when not disabled) waits for a default route, which implies an assigned IP. But it's much better to check/wait for the native ifupdown services which have exactly and only that purpose.

Here is the reason why this has been changed by Debian in the first place: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=791920
- If nothing about this has changed, it means that services which are waiting for `network.target` or `network-online.target` would be delayed if an Ethernet interface was defined/enabled (added as "allow-hotplug" in `/etc/network/interfaces`) but no cable attached. I'll check on VM how this behaves.
A counter argument is that `networking.service`, which handles `auto <iface>` entries (not used by us, but actually reasonable for onboard adapters) currently is still "Type=oneshot", hence suffers from exactly the same issue. So reverting `ifup@.service` to Type=simple was actually an inconsistent step when leaving `networking.service` as "Type=oneshot". And IMO the problem that `network-online.target` has no meaning is much larger than the problem that some services are delayed when someone unplugs an Ethernet cable without removing the related interface entry from the config file, especially on server systems which we address mostly.

Joulinar, 2021-02-19:
I hope it's not creating to much work to change the behaviour. At least it require a change on all services that require a valid network connection during start up. First thin stepping into my mind, VPN servers 😃

MichaIng, 2021-02-19:
> I hope it's not creating to much work to change the behaviour. At least it require a change on all services that require a valid network connection during start up.

Nope, it does not require any change. All server services which provide something on a network port have `After=network.target` already, so will be now delayed until interfaces have been configured, instead of waiting for the loopback interface only. But this affects systemd-controlled services only, anyway, while on DietPi many after started by dietpi-postboot.services later anyway, which is why this whole topic was only an issue in rare cases. Since Unbound is systemd-controlled, to not have dietpi-services possibly break the systems own DNS resolving, it is affected.

The services files installed by `dietpi-software` (not part of a package/installer) contain `After=network-online.target dietpi-boot.service` currently. The letter could not be removed, although it assures the default route and synced network time, which is often required as well when services perform HTTPS requests. So for now I don't see any need to change anything else 🙂.

Andymc1280, 2021-02-20:
Is this beta also useable for those of us who installed the x64 version ?

ravenclaw900, 2021-02-20:
Yes, it is.

MichaIng, 2021-02-20:
Note that `G_DEV_BRANCH` lowers the current subversion by one, hence re-applies v6.33 to v6.34 patches when being on v6.34 currently. So I would follow the instructions linked above to avoid that overhead:
```
G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=beta' /boot/dietpi.txt
dietpi-update 1
```

An yes, DietPi beta versions are generally usable on all SBCs/architectures/systems 🙂.

Joulinar, 2021-02-20:
I tested Bitwarden_RS. New certificate is working well with client Apps on W10 20H2, Android10 and iOS 14.4

MichaIng, 2021-02-21:
#### Beta v7.0.0 has been merged: https://github.com/MichaIng/DietPi/pull/4132
In case anyone is confused, since I forgot to raise the RC version string with the first beta, it was actually `v7.0.-1` 😉.

Joulinar, 2021-02-23:
@MichaIng 
can we add one step to the update process? Jellyfin pgp key expired and would need to be replaced. Challange is, it would need to executed as first step, otherwise `apt update` will fail. Or should it go via MOTD?

https://dietpi.com/phpbb/viewtopic.php?p=31985#p31985

MichaIng, 2021-02-23:
> it would need to executed as first step

This is what we have pre-patches for 🙂.

tindecken, 2021-02-25:
Hi @MichaIng, 
Is there any way to upgrade to current dietpi version: v6.34.3 to v7.0 beta?

[image]

Regards,

Joulinar, 2021-02-25:
The first post above contains a link how to apply the beta patch. https://github.com/MichaIng/DietPi/blob/master/BRANCH_SYSTEM.md

Joulinar, 2021-02-27:
@MichaIng 
I tested installation of Docker in combination with Docker-Compose and Portainer. All working fine.

MichaIng, 2021-02-27:
#### Beta v7.0.1 has been merged: #4146

MichaIng, 2021-02-28:
#### DietPi v7.0 has been released: #4126

closed 2021-02-28 by MichaIng