MichaIng/DietPi #4127: Can't see the rock64 in the network

Maaago, 2021-02-19:
#### Required Information
- DietPi: the one you get from the download page on the dietpi's site (I don't know the version as that image never worked for me, but I think it's 6.33 or 6.34)
- Power supply used: 5V 3A
- SDcard used: Samsung EVO Plus 32Gb

#### Steps to reproduce
1. Download the image from the Dietpi's site
2. Flash the card with balenaEtcher (on macOS)
3. Put it in the rock64 and start the board
4. Use an IP scanner (e.g. Angry IP Scanner)

#### Expected behaviour
See the rock64's IP in the scanner

#### Actual behaviour
Can't see the IP in the scanner

#### Extra details
1. Ethernet's green light is often off, but sometimes, when I reboot the board, the light turns on. In each case it's state doesn't change (i.e. doesn't changes from on to off or viceversa) unless reboot
2. The red led is blinking (heartbeat) at different speed: again sometimes, when I reboot, the led blinks at a high speed, but in most cases the led blinks at idle speed
3. The board worked only once (because the diagnostic was too easy): it was the first boot after, can't remember, maybe the second or the third time I flashed the card. The first boot after flashing worked, but I had instread some software related problems, the following reboots didn't worked at all. The software problems I had were with apt update that gave me an error the first time it was invoked by Dietpi-Update, rerunning the command solved the problem, but then the first boot script got stuck. I stopped it and then I run apt upgrade by my self, but after a while the whole board got stucked. I was using SSH
4. I did a fresh install of Armbian and used the prep script: it "worked" immediately. I still have some (kernel related I think) problems: the board gets stuck, sometimes with the red led blicking others with the red led off. Anyway I think it's an Armbian related problem

Maaago, 2021-02-20:
Today the board booted a few times, but I can only confirm the system is unusable: I've never completed the first login setup as it always gets stuck. It seems to get stuck quite randomly although I noticed that everytime the board was executing a job, like apt upgrade or similar. I had some kernel panic during a couple boots too. The interesting thing is that once I got a self-detected stall on CPU error and that's similar to the errors I get when I type dmesg in the Armbian+Prep script OS when it gets stuck (point 4). The error I got is (hand-copied):

[220.873840] rcu: INFO: rcu_preempt self-detected stall on CPU
[220.879592] rcu: $3-...0: (32285498 ticks this GP) idle=256/0/0x1 softirq=5196/5196 fsq=7228

At this point I can assume everything is due to a kernel related error, but I still can't explain to myself why with DietPi I can't complete the first boot script, but with Armbian+Prep script I can use the system quite "normally" (of course it always got stuck before 48h of uptime)

MichaIng, 2021-02-20:
Which Armbian image did you use, the legacy kernel 4.4 or mainline kernel 5.10 one?

And please try to limit the CPU frequency to 1.5 GHz:
```
G_CONFIG_INJECT 'CONFIG_CPU_MAX_FREQ=' 'CONFIG_CPU_MAX_FREQ=1500' /boot/dietpi.txt
echo '1500000' > /sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq
```

Maaago, 2021-02-20:
Sorry I forgot to mention it in this issue (but wrote it in #4129), I'm running the linux-5.10.16-rockchip64 kernel.
`G_CONFIG_INJECT 'CONFIG_CPU_MAX_FREQ=' 'CONFIG_CPU_MAX_FREQ=1500' /boot/dietpi.txt` fails: "[FAILED] File does not exist or cannot be written to by current user". Running with sudo gives me "sudo: G_CONFIG_INJECT: comando non trovato" (command not found), so I've changed the file manually. I had to modify manually also the scaling_max_freq file (wich already had the number 1296000). Let's see what happens...

ravenclaw900, 2021-02-20:
Yeah, you have to actually log in as root (`sudo su`) before you can run any of the `G_` commands.

Maaago, 2021-02-20:
That's true, I do it right now

MichaIng, 2021-02-20:
```
G_SUDO G_CONFIG_INJECT 'CONFIG_CPU_MAX_FREQ=' 'CONFIG_CPU_MAX_FREQ=1500' /boot/dietpi.txt
```
works. Yeah sudo by default resets the environment (hence the G_ functions), G_SUDO loads them again before invoking sudo. `su` works as well as it does not reset the environment, but since this is true for each and every variable, I generally recommend against `su`. E.g. it might passes PATH without /sbin and /usr/sbin, so commands might be missing which should actually available for root 😉.

Maaago, 2021-02-20:
I think I've found something: https://forum.armbian.com/topic/15082-rock64-focal-fossa-memory-frequency/?tab=comments#comment-108127
I remember that the first ayufan image I've tried was super unstable, but with the latest ayufan image I used (for more than a year) I've never had this issues. Maybe that image already had some downclocks (CPU or memory)... I'll see if CPU downclock is enough, if not i'll try to downclock the memory and will let you know how it works.

Maaago, 2021-02-26:
I'm back! I've tested a vanilla Armbian as well as a "patched" Armbian as said in the link in the previous post, but none of them is stable. As this is an Armbian related problem I'll close the issue.

closed 2021-02-26 by Maaago

MichaIng, 2021-02-26:
For for the record: The newest Armbian kernel does not overclock those Rockchip SBCs anymore by default, but instead an optional device tree overlay is now available to do so: #4139