MichaIng/DietPi #4129: Network dead after install Motioneye

lupa18, 2021-02-20:
### Creating a bug report/issue

#### Required Information
- DietPi version | last updated (can't access now)
- Distro version | 10.8
- Kernel version | ?
- SBC model | rock64

#### Additional Information (if applicable)
- Software title | Motioneye
- Can this issue be replicated on a fresh installation of DietPi? Yes. Is the second time that happen. First time I think it was my error but several weeks after, I tried again and I get the same problem. 

#### Steps to reproduce

1. Install Motioneye
2. It ask for reboot, so reboot
3. You lost access to your dietpi

#### Expected behaviour

- Motioneye should be installed and you must have access to your dietpi as usual

#### Actual behaviour

- I lost acces. I have to connect using a display and a keyboard.

Joulinar, 2021-02-20:
Hi
many thanks for your message. I did a test installation on my RPi4B 64bit without issues. After reboot the system was still reachable via SSH. Are there any error messages on screen once you connected a display and reboot your system? Once you connected locally, are you able to check status if interface using `ip a`

MichaIng, 2021-02-20:
Additionally to `ip a`, please show the following as well:
```
journalctl -u networking
journalctl -u ifup@eth0 # If you use Ethernet
journalctl -u ifup@wlan0 # if you use WiFi
```

Maaago, 2021-02-20:
Don't know if can be related, but as I said in #4127 SSH access worked only once for me after a fresh install, but when I rebooted the board I lost the access too. The difference is that I didn't install anything, just did apt update && apt upgrade.

MichaIng, 2021-02-20:
Well, software installs do not change something about the major network connection, neither is something changed about it between first login and first reboot, but APT packages are upgraded in both cases, including the kernel in case. So I suspect in both cases the latest Armbian kernel being the issue.

The following could be tried to downgrade to one version earlier:
```
cd /tmp
wget https://apt.armbian.com/pool/main/l/linux-5.10.12-rockchip64/linux-dtb-current-rockchip64_21.02.1_arm64.deb
wget https://apt.armbian.com/pool/main/l/linux-5.10.12-rockchip64/linux-image-current-rockchip64_21.02.1_arm64.deb
apt install ./*.deb
rm ./*.deb
reboot
```
Let's see if that helps.

There is not yet any report on their forums, but that's where to keep an eye on: https://forum.armbian.com/forum/21-rockchip-3288-3328/

Maaago, 2021-02-20:
As I wrote in the other issue I can't test these commands in the DietPi from the image downloaded from the DietPi's site. In the Armbian+Prep script OS I run the linux-5.10.16-rockchip64 and works, but after some hours of uptime I get some preempt errors

Joulinar, 2021-03-10:
Guys, can this be closed as the other issue was closed already?

MichaIng, 2021-03-10:
There as an Armbian update recently. Worth to give it a try: `apt full-upgrade`

MichaIng, 2021-03-11:
I updated the image: https://dietpi.com/downloads/images/DietPi_ROCK64-ARMv8-Buster.7z
Probably the kernel upgrade solves the network issues.

Maaago, 2021-03-11:
I didn't say anything because I want to test more my Armbian+Prep script with the 5.10.20 kernel, but yes, it seems that my network issues and also the CPU stalls were solved with this kernel update. As soon as I can I'll test the new DietPi image you've uploaded.

lupa18, 2021-03-12:
I didn't say anything because my SD is dead and I must get a new one.

lupa18, 2021-03-20:
I finally found my logs: [logs_dietpi.zip](https://github.com/MichaIng/DietPi/files/6174930/logs_dietpi.zip)

I think dietpi-boot.log has somethint to look at.

Joulinar, 2021-03-20:
hmm not sure if this is related but the system is trying to use following file `/DietPi/dietpi/.network` which does not exist anymore. The whole DietPi script directory has been moved to `/boot` a couple of version before. You could try creating a sym link pointing from `/DietPi` > `/boot`

lupa18, 2021-03-20:
Hello @Joulinar Thanks for your answer. 

The fact is, 2 times I installed motioneye, 2 times y lost connection an had to reinstall. That logs are from my previous installation. 

I don't dare to install motioneye again. Maybe when I get a new board.

MichaIng, 2021-03-21:
As MotionEye and general network have nothing to do with each other, it's 99,99%  assured that it's instead related to the previous kernel upgrade. Such issues are one reason why dietpi-software installs won't apply a general apt upgrade anymore since DietPi v7.0. But the very first boot will.

With Debian Bullseye we'll move from Armbian kernels to Debian kernels, I hope this will enhance the stability, but means that the kernel versions will be older, not as fast on bleeding edge like with Armbian :wink:.

Maaago, 2021-03-21:
After a week of testing the board never had problems with the new image, so at this point I think it would be a good idea to let users choose the kernel's version. According to me this makes even more sense if DietPi will move to another distro's kernel. I know armbian-config has a menu entry for this. I'll open a feature request.

closed 2021-05-07 by MichaIng