MichaIng/DietPi #4133: Mosquitto: In default mosquitto.conf, add a listener on port 1883

mattsmithuk, 2021-02-21:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- Mosquitto versions prior to version 2 would work out of the box. Versions after 2.0 need a listener setting up in configuration before they will work across a network

#### Describe the solution you'd like:

- Have the default mosquitto.conf file to have the following lines:
- ```listener 1883
- allow_anonymous true```
- Alternatively, have a user and password which matches global dietpi setup
 
#### Describe alternatives you've considered:

- Display instructions to setup mosquitto.conf after installation

Joulinar, 2021-02-21:
Probably something like this could be documented on our online docs https://dietpi.com/docs/software/hardware_projects/#mosquitto

MichaIng, 2021-02-22:
Relevant for Bullseye: https://packages.debian.org/mosquitto
The manpage reflexes it: https://manpages.debian.org/unstable/mosquitto/mosquitto.8.en.html#OPTIONS

Jep we should definitely set it up with a default port, that simply matches the previous default, IMO. Any preference about whether to allow anonymous/unauthenticated access by default or setting it up with username and password (via `mosquitto_passwd` to have it hashed and with dedicated stricter permissions)?

slopsjon, 2021-03-07:
Mosquitto was working no problem on version DietPi 6.34 upgraded to 7.0.2 this morning and it stopped working. No service running and would not start, so I had to reinstall Mosquitto and that got the service running but then no connections. Until I read this and put  
listener 1883
allow_anonymous true
into the conf file and now all working again.

MichaIng, 2021-03-08:
Which system are you using? RPi or other, Buster or Bullseye?
Should be only relevant on Bullseye, since on Buster there was no package update since 2019.

slopsjon, 2021-03-08:
> 
> 
> Which system are you using? RPi or other, Buster or Bullseye?
> Should be only relevant on Bullseye, since on Buster there was no package update since 2019.

It's on an RPi 3B+ Buster.
At the same time my DNS server kept crashing which is on another SBC (Pihole+Unbound). The only way to stop it crashing was to kill the Unbound service. I then added the above two lines to mosquitto.conf and rebooted the mosquitto server that worked and then rebooted the DNS server and that stopped crashing.

MichaIng, 2021-03-08:
Very strange, as Unbound runs on a very different port, and it would be too much coincidence when Mosquitto would listen/bind to port 5335 by default, when none is set.

However, it does not hurt and is probably a good idea anyway to set the port on every distro version, even if it's only to expose that option transparently.

MichaIng, 2021-04-04:
I'll go with a password file. It makes sense to not allow unauthenticated remote requests by default. Any preferences on the username? `dietpi` or `mosquitto`? Password will be the global software password of course, and if the password file already exists, it won't be touched, of course.

Joulinar, 2021-04-04:
Personally I would vote for user `mosquitto`. We have other applications where user is identical with the application name like Qbit

MichaIng, 2021-04-04:
In case of qBittorrent its necessary as the login user is the UNIX user, which is quite uncommon otherwise. But there are other cases. Would be actually nice to have this aligned, either "dietpi" or the application name. Most importantly, "root" and "admin" should be avoided (also for this we have cases) to make it not too easy for brute-force attacks 😉.

Joulinar, 2021-04-04:
> to make it not too easy for brute-force attacks 

In this case, let's auto generate users per per installation individually 🤣

MichaIng, 2021-04-04:
> In this case, let's auto generate users per per installation individually 🤣

Would be the safest option, although then better leave remote requests disabled by default 😄. But seriously, it's crazy how many bots try to login via the non-existing "admin" account on our Wordpress instance.

Actually nice would be to have a "global software user", similarly like the global software password. For ownCloud/Nextcloud, we have a `dietpi.txt` option, while by default "admin" is used (hardcoded default for the CLI install command as well) 😅. Makes sense to change that as well and make that option a global software authentication user. So I can use "MichaIng" or "Micha", which matches the user I'd want to create in most cases anyway.

But, to not mix too many changes together, "dietpi" or "mosquitto" will do for now 😄.

closed 2021-04-24 by MichaIng