MichaIng/DietPi #4134: Dietpi drive-manage cannot handle second nfs mount

PrzemekSkw, 2021-02-21:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
```
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=34
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```

- Distro version | `buster`
- Kernel version | `Linux DietPi 5.10.11-v7+ #1399 SMP Thu Jan 28 12:06:05 GMT 2021 armv7l GNU/Linux
`
- SBC model | `RPi 3 Model B+ (armv7l)` 
- Power supply used | original Rpi power supply

I have two HDD connected with usb cables (+power supply) to Rpi4. Drive 1 - Seagate, drive 2 - Toshiba . I have nfs server installed and share that drives to other devices. When I install Dietpi on other Rpi2,3 and try to mount those drives Dietpi drive-manager see only Seagate. That is larger drive 6GB. If I want to also add second drive Toshiba I have to add manually in fstab because drive-manager see only Seagate and when try to add it like 192.168.0.24:/mnt/Toshiba it add second time Seagate drive as Tosiba.
Regards,
Przemek

MichaIng, 2021-02-22:
Indeed, the issue is that we do not allow to add a sub directory but just the IP, in which case the first NFS export from the server is connected to, I guess.

Shouldn't be a big deal to allow adding a path as well, in case the server has multiple exports/shares. We already allow this for NFSv3 where a full path is required.

PrzemekSkw, 2021-02-22:
@MichaIng Hi, thanks for response. It's no problem to add second nfs drive but it will be very good option to add multiple drives in future releases.
Regards,
Przemek

Joulinar, 2021-05-08:
@MichaIng 
what about an option to use `showmount -e 192.168.x.x` and provide a list of possible mounts? I know it's more complicate but would reduce typos 😃

MichaIng, 2021-05-08:
Stephan suggested exactly that here: #2519
Now I see it's basically a duplicate. Multiple mounts from the same server and adding paths are one and the same thing, as the paths differentiate the multiple mounts. Marking this hence as closed.

closed 2021-05-08 by MichaIng