MichaIng/DietPi #4135: Nextcloud Talk won't install

BugDave, 2021-02-22:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.34.3
- Distro version | buster
- Kernel version | Linux DietPi 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux
- SBC model | RPi 4 Model B (armv7l)
- Power supply used | Canakit that came with pi4b
- SDcard used | Samsung evo

#### Additional Information (if applicable)
- Software title | Nextcloud
- Was the software title installed freshly or updated/migrated? | Fresh Install
- Can this issue be replicated on a fresh installation of DietPi? | Yes

- Bug report ID | 0f767985-7352-4bde-9e92-236a97471c40

#### Steps to reproduce

1. ... Install Nextcloud
2. ...Install Nextcloud talk

#### Expected behaviour

- ...Nextcloud talk should install but is saying it's not supported on version 21

#### Actual behaviour

- Spreed is failing to install

#### Extra details

- It looks like nextcloud 21 came out yesterday and isn't compatible with the current release of nextcloud talk? Is there a way to install v20 of nextcloud so i can install nextcloud talk until it's compatible?

Joulinar, 2021-02-22:
Hi,

many thanks for your message. Just a short question. Is that you how open same topic on or forum? https://dietpi.com/phpbb/viewtopic.php?p=31972#p31972

BugDave, 2021-02-22:
did you already have nextcloud installed? My issue is with a fresh install of both nextcloud and nextcloud talk

Joulinar, 2021-02-22:
on my test I installed them one by one separately

BugDave, 2021-02-22:
i just installed again, nextcloud binary host is being hammered but after a few timeouts it installed spreed this time without error. Maybe they pushed and update? Looks good now

closed 2021-02-22 by BugDave

Joulinar, 2021-02-22:
Probably. I had similar experiences today. Binary download was extremely slow. But this is something out of our control.

MichaIng, 2021-02-22:
Servers were overloaded likely because of today's NC 21 release and announcement with live stream: https://nextcloud.com/blog/nextcloud-hub-21-out-with-up-to-10x-better-performance-whiteboard-and-more-collaboration-features/