MichaIng/DietPi #4136: certificate not trusted

ewallis, 2021-02-22:
#### Details:
- Date           | Mon Feb 22 09:15:44 EST 2021
- DietPi version | v6.34.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 2 Model B (armv7l) (ID=2)
- Kernel version | Linux piclock 5.10.11-v7+ #1399 SMP Thu Jan 28 12:06:05 GMT 2021 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | apt-get -q update
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Hit:1 http://packages.microsoft.com/repos/code stable InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Ign:3 https://archive.raspberrypi.org/debian buster InRelease
Err:4 https://archive.raspberrypi.org/debian buster Release
  Certificate verification failed: The certificate is NOT trusted. The certificate issuer is unknown.  Could not handshake: Error in the certificate verification. [IP: 176.126.240.86 443]
Reading package lists...
E: The repository 'https://archive.raspberrypi.org/debian buster Release' no longer has a Release file.
```

Joulinar, 2021-02-22:
Hi,

many thanks for your message. Can you try to install `ca-certificates` again

`apt --reinstall install ca-certificates`

ewallis, 2021-02-22:
Thanks, just did that, same result when running apt update:

root@piclock:~# apt --reinstall install ca-certificates
Reading package lists... Done
Building dependency tree
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 0 not
upgraded.
Need to get 166 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 http://mirror.umd.edu/raspbian/raspbian buster/main armhf
ca-certificates all 20200601~deb10u2 [166 kB]
Fetched 166 kB in 1s (246 kB/s)
Preconfiguring packages ...
(Reading database ... 33825 files and directories currently installed.)
Preparing to unpack .../ca-certificates_20200601~deb10u2_all.deb ...
Unpacking ca-certificates (20200601~deb10u2) over (20200601~deb10u2) ...
Setting up ca-certificates (20200601~deb10u2) ...
Updating certificates in /etc/ssl/certs...
0 added, 0 removed; done.
Processing triggers for ca-certificates (20200601~deb10u2) ...
Updating certificates in /etc/ssl/certs...
0 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d...
done.
root@piclock:~# apt update
Hit:1 http://packages.microsoft.com/repos/code stable InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Ign:3 https://archive.raspberrypi.org/debian buster InRelease
Err:4 https://archive.raspberrypi.org/debian buster Release
  Certificate verification failed: The certificate is NOT trusted. The
certificate issuer is unknown.  Could not handshake: Error in the
certificate verification. [IP: 176.126.240.86 443]
Reading package lists... Done
E: The repository 'https://archive.raspberrypi.org/debian buster Release'
no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore
disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration
details.

MichaIng, 2021-02-22:
Your system clock is correct? `date`
If so, could you paste the output of the following command:
```
openssl s_client -connect archive.raspberrypi.org:443
```

ewallis, 2021-02-22:
Sorry for wasting your time, looks like my org is doing ssl inspection and
that is causing some trust issues.

Joulinar, 2021-02-22:
I'm going to close the call, feel free re-open if needed

closed 2021-02-22 by Joulinar