MichaIng/DietPi #4137: Network mount inside Windows Domain not working anymore with 6.34.3

Loader23, 2021-02-22:
### Creating a bug report/issue

#### Required Information
- DietPi version | `
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=34
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
`
- Distro version | `buster`
- Kernel version | `Linux Dietpi 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux
`
- SBC model | `RPi4
- Power supply used | RPi Power Supply
- SDcard used | SanDisk

#### Steps to reproduce

1. Have a DietPi inside a Windows domain with Distributed File System (DFS)
2. Try to mount

#### Expected behaviour

- it should work like it did with 6.33.3

#### Actual behaviour

- │
│ [FAILED] Samba mount failed with the following error output:                 │
│                                                                              │
│ mount error(2): No such file or directory                                    │
│ Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)                 │
│ mount error(126): Required key not available                                 │
│ Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)                 │
│ mount error(126): Required key not available                                 │
│ Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)                 │
│ mount error(126): Required key not available                                 │
│ Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)                 │
│ mount error(126): Required key not available                                 │
│ Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)  

#### Extra details

- Everything was working fine with 6.33.3 and after the update to 6.34.3 it stopped working.
- What has changed and how to revert it?

MichaIng, 2021-02-22:
Many thanks for your report.

I don't know much about DFS, it's intended to be mountable as Samba/CIFS mount? How does the fstab entry for that mount look like, respectively the mount command arguments?

Loader23, 2021-02-23:
Yes, normal cifs mount. Thats the entry:

`//192.168.178.1/Folder /mnt/Folder cifs username=user,password=pass,iocharset=utf8,uid=dietpi,gid=dietpi,file_mode=0770,dir_mode=0770,vers=3.1.1,_netdev,nodfs,nofail,noauto,x-systemd.automount`

MichaIng, 2021-02-23:
And what is the output if you do:
```
mount /mnt/Folder
```

Loader23, 2021-02-23:
`Couldn't chdir to /mnt/Folder: No such device`

Joulinar, 2021-02-23:
I guess `Folder` is/was a placeholder and would need to be replace with the existing target folder located in `/mnt`

Loader23, 2021-02-23:
I have of course replaced the placeholder :D

MichaIng, 2021-02-23:
First of all, `ping`ing that CIFS share IP from the DietPi system works, right?

Can you try to remove the `nodfs` option? It seems to have been required for very old servers but is not even listed on current man pages as valid option, should hence definitely not be required for SMB 3.1.1 protocol. If that does not help, please try to remove the `vers=3.1.1` option as well to allow auto-negotiation.

Loader23, 2021-02-23:
Yes a ping is working fine. Removing the nodfs did not work and removing the vers is also not working.

I just tested it with another Pi on version 6.33.3. The Pi never had a mount before. I added the line to fstab and its working fine. I Updated to 6.34.3 -> broken.

Joulinar, 2021-02-23:
what happen if on the Pi running 6.33.3 you are going to do `apt update && apt upgrade` without changing DietPi version. What kernel you are running on 6.33.3? `uname -a`

Loader23, 2021-02-23:
That was the last Pi i had on 6.33.3 :-(

Joulinar, 2021-02-23:
no backup? no saved image stored?

Loader23, 2021-02-23:
I have an Image of dietpi 6.32.x but installing that would also update it to the latest version at startup.

MichaIng, 2021-02-23:
Please run `apt update && apt upgrade` on that system only, at best after a backup 😉, and store the list of upgraded packages somewhere to hopefully find the culprit. Since the last DietPi update did not include any related changes, I'm also pretty sure that it is an APT package related to SMB/CIFS mounts, or the kernel upgrade from 5.4 to 5.10.

Loader23, 2021-02-23:
Maybe i have an Update from 6.33.3.^^ I have to check that tomorrow, the device is offline now. I will post the kernel info then.
Here is the dietpi-update.log:
```
[ SUB1 ] DietPi-Update > Applying pre-patches
[  OK  ] DietPi-Update | Downloading pre-patch file
[  OK  ] DietPi-Update | Applying execute permission
[  OK  ] DietPi-Update | Successfully applied pre-patches
[ SUB1 ] DietPi-Update > Upgrading APT packages
[ INFO ] DietPi-Update | APT update, please wait...
Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:2 https://archive.raspberrypi.org/debian buster InRelease [32.8 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian buster/non-free armhf Packages [1                                                                                                                                                             04 kB]
Get:4 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0                                                                                                                                                              MB]
Get:5 https://archive.raspberrypi.org/debian buster/main armhf Packages [360 kB]
Fetched 13.5 MB in 12s (1084 kB/s)
Reading package lists...
[  OK  ] DietPi-Update | APT update
[ INFO ] DietPi-Update | APT upgrade, please wait...
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 26264 files and directories currently installed.)
Preparing to unpack .../base-files_10.3+rpi1+deb10u8_armhf.deb ...
Unpacking base-files (10.3+rpi1+deb10u8) over (10.3+rpi1+deb10u6) ...
Setting up base-files (10.3+rpi1+deb10u8) ...
Installing new version of config file /etc/debian_version ...
(Reading database ... 26264 files and directories currently installed.)
Preparing to unpack .../xserver-common_2%3a1.20.4-1+rpt2+deb10u3_all.deb ...
Unpacking xserver-common (2:1.20.4-1+rpt2+deb10u3) over (2:1.20.4-1+rpt2+deb10u1                                                                                                                                                             ) ...
Preparing to unpack .../libsystemd0_241-7~deb10u6+rpi1_armhf.deb ...
Unpacking libsystemd0:armhf (241-7~deb10u6+rpi1) over (241-7~deb10u4+rpi1) ...
Setting up libsystemd0:armhf (241-7~deb10u6+rpi1) ...
(Reading database ... 26264 files and directories currently installed.)
Preparing to unpack .../xserver-xorg-core_2%3a1.20.4-1+rpt2+deb10u3_armhf.deb ..                                                                                                                                                             .
Unpacking xserver-xorg-core (2:1.20.4-1+rpt2+deb10u3) over (2:1.20.4-1+rpt2+deb1                                                                                                                                                             0u1) ...
Preparing to unpack .../systemd_241-7~deb10u6+rpi1_armhf.deb ...
Unpacking systemd (241-7~deb10u6+rpi1) over (241-7~deb10u4+rpi1) ...
Preparing to unpack .../udev_241-7~deb10u6+rpi1_armhf.deb ...
Unpacking udev (241-7~deb10u6+rpi1) over (241-7~deb10u4+rpi1) ...
Preparing to unpack .../libudev1_241-7~deb10u6+rpi1_armhf.deb ...
Unpacking libudev1:armhf (241-7~deb10u6+rpi1) over (241-7~deb10u4+rpi1) ...
Setting up libudev1:armhf (241-7~deb10u6+rpi1) ...
Setting up systemd (241-7~deb10u6+rpi1) ...
(Reading database ... 26264 files and directories currently installed.)
Preparing to unpack .../0-systemd-sysv_241-7~deb10u6+rpi1_armhf.deb ...
Unpacking systemd-sysv (241-7~deb10u6+rpi1) over (241-7~deb10u4+rpi1) ...
Preparing to unpack .../1-libegl1-mesa_19.3.2-1~bpo10+1~rpt3_armhf.deb ...
Unpacking libegl1-mesa:armhf (19.3.2-1~bpo10+1~rpt3) over (19.3.2-1~bpo10+1~rpt2                                                                                                                                                             ) ...
Preparing to unpack .../2-libgl1-mesa-dri_19.3.2-1~bpo10+1~rpt3_armhf.deb ...
Unpacking libgl1-mesa-dri:armhf (19.3.2-1~bpo10+1~rpt3) over (19.3.2-1~bpo10+1~r                                                                                                                                                             pt2) ...
Preparing to unpack .../3-libglx-mesa0_19.3.2-1~bpo10+1~rpt3_armhf.deb ...
Unpacking libglx-mesa0:armhf (19.3.2-1~bpo10+1~rpt3) over (19.3.2-1~bpo10+1~rpt2                                                                                                                                                             ) ...
Preparing to unpack .../4-libegl-mesa0_19.3.2-1~bpo10+1~rpt3_armhf.deb ...
Unpacking libegl-mesa0:armhf (19.3.2-1~bpo10+1~rpt3) over (19.3.2-1~bpo10+1~rpt2                                                                                                                                                             ) ...
Preparing to unpack .../5-libglapi-mesa_19.3.2-1~bpo10+1~rpt3_armhf.deb ...
Unpacking libglapi-mesa:armhf (19.3.2-1~bpo10+1~rpt3) over (19.3.2-1~bpo10+1~rpt                                                                                                                                                             2) ...
Preparing to unpack .../6-libgbm1_19.3.2-1~bpo10+1~rpt3_armhf.deb ...
Unpacking libgbm1:armhf (19.3.2-1~bpo10+1~rpt3) over (19.3.2-1~bpo10+1~rpt2) ...
Preparing to unpack .../7-libp11-kit0_0.23.15-2+deb10u1_armhf.deb ...
Unpacking libp11-kit0:armhf (0.23.15-2+deb10u1) over (0.23.15-2) ...
[... cut ...]                  APT_UPDATES=1
[  OK  ] DietPi-Set_software | dietpi.txt verification completed
[  OK  ] verify_dietpi.txt  | Completed
[ SUB1 ] DietPi-Update > Applying incremental patches
[ INFO ] DietPi-Update | Current version : v6.33.3
[ INFO ] DietPi-Update | Latest version  : v6.34.3
[ INFO ] DietPi-Patch | Patching 6.33 to 6.34
[ INFO ] DietPi-Patch | APT autopurge, please wait...
[  OK  ] DietPi-Patch | APT autopurge
[  OK  ] DietPi-Update | Incremental patching to v6.34.3 completed
```

Joulinar, 2021-02-23:
it's doing the Raspberry Pi kernel update. propaly we could try to switch back to 5.4. It's highly recommended to clone your SD card before or do an appropriate backup. 😉 

```
apt update
apt install rpi-update
sudo rpi-update 0642816ed05d31fb37fc8fbbba9e1774b475113f
reboot
```

This will downgrade the kernel to 5.4.79

Loader23, 2021-02-24:
The downgrade to 5.4.79 did not help. But I could recover that one Pi to 6.33.3, the kernel is 5.4.72 there.

Joulinar, 2021-02-24:
Ok If you are ok let's try some more test. Keep the backup of your system on a save spot. On the current system running v6.33.3, just do an `apt update && apt upgrade`

Loader23, 2021-02-24:
After apt update && apt upgrade + reboot its not working anymore.

Joulinar, 2021-02-24:
Than it is one of the apt packages and not the DietPi update. Now we would need to find out which one. Can you go back to the backup before running apt upgrade? And share the list of packages to be upgraded (without actually upgrading them)

Loader23, 2021-02-24:
Here it is :-)

```
  apt base-files ca-certificates chromium-browser chromium-codecs-ffmpeg-extra device-tree-compiler firmware-atheros firmware-brcm80211 firmware-iwlwifi firmware-misc-nonfree firmware-realtek iproute2 libapt-pkg5.0 libcups2
  libdns-export1104 libegl-mesa0 libegl1-mesa libgbm1 libgl1-mesa-dri libglapi-mesa libglx-mesa0 libgnutls30 libisc-export1100 libjpeg62-turbo libldap-2.4-2 libldap-common libp11-kit0 libraspberrypi-bin libraspberrypi0 libsqlite3-0
  libssl1.1 libsystemd0 libudev1 libxml2 libzstd1 openssl raspberrypi-bootloader raspberrypi-kernel raspberrypi-sys-mods rpi-eeprom sudo systemd systemd-sysv tzdata udev unzip xserver-common xserver-xorg-core
```

Joulinar, 2021-02-24:
quite a long list 😃 

@MichaIng 
would it make sense to try to install one by one to see where it is failing?

Loader23, 2021-02-24:
After installing raspberrypi-kernel it is no longer working.

MichaIng, 2021-02-24:
So upgrading the kernel alone broke it, but downgrading the kernel alone did not fix it 🤔? But you did a reboot after kernel upgrade/downgrade, didn't you?

Loader23, 2021-02-24:
Yes, of course.^^

Joulinar, 2021-02-24:
Is this something to report back to Raspberry Foundation as issue? Interesting would be as well how plain Raspberry OS behave. Before and after `apt upgrade`. As well as checking the kernel downgrade.

MichaIng, 2021-02-24:
> Interesting would be as well how plain Raspberry OS behave.

I agree. I'm unsure how to setup such a Windows DFS to replicate/test, which would be great before forwarding to RPi devs. Probably we find some related report on their forum or GitHub.

Ah, when mounting fails, does `dmesg` show any related kernel log entry (at the end of the output)?

Loader23, 2021-02-24:
These are the last lines of dmesg after a reboot:
```
[   14.846642] CIFS: Attempting to mount //IP/Folder
[   15.654995] CIFS: VFS: cifs_mount failed w/return code = -2
[   15.727164] CIFS: Attempting to mount //IP/Folder
[   16.288569] CIFS: VFS: cifs_mount failed w/return code = -126
```

MichaIng, 2021-02-24:
Mount error codes: https://manpages.debian.org/buster/mount/mount.8.en.html#RETURN_CODES

The first error could be related to missing network. Can you please paste the following:
```
journalctl -u mnt-Folder.mount -u mnt-Folder.automount -u network.target -u network-online.target
```
All parts of the mount point path need to be concatenated with a dash, so /mnt/this/point would be `mnt-this-point.mount` and `mnt-this-point.automount`.

Probably a change we do with upcoming release does already fix that part:
```
mkdir -p /etc/systemd/system/ifup@.service.d
curl -sSfL 'https://raw.githubusercontent.com/MichaIng/DietPi/dev/rootfs/etc/systemd/system/ifup%40.service.d/dietpi.conf' > /etc/systemd/system/ifup@.service.d/dietpi.conf
```
- The `_netdev` mount option should assure that the mount is not done before `network-online.target` is reached.
- The above config assures that `network-online.target` is not reached before all network adapters have been fully configured, hence give that target much more meaning.

But actually, the `x-systemd.automount` mount option means that the share is not mounted before the mount point is accessed the first time. Do you have any service or task at boot that accesses that mount so early (~15 seconds after boot?). Usually such network shares are accessed somehow later, which is why the issue with missing network rarely shows up.
__________
But mounting the drive fails as well later + we have a second error code (126), which is a combination of all errors but "incorrect invocation or permissions (1)", not helpful 😄.

Let's have a look at the `sec` mount option, which is about how the password is transferred and where the default seems to depend on the kernel version: https://manpages.debian.org/testing/cifs-utils/mount.cifs.8.en.html#OPTIONS
> ### sec=arg
> Security mode. Allowed values are:
none - attempt to connection as a null user (no name)
krb5 - Use Kerberos version 5 authentication
krb5i - Use Kerberos authentication and forcibly enable packet signing
ntlm - Use NTLM password hashing
ntlmi - Use NTLM password hashing and force packet signing
ntlmv2 - Use NTLMv2 password hashing
ntlmv2i - Use NTLMv2 password hashing and force packet signing
ntlmssp - Use NTLMv2 password hashing encapsulated in Raw NTLMSSP message
ntlmsspi - Use NTLMv2 password hashing encapsulated in Raw NTLMSSP message, and force packet signing
>
> The default in mainline kernel versions prior to v3.8 was sec=ntlm. In v3.8, the default was changed to sec=ntlmssp.
>
> If the server requires signing during protocol negotiation, then it may be enabled automatically. Packet signing may also be enabled automatically if it's enabled in /proc/fs/cifs/SecurityFlags.

I was trying to find out whether there was a change recently: https://wiki.samba.org/index.php/LinuxCIFSKernel#5.5_kernel_.28module_version_2.24.2C_released_January_26th.2C_2020._61_changesets.29
> ### 5.9 kernel (module version 2.28 30 changesets)
Fix idsfromsid mount option when creating directories. **Various DFS (global namespace) fixes.**

But no recent word about sec default. However, could you try:
```
mount -o sec=ntlmssp /mnt/Folder
```
and in case go through the other possible options from the list above? Else we must assume that the "various DFS fixes" contain a regression.

Loader23, 2021-02-24:
Journal:
```
-- Logs begin at Wed 2021-02-24 15:15:07 CET, end at Wed 2021-02-24 16:27:16 CET. --
Feb 24 15:15:10 Pi01 systemd[1]: Reached target Network.
Feb 24 15:15:32 Pi01 systemd[1]: Reached target Network is Online.
```
I have to test the change you have posted tomorrow (device offline now) but `mount -o sec=ntlmssp` also resulted in `No such device
`

MichaIng, 2021-02-24:
`journalctl -u mnt-Folder.mount -u mnt-Folder.automount` doesn't output anything (with adjusted path elements)?

Does `systemctl status mnt-Folder.mount -u mnt-Folder.automount` show something instead?

> I have to test the change you have posted tomorrow (device offline now)

Okay. Probably I find time to setup a DFS here myself in the meantime.

Loader23, 2021-02-24:
Sorry, I missread the dash part.^^Thats the Log:

```
Feb 24 15:15:10 Pi01 systemd[1]: Reached target Network.
Feb 24 15:15:32 Pi01 systemd[1]: mnt-Folder.automount: Got automount request for /mnt/Folder, triggered by 447 (cp)
Feb 24 15:15:32 Pi01 systemd[1]: Reached target Network is Online.
Feb 24 15:15:32 Pi01 systemd[1]: Mounting /mnt/Folder...
Feb 24 15:15:33 Pi01 mount[449]: mount error(2): No such file or directory
Feb 24 15:15:33 Pi01 mount[449]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
Feb 24 15:15:33 Pi01 systemd[1]: mnt-Folder.mount: Mount process exited, code=exited, status=32/n/a
Feb 24 15:15:33 Pi01 systemd[1]: mnt-Folder.mount: Failed with result 'exit-code'.
Feb 24 15:15:33 Pi01 systemd[1]: Failed to mount /mnt/Folder.
Feb 24 15:15:33 Pi01 systemd[1]: mnt-Folder.automount: Got automount request for /mnt/Folder, triggered by 497 (find)
Feb 24 15:15:33 Pi01 systemd[1]: Mounting /mnt/Folder...
Feb 24 15:15:34 Pi01 mount[498]: mount error(126): Required key not available
Feb 24 15:15:34 Pi01 mount[498]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
Feb 24 15:15:34 Pi01 systemd[1]: mnt-Folder.mount: Mount process exited, code=exited, status=32/n/a
Feb 24 15:15:34 Pi01 systemd[1]: mnt-Folder.mount: Failed with result 'exit-code'.
Feb 24 15:15:34 Pi01 systemd[1]: Failed to mount /mnt/Folder.
Feb 24 15:33:36 Pi01 systemd[1]: mnt-Folder.automount: Got automount request for /mnt/Folder, triggered by 777 (sftp-server)
Feb 24 15:33:36 Pi01 systemd[1]: Mounting /mnt/Folder...
Feb 24 15:33:36 Pi01 mount[778]: mount error(2): No such file or directory
Feb 24 15:33:36 Pi01 mount[778]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
Feb 24 15:33:36 Pi01 systemd[1]: mnt-Folder.mount: Mount process exited, code=exited, status=32/n/a
Feb 24 15:33:36 Pi01 systemd[1]: mnt-Folder.mount: Failed with result 'exit-code'.
Feb 24 15:33:36 Pi01 systemd[1]: Failed to mount /mnt/Folder.
Feb 24 15:34:40 Pi01 systemd[1]: mnt-Folder.automount: Got automount request for /mnt/Folder, triggered by 523 (bash)
Feb 24 15:34:40 Pi01 systemd[1]: Mounting /mnt/Folder...
Feb 24 15:34:41 Pi01 mount[798]: mount error(2): No such file or directory
Feb 24 15:34:41 Pi01 mount[798]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
Feb 24 15:34:41 Pi01 systemd[1]: mnt-Folder.mount: Mount process exited, code=exited, status=32/n/a
Feb 24 15:34:41 Pi01 systemd[1]: mnt-Folder.mount: Failed with result 'exit-code'.
Feb 24 15:34:41 Pi01 systemd[1]: Failed to mount /mnt/Folder.
Feb 24 15:34:41 Pi01 systemd[1]: mnt-Folder.automount: Got automount request for /mnt/Folder, triggered by 523 (bash)
Feb 24 15:34:41 Pi01 systemd[1]: Mounting /mnt/Folder...
Feb 24 15:34:41 Pi01 mount[803]: mount error(126): Required key not available
Feb 24 15:34:41 Pi01 mount[803]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
Feb 24 15:34:41 Pi01 systemd[1]: mnt-Folder.mount: Mount process exited, code=exited, status=32/n/a
Feb 24 15:34:41 Pi01 systemd[1]: mnt-Folder.mount: Failed with result 'exit-code'.
Feb 24 15:34:41 Pi01 systemd[1]: Failed to mount /mnt/Folder.
Feb 24 16:26:21 Pi01 systemd[1]: mnt-Folder.automount: Got automount request for /mnt/Folder, triggered by 1431 (mount.cifs)
Feb 24 16:26:21 Pi01 systemd[1]: Mounting /mnt/Folder...
Feb 24 16:26:21 Pi01 mount[1432]: mount error(2): No such file or directory
Feb 24 16:26:21 Pi01 mount[1432]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
Feb 24 16:26:21 Pi01 systemd[1]: mnt-Folder.mount: Mount process exited, code=exited, status=32/n/a
Feb 24 16:26:21 Pi01 systemd[1]: mnt-Folder.mount: Failed with result 'exit-code'.
Feb 24 16:26:21 Pi01 systemd[1]: Failed to mount /mnt/Folder.
```

MichaIng, 2021-02-24:
Okay, so indeed there is a `cp` and a `find` command during boot that trigger the automount, so adding the `ifup@.service` configuration might solve part of the issue. Probably you can identify the service/script that accesses that mount during boot and can delay it, to not have it facing an empty mount point unexpectedly. But the kernel-related one is to be investigated.

Loader23, 2021-02-25:
The cp and find command during boot is caused by commands inside /var/lib/dietpi/dietpi-autostart/custom.sh for the Kiosk System. Installing ifup@.service.d and rebooting did not fix the cp and find errors. But there is also a cron job for that so I can live with that.^^

MichaIng, 2021-02-26:
I made some tests and even with x-systemd.automount, any access to a network mount point, that triggers a mount request, is delayed until the network is online (the target reached). I even caused hanging test systems with this by adding a mount point access to our preboot script, which then was waiting forever until network, since network on the other hand waited for preboot to finish 😄.

However, let's try to completely rule out any network init related effect by disabling that service temporarily and see if a single manual access still fails to mount the share properly:
```
systemctl disable dietpi-autostart_custom
reboot
ls -l /mnt/Folder
df
journalctl -u mnt-Folder.mount -u mnt-Folder.automount
```

Loader23, 2021-02-26:
I have set up a fresh dietpi with no autostart and the mount does still fail. No matter if setup via manual fstab entry or via drive manager.

MichaIng, 2021-02-26:
Okay, and downgrading the kernel does as well not resolve it on that system?

Loader23, 2021-02-26:
No :-(

MichaIng, 2021-02-26:
You tried the other `sec=` options as well, right?

I had a look through the other upgraded packages an `libgnutls30` is used by the Samba library. Luckily the older package version is available in Debian snapshots. It won't be armv6hf but armv7hf, but in case or RPi 4 that shouldn't matter:
```
cd /tmp
wget 'https://snapshot.debian.org/archive/debian/20210101T024031Z/pool/main/g/gnutls28/libgnutls30_3.6.7-4%2Bdeb10u5_armhf.deb'
dpkg -i libgnutls30_*_armhf.deb
rm libgnutls30_*_armhf.deb
```
Everything else looks completely unrelated to me. `libcups2` is used by the library as well, but for printer servers. Of course one could go through the whole package list and downgrade everything from the snapshots archive, as something must have changed. I had another look through the patches of DietPi v6.34 but there is nothing even close to be related.

So I'm out of ideas then 😢.

Since we do not know whether it's RPi-related or not, probably best is to ask on a more general forum for help about this, like StackExchange superuser: https://superuser.com/
Probably someone which more Samba experience has further debugging ideas or knows more about possible issues in combination with DFS.

Joulinar, 2021-02-26:
Still would be interesting to see how plan Raspberry OS is behaving

Loader23, 2021-03-04:
I have setup Raspbian Lite, made update and upgrade and added the line to fstab. The mount line is unchanged except the uid and gid. After reboot the mount is available with no problems.
with dmesg I get this warning, but its working anyway.
```
[   38.314167] CIFS: VFS: Autodisabling the use of server inode numbers on new server
[   38.314177] CIFS: VFS: The server doesn't seem to support them properly or the files might be on different servers (DFS)
[   38.314185] CIFS: VFS: Hardlinks will not be recognized on this mount. Consider mounting with the "noserverino" option to silence this message.
```
I did not get this warning with dietpi though.

MichaIng, 2021-03-04:
Did you do an `apt full-upgrade` and `reboot` on Raspberry Pi OS, before testing it?

Loader23, 2021-03-04:
Hmm, no.^^ Just looked at the kernel, that was updated.^^

MichaIng, 2021-03-04:
Is done of those two packages installed?
```
dpkg -l | grep -E (keyutils|winbind)
```

MichaIng, 2021-03-04:
Ah the second one will be it: https://packages.debian.org/buster/winbind

Loader23, 2021-03-07:
Thats the solution! 😄
I have installed keyutils (not winbind) and its working! Still confusing why it was working with the old version and stopped working after Kernel Update. Iam going to include this package in all my DietPi scripts. Thanks alot to both of you 😃 👍

closed 2021-03-07 by Loader23

MichaIng, 2021-03-08:
Great! It's for kernel key management, so then it's likely due to the kernel upgrade, although strange that a downgrade did not work. We'll keep it in mind an in case create some hint in the docs.