MichaIng/DietPi #4140: dietpi-banner 0, dietpi-banner 1 or dietpi-banner 2 all open menu

maartenlangeveld, 2021-02-24:
DietPi v6.34.3

Seems that dietpi-banner does not process parameter right (or at all).

`dietpi-banner`
`dietpi-banner 0`
`dietpi-banner 1`
`dietpi-banner 2`

have same output, i.e. all result in opening the selection menu:

```
                    ┌──────────────────────────────────────────────────┤ DietPi-Banner ├───────────────────────────────────────────────────┐
                    │ Please (de)select options via spacebar to be shown in the DietPi-Banner:                                             │
                    │                                                                                                                      │
                    │    [*] 0   Device model                                                                                              │
                    │    [*] 1   Uptime                                                                                                    │
                    │    [*] 2   CPU temp                                                                                                  │
                    │    [*] 3   FQDN/hostname                                                                                             │
                    │    [ ] 4   NIS domainname                                                                                            │
                    │    [*] 5   LAN IP                                                                                                    │
                    │    [ ] 6   WAN IP                                                                                                    │
                    │    [*] 7   Freespace (RootFS)                                                                                        │
                    │    [ ] 8   Freespace (userdata)                                                                                      │
                    │    [ ] 9   Weather (wttr.in)                                                                                         │
                    │    [*] 10  Freespace (music)                                                                                         │
                    │    [*] 11  Display DietPi useful commands?                                                                           │
                    │    [*] 12  MOTD                                                                                                      │
                    │    [ ] 13  NordVPN status                                                                                            │
                    │                                                                                                                      │
                    │                                  <Ok>                                      <Cancel>                                  │
                    │                                                                                                                      │
                    └──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┘

```

Joulinar, 2021-02-24:
Hi,

this is a know issue and will be fixed on next release https://github.com/MichaIng/DietPi/blob/56b5f85cd1823f9f40a97f2fbcc51f2c618f714b/CHANGELOG.txt#L18

maartenlangeveld, 2021-02-24:
Ok thanks!

Joulinar, 2021-02-24:
I'm going to close this one. Feel free to reopen if needed.

closed 2021-02-24 by Joulinar

MichaIng, 2021-02-24:
Hehe, always great to get a report for an issue that has just been addressed, so I know that it was worth it aside of my own thoughts 😃. Yes, until now, the alias is `alias dietpi-banner='/boot/dietpi/dietpi-banner 2'` so that the menu call is hardcoded while other calls would require the full path, like `/boot/dietpi/dietpi-banner 0`, but now it is as it should be intuitively that the alias does not include the argument. To keep it aligned with other scripts, that calling the alias without argument opens the menu, the default was switched script-internally.