MichaIng/DietPi #4141: Python 3 won't install.

Andymc1280, 2021-02-25:
#### Details:
- Date           | Thu Feb 25 22:05:26 GMT 2021
- DietPi version | v7.0.0 (MichaIng/beta)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite 64-bit
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | Linux DietPihole 5.10.11-v8+ #1399 SMP PREEMPT Thu Jan 28 12:14:03 GMT 2021 aarch64 GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=0)
- Command        | mv get-pip.py
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. ... Install Python 3 from "Other software"
2. ...
#### Expected behaviour:

- ... It should install, but tries to move file somewhere it shouldn't.  Not the most helpful description BUT....
#### Actual behaviour:

- ... Install fails after trying to move file with error below.
#### Extra details:

- ...
#### Additional logs:
```
mv: cannot move 'get-pip.py' to '': No such file or directory
```

Joulinar, 2021-02-25:
Hi,

many thanks for your message. Usually it should have been fixed already https://github.com/MichaIng/DietPi/commit/bb226bc80cb5770fdf165d4bb66d57d6c977663a

Let me verify

MichaIng, 2021-02-26:
My fault, I implemented the logic false and obviously didn't test it. Fixed it now and tested all possible combinations file download file ending and target path given or not. Thanks for the report @Andymc1280!

Andymc1280, 2021-02-26:
So how do I get the fix ?

MichaIng, 2021-02-26:
To avoid switching to the dev branch:
```
wget 'https://github.com/MichaIng/DietPi/blob/dev/dietpi/dietpi-software' -O /boot/dietpi/dietpi-software
```

Andymc1280, 2021-02-26:
> 
> 
> To avoid switching to the dev branch completely:
> 
> ```
> wget 'https://github.com/MichaIng/DietPi/blob/dev/dietpi/dietpi-software' -O /boot/dietpi/dietpi-software
> ```

Connecting to github.com (github.com)|140.82.121.3|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/html]
Saving to: ‘/boot/dietpi/dietpi-software’

/boot/dietpi/dietpi-software             [   <=>                                                              ]   4.77M  10.5MB/s    in 0.5s

2021-02-26 19:12:45 (10.5 MB/s) - ‘/boot/dietpi/dietpi-software’ saved [5001778]

root@DietPihole:~# dietpi-software
/boot/dietpi/dietpi-software: line 7: syntax error near unexpected token `newline'
/boot/dietpi/dietpi-software: line 7: `<!DOCTYPE html>'

MichaIng, 2021-02-26:
Ah sorry, it needs to be the raw file URL of course:
```
wget 'https://raw.githubusercontent.com/MichaIng/DietPi/dev/dietpi/dietpi-software' -O /boot/dietpi/dietpi-software
```

closed 2021-02-28 by MichaIng