MichaIng/DietPi #4147: Image | NanoPi NEO2 Black hanging on first boot

chippey5, 2021-02-27:
### Creating a bug report/issue
I'm trying to install DietPi on my NanoPi NEO2 Black, but it doesn't appear on the network, so I grabbed one of my UART adapters and hooked it up to the board:

#### Required Information
- DietPi version | N/A
- Distro version | Buster
- Kernel version | N/A
- SBC model | NanoPi NEO2 Black
- Power supply used | 5v 2a
- SDcard used | Sandisk Ultra 16GB A1 & Samsung Evo 32GB 

First I tried with my Sandisk Ultra 16GB. SHA256 are passing, build from `Sat May 23 21:50:51 CEST 2020`. First boot:

```
U-Boot SPL 2019.10-armbian (Feb 17 2020 - 07:04:50 +0100)
DRAM: 1024 MiB
Trying to boot from MMC1
NOTICE:  BL31: v2.2(debug):572fcdd-dirty
NOTICE:  BL31: Built : 07:04:39, Feb 17 2020
NOTICE:  BL31: Detected Allwinner H5 SoC (1718)
NOTICE:  BL31: Found U-Boot DTB at 0x40923e8, model: FriendlyARM NanoPi NEO Core 2
INFO:    ARM GICv2 driver initialized
INFO:    Configuring SPC Controller
NOTICE:  PMIC: Assuming H5 reference regulator design
INFO:    BL31: Platform setup done
INFO:    BL31: Initializing runtime services
INFO:    BL31: cortex_a53: CPU workaround for 855873 was applied
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x4a000000
INFO:    SPSR = 0x3c9

U-Boot 2019.10-armbian (Feb 17 2020 - 07:04:50 +0100) Allwinner Technology

CPU:   Allwinner H5 (SUN50I)
Model: FriendlyARM NanoPi NEO Core 2
DRAM:  1 GiB
MMC:   mmc@1c0f000: 0, mmc@1c11000: 1
Loading Environment from EXT4... ** File not found /boot/boot.env **

** Unable to read "/boot/boot.env" from mmc0:1 **
In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.
Autoboot in 1 seconds, press <Space> to stop
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot/boot.scr
3033 bytes read in 3 ms (987.3 KiB/s)
## Executing script at 4fc00000
U-boot loaded from SD
Boot script loaded from mmc
219 bytes read in 2 ms (106.4 KiB/s)
28191 bytes read in 6 ms (4.5 MiB/s)
504 bytes read in 4 ms (123 KiB/s)
Applying kernel provided DT overlay sun50i-h5-usbhost1.dtbo
504 bytes read in 5 ms (97.7 KiB/s)
Applying kernel provided DT overlay sun50i-h5-usbhost2.dtbo
4161 bytes read in 5 ms (812.5 KiB/s)
Applying kernel provided DT fixup script (sun50i-h5-fixup.scr)
## Executing script at 44000000
7836994 bytes read in 382 ms (19.6 MiB/s)
15687688 bytes read in 766 ms (19.5 MiB/s)
## Loading init Ramdisk from Legacy Image at 4fe00000 ...
   Image Name:   uInitrd
   Image Type:   AArch64 Linux RAMDisk Image (gzip compressed)
   Data Size:    7836930 Bytes = 7.5 MiB
   Load Address: 00000000
   Entry Point:  00000000
[... 628 lines cut ...]
[  OK  ] Started Apply Kernel Variables.
[  OK  ] Started udev Coldplug all Devices.
         Starting Helper to synchronize boot up for ifupdown...
[  OK  ] Started Set the console keyboard layout.
[  OK  ] Started Helper to synchronize boot up for ifupdown.
[  OK  ] Started Create System Users.
         Starting Create Static Device Nodes in /dev...
[  OK  ] Started Create Static Device Nodes in /dev.
         Starting udev Kernel Device Manager...
[  OK  ] Reached target Local File Systems (Pre).
         Mounting /tmp...
[  OK  ] Mounted /tmp.
[  OK  ] Reached target Local File Systems.
         Starting Armbian ZRAM config...
         Starting Set console font and keymap...
         Starting Mark the need to relabel after reboot...
         Starting Raise network interfaces...
[  OK  ] Started Set console font and keymap.
[  OK  ] Started Mark the need to relabel after reboot.
[  OK  ] Started udev Kernel Device Manager.
[  OK  ] Found device /dev/ttyGS0.
[  OK  ] Found device /dev/ttyS1.
``` 

Before I flashed these images i had an old Armbian image on it that was working which used the same SD Card. Unfortunately I wiped the SD card and I don't have that image anymore because I wanted to use DietPi instead. I'm not sure what to do from this point. Any help would be greatly appreciated.

chippey5, 2021-02-27:
Issue also reported to the [armbian forums](https://forum.armbian.com/topic/17150-nanopi-neo2-black-fresh-install-fails-to-boot-properly/?tab=comments#comment-120103).

chippey5, 2021-02-27:
Update: The issue is most likely caused by insufficient amp throughput caused by the cable. Updating this issue once I have tried with a new cable.

chippey5, 2021-02-27:
Switched the cable and the board boots fine.

closed 2021-02-27 by chippey5

MichaIng, 2021-02-27:
Great. It always hanged on the partition resize, indeed possible that the implied disk writes triggered the voltage drop.

chippey5, 2021-02-28:
> Great. It always hanged on the partition resize, indeed possible that the implied disk writes triggered the voltage drop.

Indeed. There was a false positive before I reflashed the board - It was working with the same cable but everything was extremely sluggish, hence I wanted to reflash the image as that installation was more than a year old (and because I wanted to use DietPi over Armbian). So given the fact that the board successfully booted before reflashing I never thought it'd be the cable.

Always fun to tackle new problems, thanks for your effort in the dietpi forums as always @MichaIng