MichaIng/DietPi #4149: DietPi-Software | WiringPi: Migrate to new maintained branch

MichaIng, 2021-02-27:
I'm sorry to read such a frustrating deprecation message of the founder of WiringPi: http://wiringpi.com/wiringpi-deprecated/
But there is a new repository which further maintains the code: https://github.com/WiringPi/WiringPi

For RPi, we ship a fixed commit that seems to be the second last official release before Gordon stopped developing it: https://github.com/WiringPi/WiringPi/commit/093e0a17a40e064260c1f3233b1ccdf7e4c66690
A final release v2.50 has been made and now the further maintained repository. Which we should use.

I'll have a closer look into this with next development cycle.
_______
This is the RPi version, not supporting other SBCs, while other manufacturers have their own repos:
- Odroid: https://github.com/hardkernel/wiringPi
- NanoPi: https://github.com/friendlyarm/WiringNP
- BananaPi: https://github.com/LeMaker/WiringBP

MichaIng, 2021-05-01:
Support for some NanoPi models added via FriendlyARM's fork "WiringNP": #4326

Please help testing it on all those models:
- [ ] NanoPi NEO
- [ ] NanoPi NEO Air
- [ ] NanoPi NEO2
- [ ] NanoPi NEO Plus2
- [ ] NanoPi M1
- [ ] NanoPi K1 Plus

MichaIng, 2021-05-06:
@StephanStS 
Pinging you here for testing.

StephanStS, 2021-05-07:
@MichaIng: 
Does this also cover [The access to smbus, i2c and gpio is missing on NanoPi NEO2](https://github.com/MichaIng/DietPi/issues/3746)?

It is also related to [DietPi-Software | WiringNP](https://github.com/MichaIng/DietPi/issues/3514)?

MichaIng, 2021-05-07:
How could I forget the second issue you link 😄. So indeed the issue as that the device name/identifier is reported differently by the Armbian kernel (in case of NanoPi NEO(1)). Probably with M1 Plus it's the same.

I'm not good enough in C code reading to understand where those identifiers are read from, respectively where we can find the correct one on the Armbian system, e.g. via /sys or /proc contained files, like `/proc/cpuinfo` or others. Here it a list which associates such info, but I'm not exactly sure which of the array entries refers to the kernel/firmware reported identifier and which to the name that is shown/reported by WiringPi/gpio lib/util: https://github.com/friendlyarm/WiringNP/blob/master/wiringPi/boardtype_friendlyelec.c
The variables seem to be addresses, defined in the header: https://github.com/friendlyarm/WiringNP/blob/master/wiringPi/boardtype_friendlyelec.h

Hmm, shall we switch over to the other issue and keep this one as RPi sources update finished?

The first link btw is not directly but indirectly related. WiringPi also allows to manage I2C, SPI, UART etc. But this can only work when the related device tree overlays are enabled. So if you face any errors regarding those, at least we should keep it in mind. Not sure whether e.g. the build or the gpio util tries to access those without explicitly calling the related command/function.

StephanStS, 2021-05-09:
Played a bit with debugging the .c file. 
Ended up in this thread which is still valid: <https://forum.armbian.com/topic/5647-nanopi-neo-wiringnp-missing-sunxi_infosys_info>.
I stopped there instead of playing with the given workaround via /etc/sys_info.
If needed the repo <https://github.com/friendlyarm/WiringNP> and the kernel need to be extended.

MichaIng, 2021-05-09:
Great find!

Interesting that it worked on NEO2 without the change. But considering the bad code quality as well, shall we give up on this for all NanoPi's? Otherwise I think we'd need to fork the FriendlyARM repo to add mainline kernel support. We could also fix some build warnings then. I'm just not sure if it's worth the effort, i.e. how many NanoPi users would really use WiringPi.

At least we now know which file is actually used to derive the model: `/sys/class/sunxi_info/sys_info`
Does this even exist on our NanoPi images?

StephanStS, 2021-05-09:
This file /sys/class/sun*) I cannot see on any NanoPi image. But the fopen() command on this seems to run.
The first basic hardware info is read from /proc/cpuinfo (e.g. `Hardware: sun50iw1p1` in case of a NEO2).

Further details shall be read via fopen("/sys/class/sunxi_info/sys_info", "r") which leads to the problem. I do not know how to add kernel code to the friendlyelec code base to get it running.

StephanStS, 2021-05-10:
Checked as **"Not working"** for 
- NanoPi NEO4
- NanoPi M4v2
- NanoPi K2

MichaIng, 2021-05-10:
Thanks for testing. But see the list above, they were not support to work 😉: https://github.com/MichaIng/DietPi/issues/4149#issuecomment-830688008

MichaIng, 2021-05-10:
Got it: `/proc/cpuinfo` only provides the SoC/CPU ID but not the board ID. In case of Allwinner H3 and H5, many different SBCs share the same SoC but may have a different GPIO layout. In those cases another board ID is provided via `/sys/class/sunxi_info/sys_info`. Not sure whether the kernel itself on Armbian images provides such information elsewhere, but indeed it looks like we'd need to use a file like /boot/dietpi/.hw_model to take this info from, like discussed in the other issue as well.

Interestingly the newer RK3399 SBCs are as well in the code already, so indeed the test was worth it: https://github.com/friendlyarm/WiringNP/blob/master/wiringPi/boardtype_friendlyelec.c#L81-L88

MichaIng, 2021-05-18:
Marking this as closed. Let's discuss WiringNP in the other issue, but I'll remove it for v7.2 as it requires larger rework and testing and code fixes as well: #3514

closed 2021-05-18 by MichaIng