MichaIng/DietPi #4151: mpeg-streamer is unable to install because of missing blank in dependend apt-get install

mhansi12, 2021-02-27:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=0
G_DIETPI_VERSION_RC=1
G_GITBRANCH='beta'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
      10.8
- Kernel version | `uname -a`
      Linux DietPi 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
      RPi 4 Model B (armv7l)
- Power supply used | (EG: 5V 1A RAVpower)
   2,5A
- SDcard used | (EG: SanDisk ultra)
  SanDisk 32 GB
#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
  mpeg-streamer
- Was the software title installed freshly or updated/migrated?
   fresh install, because it was not available in 6.34
- Can this issue be replicated on a fresh installation of DietPi?
   I think so, I have upgraded from 6.34.5, but problem is probebly the same

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

Install 6.34.5 ( maybe new install is also ok )
Update to 7.0.0
install "mpeg-streamer"  with "dietpi-software" --> "Software Optimised" --> mpeg-streamer
Problem: unable to install dependend libxx-dev files, because of missing blank (" ")
Line shows:
 `apt-get -qq --allow-change-held-packages install make cmake gcc libc6-dev **libjpeg-devlibraspberrypi-dev**`

#### Expected behaviour
 `apt-get -qq --allow-change-held-packages install make cmake gcc libc6-dev libjpeg-dev libraspberrypi-dev`

#### Actual behaviour

  apt-get install error

#### Extra details
  with blank between "libjpeg-devlibraspberrypi-dev"  everything installs fine
  Just updated to 7.0.1 , so it is not clear to me if this problem still exists ( I do not see this in changelog), so I made this report

MichaIng, 2021-02-27:
Many thanks for reporting, I just fixed it: https://github.com/MichaIng/DietPi/commit/b2959fa29fa56f56793313f1387386216730d060

mhansi12, 2021-02-28:
Thank your for your fast response - I fix the name in my report and subject- other people should find ... :-)

closed 2021-02-28 by MichaIng