MichaIng/DietPi #4153: PoE Hat (official) fan not starting in 5.10.11 kernel

dazzathewiz, 2021-02-28:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=0
G_DIETPI_VERSION_RC=1
G_GITBRANCH='beta'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
buster

- Kernel version | `uname -a`
Linux piFour 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
RPi 4 Model B (armv7l)

- Power supply used | (EG: 5V 1A RAVpower)
PoE Hat (official RPi)

- SDcard used | (EG: SanDisk ultra)
SanDisk Ultra 16GB

#### Additional Information (if applicable)

- Since upgrade to DietPi v6.34.3 the PoE Fan no longer works. My previous version was v6.33.3 before directly upgrading to v6.34.3. The non operational fan affects the Pi4 temperature by adding ~27c which overheats my Pi.
- dmesg | grep fan
[    3.710175] rpi-poe-fan rpi-poe-fan@0: Failed to get default PWM value: -5
[    3.710235] rpi-poe-fan: probe of rpi-poe-fan@0 failed with error -5
- This post notes it's resolved in latest firmware: https://www.raspberrypi.org/forums/viewtopic.php?t=298985
- However, I cannot find a way to get to 5.10.14 kernel with DietPi, including performing:
sudo apt-get install --reinstall raspberrypi-bootloader raspberrypi-kernel
- I have confirmed not a hardware issue by downloading latest image and on boot the fan works, but as soon as apt update/upgrade completes and reboots, the fan stops working again.

#### Steps to reproduce
1. Download fresh image (v6.34.3)
2. On setup, the fan works first boot
3. After apt update / upgrade and reboot, the fan stops working
4. Upgrade to beta 7.0.1, behaviour remains the same (no newer kernel)

#### Expected behaviour

- PoE fan starts on boot
- get to kernel: Linux raspberrypi 5.10.14-v7l+ #1401 SMP Mon Feb 8 14:27:07 GMT 2021 armv7l GNU/Linux

#### Actual behaviour

- PoE fan briefly moves on boot but then stops working

#### Extra details
- https://www.raspberrypi.org/forums/viewtopic.php?t=298985
- https://github.com/home-assistant/operating-system/issues/1208

Joulinar, 2021-02-28:
Hi,

Many thanks for your message. DietPi is not creating own kernel. As you running a Raspberry Pi, you are using Raspberry OS as base image. Means you would need official Raspberry Foundation tools like `rpi-update` to update or downgrade to a specific kernel version. https://github.com/Hexxeh/rpi-update#options

dazzathewiz, 2021-02-28:
Hi there - thank you this helps me a lot. I figured the kernel was part of the Raspberry OS base, but wasn't sure and just assumed at some point the newer kernels are included in DietPi.

Thanks for your link, installing and running rpi-update has put my pi onto 5.10.17 kernel and the problem has gone away!
uname -a
Linux piFour 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux

Thanks for your patients - I'm grateful that I got an answer which fixed it for me :)

Joulinar, 2021-02-28:
yes, the kernel is part of Raspberry OS and not influenced or changed by DietPi. Keep in mind, `rpi-update` moved your kernel away from stable into testing state. 😉 

I will it mark this as solved for now.

closed 2021-02-28 by Joulinar