MichaIng/DietPi #4154: Odroid C2 LCD 3.5 not working with Linux 3.16.85+

MarcProux, 2021-02-28:
### Creating a bug report/issue

#### Required Information
- DietPi version | `6.34.3`
- Distro version | `buster`
- Kernel version | `Linux ODROID-CL01-SBC-M1 3.16.85+ #1 SMP PREEMPT Tue Jun 30 19:02:35 CEST 2020 aarch64 GNU/Linux
`
- SBC model | `Odroid C2 (aarch64)`
- Power supply used | `5V 2A Hardkernel`
- eMMC used | `C2 Red 16Go`

#### Additional Information (if applicable)
- Can this issue be replicated on a fresh installation of DietPi? `yes`

#### Steps to reproduce

1. Instrall 'chromium' as autostart and configure dietpi as user.
2. Right after installation activate the `odroid-lcd35` in `dietpi-config` under Display Options

#### Expected behaviour

- Get the chromium desktop

#### Actual behaviour

- Backlight is on. And nothing more.

#### Extra details
```shell
root@ODROID-CL01-SBC-M1:~# systemctl status odroid-lcd35
● odroid-lcd35.service - Odroid LCD 3.5
   Loaded: loaded (/etc/systemd/system/odroid-lcd35.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2021-02-28 13:08:01 CET; 20min ago
  Process: 343 ExecStart=/bin/dash -c echo 500000 > /sys/devices/pwm-ctrl/freq0 (code=exited, status=0/SUCCESS)
  Process: 358 ExecStart=/bin/dash -c echo 1 > /sys/devices/pwm-ctrl/enable0 (code=exited, status=0/SUCCESS)
  Process: 364 ExecStart=/bin/dash -c echo 1023 > /sys/devices/pwm-ctrl/duty0 (code=exited, status=0/SUCCESS)
  Process: 379 ExecStart=/usr/bin/setterm -blank 0 -powersave off (code=exited, status=0/SUCCESS)
  Process: 416 ExecStart=/bin/con2fbmap 1 2 (code=exited, status=1/FAILURE)
 Main PID: 416 (code=exited, status=1/FAILURE)

Feb 28 13:08:01 ODROID-CL01-SBC-M1 systemd[1]: Starting Odroid LCD 3.5...
Feb 28 13:08:01 ODROID-CL01-SBC-M1 systemd[1]: odroid-lcd35.service: Main process exited, code=exited, status=1/FAILURE
Feb 28 13:08:01 ODROID-CL01-SBC-M1 systemd[1]: odroid-lcd35.service: Failed with result 'exit-code'.
Feb 28 13:08:01 ODROID-CL01-SBC-M1 systemd[1]: Failed to start Odroid LCD 3.5.
```
```shell
root@ODROID-CL01-SBC-M1:~# cat /proc/fb
0 OSD FB
1 OSD FB
```
```shell
root@ODROID-CL01-SBC-M1:~# cat /etc/modules
# /etc/modules: kernel modules to load at boot time.
#
# This file contains the names of kernel modules that should be loaded
# at boot time, one per line. Lines beginning with "#" are ignored.

fbtft_device
flexfb
sx865x
aml_i2c
pwm-meson
pwm-ctrl
```
```shell
root@ODROID-CL01-SBC-M1:~# cat /etc/modprobe.d/odroid-lcd35.conf 
options fbtft_device name=flexpfb rotate=270 debug=7
options flexfb chip=ili9488
```
```shell
root@ODROID-CL01-SBC-M1:~# lsmod
Module                  Size  Used by
w1_gpio                 3801  0
wire                   28753  1 w1_gpio
meson_gpiomem           4064  0
gxbb_wdt                3411  0
pwm_ctrl                6619  0
pwm_meson               4530  1
aml_i2c                16850  0
sx865x                  5235  0
flexfb                 16696  0
fbtft_device           36079  0
fbtft                  37559  2 flexfb,fbtft_device
syscopyarea             2905  1 fbtft
sysfillrect             3731  1 fbtft
sysimgblt               1933  1 fbtft
fb_sys_fops             1365  1 fbtft
ip_tables              17301  0
ipv6                  386185  30
autofs4                27654  2
spidev                  6643  0
spi_gpio                6551  0
spi_bitbang             4071  1 spi_gpio
```
So module is registered as succesfully loaded.

```shell
root@ODROID-CL01-SBC-M1:~# dmesg | grep fbtft
[    6.086947] fbtft_device:  SPI devices registered:
[    6.089133] fbtft_device:      spidev spi0.0 500kHz 8 bits mode=0x00
[    6.094567] fbtft_device:  'fb' Platform devices registered:
[    6.100173] fbtft_device:      meson-fb id=-1 pdata? no
[    6.105463] fbtft_device:  GPIOS used by 'flexpfb':
[    6.110010] fbtft_device:    'reset' = GPIO247
[    6.114447] fbtft_device:    'dc' = GPIO228
[    6.118384] fbtft_device:    'wr' = GPIO229
[    6.122427] fbtft_device:    'cs' = GPIO238
[    6.126481] fbtft_device:    'db00' = GPIO233
[    6.130709] fbtft_device:    'db01' = GPIO231
[    6.134942] fbtft_device:    'db02' = GPIO236
[    6.139125] fbtft_device:    'db03' = GPIO230
[    6.143346] fbtft_device:    'db04' = GPIO232
[    6.147558] fbtft_device:    'db05' = GPIO235
[    6.151801] fbtft_device:    'db06' = GPIO237
[    6.156097] fbtft_device:    'db07' = GPIO239
[    6.160346] fbtft_device:  'fb' Platform devices registered:
[    6.166005] fbtft_device:      meson-fb id=-1 pdata? no
[    6.171189] fbtft_device:      flexpfb id=0 pdata? yes
[    6.183905] flexfb flexpfb.0: fbtft_request_gpios_match('reset')
[    6.188822] flexfb flexpfb.0: fbtft_request_gpios: 'reset' = GPIO247
[    6.195042] flexfb flexpfb.0: fbtft_request_gpios_match('dc')
[    6.200723] flexfb flexpfb.0: fbtft_request_gpios: 'dc' = GPIO228
[    6.206710] flexfb flexpfb.0: fbtft_request_gpios_match('wr')
[    6.212361] flexfb flexpfb.0: fbtft_request_gpios: gpio_request_one('wr'=229) failed with -16
```

MichaIng, 2021-02-28:
Many thanks for your report.

`con2fbmap` failed. Can you run the following from the console, e.g. via SSH and paste the output here?
```
con2fbmap 1 2
```
Based on your output, `/dev/fb2` seems to simply not exist.

When `/dev/fb0` is HDMI, I'm wondering what `/dev/fb1` is, when `/dev/fb2` is assumed to be the LCD. Probably this has changed with a recent kernel upgrade.

Does this work?
```
con2fbmap 1 1
```

I checked where we got the instructions from: https://wiki.odroid.com/accessory/display/3.5inch_lcd_shield/3.5inch_lcd_shield
> The frame buffer 0, 1 (/dev/fb{0,1}) always exists. Proceed with /dev/fb2 or other named flexfb.

Yours are called `OSD FB`, not `flexfb`.

```
fbtft_request_gpios: gpio_request_one('wr'=229) failed with -16
```
This means that something else is using the GPIO already, but not sure if it is related here.
... ah, the XU3/XU3 box reveals some hint: https://wiki.odroid.com/accessory/display/3.5inch_lcd_shield/3.5inch_lcd_shield#tab__odroid-xu3xu4
> If the device /dev/fb* is not created and the following messages appears from dmesg,
> ```
> [  621.952761] flexfb_probe_common : ioremap gpiox register success!
> [  621.957496] flexfb flexpfb.0: fbtft_request_gpios: gpio_request_one('wr'=190) failed with -16
> ```
> Enter the following command to unload SPI modules and try again.
> ```
> sudo modprobe -r spidev
> ```

So can you please try the following:
```
modprobe -r spidev
modprobe fbtft_device name=flexpfb rotate=90
modprobe flexfb chip=ili9488
con2fbmap 1 2
```

MarcProux, 2021-02-28:
> con2fbmap failed. Can you run the following from the console, e.g. via SSH and paste the output here?
> `con2fbmap 1 2`
```shell
root@ODROID-CL01-SBC-M1:~# con2fbmap 1 2
ioctl FBIOPUT_CON2FBMAP: Invalid argument
```

> Does this work?
> `con2fbmap 1 1`

This work and map the console 1 to the framebuffer 1:
```shell
root@ODROID-CL01-SBC-M1:~# con2fbmap 1
console 1 is mapped to framebuffer 1
```

After this, I tried to unload the SPI but does nothing more. Framebuffer 2 still missing:
```shell
root@ODROID-CL01-SBC-M1:~# modprobe -r spidev
root@ODROID-CL01-SBC-M1:~# modprobe fbtft_device name=flexpfb rotate=90
root@ODROID-CL01-SBC-M1:~# modprobe flexfb chip=ili9488
root@ODROID-CL01-SBC-M1:~# lsmod
Module                  Size  Used by
w1_gpio                 3801  0
wire                   28753  1 w1_gpio
meson_gpiomem           4064  0
gxbb_wdt                3411  0
pwm_ctrl                6619  0
pwm_meson               4530  1
aml_i2c                16850  0
sx865x                  5235  0
flexfb                 16696  0
fbtft_device           36079  0
fbtft                  37559  2 flexfb,fbtft_device
syscopyarea             2905  1 fbtft
sysfillrect             3731  1 fbtft
sysimgblt               1933  1 fbtft
fb_sys_fops             1365  1 fbtft
ip_tables              17301  0
ipv6                  386185  30
autofs4                27654  2
spi_gpio                6551  0
spi_bitbang             4071  1 spi_gpio
root@ODROID-CL01-SBC-M1:~# con2fbmap 1 2
ioctl FBIOPUT_CON2FBMAP: Invalid argument
root@ODROID-CL01-SBC-M1:~# cat /proc/fb
0 OSD FB
1 OSD FB
```

MichaIng, 2021-02-28:
Hmm there are two other SPI modules, lets try to remove them as well:
```
modprobe -r spi_gpio
modprobe -r spi_bitbang
modprobe fbtft_device name=flexpfb rotate=90
modprobe flexfb chip=ili9488
con2fbmap 1 2
```
And if that does not work yet, could you try to reboot with them blacklisted and then check again the related kernel messages:
```
echo -e 'blacklist spidev\nblacklist spi_gpio\nblacklist spi_bitbang' > /etc/modprobe.d/disable-spi.conf
reboot
dmesg | grep flexfb
```

MarcProux, 2021-02-28:
After many try, I found a solution for loading the LCD.
```shell
echo -e 'blacklist spidev\nblacklist spi_gpio\nblacklist spi_bitbang' > /etc/modprobe.d/blacklist.conf
update-initramfs -u
```

By doing this way, the default blacklist is updated.

I don't think this is the right way and most portable but it works.

MichaIng, 2021-02-28:
Ah I forgot the initramfs, great that it works by updating it. It doesn't matter if those blacklists are added to `blacklist.conf` or any other file name inside there, to have them removed from the initramfs, right? We can add this to our config scripts, or even batter a general SPI interface toggle which is then switched off when enabling any non-SPI LCD on Odroid C2.

MichaIng, 2021-12-08:
Solved with: https://github.com/MichaIng/DietPi/commit/4e49b26

closed 2021-12-11 by MichaIng