MichaIng/DietPi #4155: Write /boot/dietpi/.hw_model only if it has changed

ferbar, 2021-02-28:
### Creating a bug report/issue
If you configured /boot as readonly device /boot/dietpi/preboot will print an error message on boot.

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=34
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
bustercat /etc/debian_version

- Kernel version | `uname -a`
Linux raspi-rhb-182 5.10.11+ #1399 Thu Jan 28 12:02:28 GMT 2021 armv6l GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
RPi Zero W (armv6l)

#### Additional Information (if applicable)

#### Steps to reproduce
- mount /boot as readonly (fstab od dietpi-device_manager)
- reboot
- /boot/dietpi/preboot
Error Message:
/boot/dietpi/func/dietpi-obtain_hw_model: line 614: /boot/dietpi/.hw_model: Read-only file system
/boot/dietpi/func/dietpi-obtain_hw_model: line 622: /boot/dietpi/.hw_model: Read-only file system

#### Expected behaviour
a) /boot/dietpi/.hw_model is not written if /boot is mounted readonly or
b) /boot/dietpi/.hw_model is written only if it's different from the current .hw_model

#### Actual behaviour
/boot/dietpi/.hw_model is written on every boot

#### Extra details
I could write a PR with a check to only write .hw_model if the information has changed.

MichaIng, 2021-02-28:
Many thanks for your request. I was thinking to apply this only on RPi, where it's possible to swap the SD card to a different model. On all other SBCs it isn't possible anyway, so the information should only chance in case on DietPi updates, where we call the scripts once anyway.

... ah, after a distribution update, info changes as well. That was the reason I left things as they are now.

Writing that file is probably a faster task than comparing it with what would be written, and it does not solve the actual issue that writing can be attempted even that the drive is read-only. Since scripts currently depend on the correctness of the information, it can be seen as required that /boot is mounted R/W, which is as well true for a lot of other scripts. What we need is a read-only mode, in which case all scripts handle the situation gracefully, print meaningful error messages etc.

What we could do here, is to check for R/W mount before even calling `dietpi-obtain_hw_model`. But to decide which workaround works best for you, let me understand first the motivation/issue you have with the current situation:
- `dietpi-preboot.service` still succeeds, despite the error, doesn't it?
- So it would be a purely visual annoyance?
- If we check for R/W mount, currently I'd still print an error message or warning, until we have a definite read-only mode. So the situation for you would not change that much, but probably a little less overhead only.
- A very different approach would be, in case the drive is R/O mounted, to temporarily mount it R/W and R/O back again. Not sure if it's worth it, and probably not even wanted when someone already chooses to mount it R/O for some reason.

ferbar, 2021-02-28:
Thanks for the insights!
The motivation is a raspberry pi zero batterypowered. My experience is that a rw - mounted filesystem (expecially fat) breaks if the battery gets empty. The easiest way to fix that is ow mounting the filesystems.

> `dietpi-preboot.service` still succeeds, despite the error, doesn't it?
right
> So it would be a purely visual annoyance?
right (could not read all the other error messages ;-) )
> If we check for R/W mount, currently I'd still print an error message or warning,
right
> A very different approach would be, in case the drive is R/O mounted, to temporarily mount it R/W and R/O back again.
If battery power is low during boot that would destroy the filesystem and nobody would find that => better not
> Writing that file is probably a faster task than comparing it [snip]
/boot/dietpi/.hw_model is already read in
https://github.com/MichaIng/DietPi/blob/master/dietpi/func/dietpi-obtain_hw_model#L110
so it should already be in the filesystem cache and not cause additional delays. 

Readonly mounting / and /boot takes some time to change some config files and add some softlinks but isn't too hard for a headless installation.

Thanks for your great work!

Joulinar, 2021-02-28:
In past I was testing with overlay file system `overlayroot`. It was working quite well and keeps all services running. Especially for services who require r/w file system.

MichaIng, 2021-02-28:
As long as nothing is actually written to the drive, the fact that it's R/W mounted only, should not cause file system corruption on power loss. But I understand the concerns then, and a R/W mount of course theoretically enables writes.

> /boot/dietpi/.hw_model is already read in
https://github.com/MichaIng/DietPi/blob/master/dietpi/func/dietpi-obtain_hw_model#L110
so it should already be in the filesystem cache and not cause additional delays.

It's in cache due to `vmtouch` running much earlier at boot 😉. But re-reading that cached file and comparing the string to the output string is not that efficiently done in shells. So I think skipping the whole script is likely the best solution. Does `[[ -w '/boot/dietpi/.hw_model' ]]` respects permissions only or the mount type as well 🤔? That would be pretty easy and fast. Else checking the mount options would be a little more overhead.

Since it's visual only, and there are other error messages and other actual errors expected, I have to think through it whether we start implementing such checks here and there, or better add a full read-only "system", i.e. a flag derived at boot that is read by DietPi-Globals and used by all scripts to gracefully handle R/O systems, limiting features or exiting with meaningful error messages. That could be even beneficial to quickly detect file system corruption which often causes a read-only remount of the affected file system.

Again a different idea: Since we write `.hw_model` on every boot, it can be written to a tmpfs, like `/run/dietpi/.hw_model` model. Only our UUID needs to stay at disk. But that one must never change anyway, so it can stay inside of an individual R/O file.

Generally the overlayfs method indeed is great for R/O systems, since it's not our scripts only which do not gracefully deal with it. I can only recommend that approach, which also enables a lot more features, even that writes are not boot-persistent.