MichaIng/DietPi #4156: DietPi installs Microsoft vscode repository and Microsoft GPG key

kralqt, 2021-02-28:
#### Required Information

Detail | Value
------------ | -------------
DietPi version |  G_DIETPI_VERSION_CORE=6<br />G_DIETPI_VERSION_SUB=34<br />G_DIETPI_VERSION_RC=3<br />G_GITBRANCH='master'<br />G_GITOWNER='MichaIng'
Distro version  | 10.8
Kernel version  | Linux DietPi 5.10.5-v8+ #1392 SMP PREEMPT Sat Jan 9 18:56:30 GMT 2021 aarch64 GNU/Linux
SBC model  | RPi 4 Model B (aarch64) with 4 GB RAM
Power supply used | EG: 5V 2.4A Samsung
SDcard used | Lexar 32GB A1

#### Additional Information
- Fresh install of DietPi

#### Steps to reproduce
1. Download the DietPi image, put it on SD Card and boot it on RPi 4
2. Go through the installation process selecting OpenSSH as SSH server
3. Connect to the DietPi and do apt-get update:
 > dietpi@DietPi:~$ sudo apt-get update
Hit:1 http://packages.microsoft.com/repos/code stable InRelease
Hit:2 https://deb.debian.org/debian buster InRelease
Hit:3 https://deb.debian.org/debian buster-updates InRelease
Hit:4 https://archive.raspberrypi.org/debian buster InRelease
Hit:5 https://deb.debian.org/debian-security buster/updates InRelease
Hit:6 https://deb.debian.org/debian buster-backports InRelease
Reading package lists... Done

Also:
> dietpi@DietPi:~$ ls -1 /etc/apt/trusted.gpg.d/
debian-archive-buster-automatic.gpg
debian-archive-buster-security-automatic.gpg
debian-archive-buster-stable.gpg
debian-archive-jessie-automatic.gpg
debian-archive-jessie-security-automatic.gpg
debian-archive-jessie-stable.gpg
debian-archive-stretch-automatic.gpg
debian-archive-stretch-security-automatic.gpg
debian-archive-stretch-stable.gpg
microsoft.gpg

#### Expected behaviour
- The DietPi advertises itself as "Highly optimised minimal Debian OS". It shouldn't add Microsoft repos/keys.

#### Actual behaviour
- On every `sudo apt-get update` we hit http://packages.microsoft.com/repos/code and the Microsoft GPG key is installed as trusted
- The controversial move from RPi Foundation to silently add Microsoft vscode repo and install Microsoft GPG is replicated in DietPi. I certainly didn't expect it to to do this on my headless RPi setup!

#### Extra details
See https://www.windowscentral.com/microsoft-repo-silently-added-raspberry-pi-os-folks-begin-freak-out
> ... And as one particularly vocal commenter pointed out, modifying the sources.list in Linux without consent just doesn't happen. It also doesn't just apply to new images, it has been built out to be added to existing machines, too. Even Lite images of Raspberry Pi built for headless deployment. They all have the Microsoft repo now.

See the comments in the announcement: https://www.raspberrypi.org/blog/visual-studio-code-comes-to-raspberry-pi/

Also see https://www.reddit.com/r/linux/comments/lbu0t1/microsoft_repo_installed_on_all_raspberry_pis/ with 1K comments showing how hot topic this is

#### How to manually remove the Microsoft repo
```
sudo rm /etc/apt/sources.list.d/vscode.list
sudo rm /etc/apt/trusted.gpg.d/microsoft.gpg
sudo apt update
```
This is just a workaround and the issue needs to be addessed in the DietPi process for building the RPi image

If you want to disable further such modifications:
```
sudo apt-mark hold raspberrypi-sys-mods
```

Joulinar, 2021-02-28:
Hi,

many thanks for your message. This is know and we will handle it on next release. This has been introduced by RPi Foundation and not by DietPi. Some background can be found on this post https://github.com/MichaIng/DietPi/issues/4083

kralqt, 2021-02-28:
Closing as a duplicate pf #4083

closed 2021-02-28 by kralqt