MichaIng/DietPi #4159: Unbound error during installation

redschumi, 2021-03-01:
#### Details:
- Date           | Mon Mar  1 13:17:25 CET 2021
- Bug report     | 5f6c09cf-a0a5-42e6-bbe2-475daaed2825
- DietPi version | v7.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | Linux dietpi 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | systemctl restart unbound
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Warning: The unit file, source configuration file or drop-ins of unbound.service changed on disk. Run 'systemctl daemon-reload' to reload units.
Job for unbound.service failed because the control process exited with error code.
See "systemctl status unbound.service" and "journalctl -xe" for details.
```

MichaIng, 2021-03-01:
Many thanks for your report.

Did you retry the same command from the error prompt? Otherwise please paste the output of:
```
journalctl -u unbound
```

redschumi, 2021-03-01:
root@dietpi:~# journalctl -u unbound
-- Logs begin at Mon 2021-03-01 13:07:53 CET, end at Mon 2021-03-01 16:10:01 CET. --
Mär 01 13:08:04 dietpi systemd[1]: Starting Unbound DNS server...
Mär 01 13:08:18 dietpi package-helper[521]: /var/lib/unbound/root.key has content
Mär 01 13:08:18 dietpi package-helper[521]: success: the anchor is ok
Mär 01 13:08:19 dietpi unbound[541]: [1614600499] unbound[541:0] error: cannot parse ip address: '.0/24'
Mär 01 13:08:19 dietpi unbound[541]: [1614600499] unbound[541:0] error: cannot parse access control: .0/24 allow
Mär 01 13:08:19 dietpi unbound[541]: [1614600499] unbound[541:0] fatal error: Could not setup access control list
Mär 01 13:08:19 dietpi systemd[1]: unbound.service: Main process exited, code=exited, status=1/FAILURE
Mär 01 13:08:19 dietpi systemd[1]: unbound.service: Failed with result 'exit-code'.
Mär 01 13:08:19 dietpi systemd[1]: Failed to start Unbound DNS server.
Mär 01 13:08:19 dietpi systemd[1]: unbound.service: Service RestartSec=100ms expired, scheduling restart.
Mär 01 13:08:19 dietpi systemd[1]: unbound.service: Scheduled restart job, restart counter is at 1.
Mär 01 13:08:19 dietpi systemd[1]: Stopped Unbound DNS server.
Mär 01 13:08:19 dietpi systemd[1]: Starting Unbound DNS server...
Mär 01 13:08:19 dietpi package-helper[593]: /var/lib/unbound/root.key has content
Mär 01 13:08:19 dietpi package-helper[593]: success: the anchor is ok
Mär 01 13:08:19 dietpi unbound[607]: [1614600499] unbound[607:0] error: cannot parse ip address: '.0/24'
Mär 01 13:08:19 dietpi unbound[607]: [1614600499] unbound[607:0] error: cannot parse access control: .0/24 allow
Mär 01 13:08:19 dietpi unbound[607]: [1614600499] unbound[607:0] fatal error: Could not setup access control list
Mär 01 13:08:19 dietpi systemd[1]: unbound.service: Main process exited, code=exited, status=1/FAILURE
Mär 01 13:08:19 dietpi systemd[1]: unbound.service: Failed with result 'exit-code'.
Mär 01 13:08:19 dietpi systemd[1]: Failed to start Unbound DNS server.
Mär 01 13:08:19 dietpi systemd[1]: unbound.service: Service RestartSec=100ms expired, scheduling restart.
Mär 01 13:08:19 dietpi systemd[1]: unbound.service: Scheduled restart job, restart counter is at 2.
Mär 01 13:08:19 dietpi systemd[1]: Stopped Unbound DNS server.
Mär 01 13:08:19 dietpi systemd[1]: Starting Unbound DNS server...
Mär 01 13:08:19 dietpi package-helper[618]: /var/lib/unbound/root.key has content
Mär 01 13:08:19 dietpi package-helper[618]: success: the anchor is ok
Mär 01 13:08:20 dietpi unbound[638]: [1614600500] unbound[638:0] error: cannot parse ip address: '.0/24'
Mär 01 13:08:20 dietpi unbound[638]: [1614600500] unbound[638:0] error: cannot parse access control: .0/24 allow
Mär 01 13:08:20 dietpi unbound[638]: [1614600500] unbound[638:0] fatal error: Could not setup access control list
Mär 01 13:08:20 dietpi systemd[1]: unbound.service: Main process exited, code=exited, status=1/FAILURE
Mär 01 13:08:20 dietpi systemd[1]: unbound.service: Failed with result 'exit-code'.
Mär 01 13:08:20 dietpi systemd[1]: Failed to start Unbound DNS server.
Mär 01 13:08:20 dietpi systemd[1]: unbound.service: Service RestartSec=100ms expired, scheduling restart.
Mär 01 13:08:20 dietpi systemd[1]: unbound.service: Scheduled restart job, restart counter is at 3.
Mär 01 13:08:20 dietpi systemd[1]: Stopped Unbound DNS server.
Mär 01 13:08:20 dietpi systemd[1]: Starting Unbound DNS server...
Mär 01 13:08:20 dietpi package-helper[656]: /var/lib/unbound/root.key has content
Mär 01 13:08:20 dietpi package-helper[656]: success: the anchor is ok
Mär 01 13:08:20 dietpi unbound[663]: [1614600500] unbound[663:0] error: cannot parse ip address: '.0/24'
Mär 01 13:08:20 dietpi unbound[663]: [1614600500] unbound[663:0] error: cannot parse access control: .0/24 allow
Mär 01 13:08:20 dietpi unbound[663]: [1614600500] unbound[663:0] fatal error: Could not setup access control list
Mär 01 13:08:20 dietpi systemd[1]: unbound.service: Main process exited, code=exited, status=1/FAILURE
Mär 01 13:08:20 dietpi systemd[1]: unbound.service: Failed with result 'exit-code'.
Mär 01 13:08:20 dietpi systemd[1]: Failed to start Unbound DNS server.
Mär 01 13:08:20 dietpi systemd[1]: unbound.service: Service RestartSec=100ms expired, scheduling restart.
Mär 01 13:08:20 dietpi systemd[1]: unbound.service: Scheduled restart job, restart counter is at 4.
Mär 01 13:08:20 dietpi systemd[1]: Stopped Unbound DNS server.
Mär 01 13:08:20 dietpi systemd[1]: Starting Unbound DNS server...
Mär 01 13:08:20 dietpi package-helper[689]: /var/lib/unbound/root.key has content
Mär 01 13:08:20 dietpi package-helper[689]: success: the anchor is ok
Mär 01 13:08:20 dietpi unbound[694]: [1614600500] unbound[694:0] error: cannot parse ip address: '.0/24'
Mär 01 13:08:20 dietpi unbound[694]: [1614600500] unbound[694:0] error: cannot parse access control: .0/24 allow
Mär 01 13:08:20 dietpi unbound[694]: [1614600500] unbound[694:0] fatal error: Could not setup access control list
Mär 01 13:08:20 dietpi systemd[1]: unbound.service: Main process exited, code=exited, status=1/FAILURE
Mär 01 13:08:20 dietpi systemd[1]: unbound.service: Failed with result 'exit-code'.
Mär 01 13:08:20 dietpi systemd[1]: Failed to start Unbound DNS server.
Mär 01 13:08:20 dietpi systemd[1]: unbound.service: Service RestartSec=100ms expired, scheduling restart.
Mär 01 13:08:20 dietpi systemd[1]: unbound.service: Scheduled restart job, restart counter is at 5.
[... cut ...]
Mär 01 13:17:27 dietpi systemd[1]: unbound.service: Service RestartSec=100ms expired, scheduling restart.
Mär 01 13:17:27 dietpi systemd[1]: unbound.service: Scheduled restart job, restart counter is at 5.
Mär 01 13:17:27 dietpi systemd[1]: Stopped Unbound DNS server.
Mär 01 13:17:27 dietpi systemd[1]: unbound.service: Start request repeated too quickly.
Mär 01 13:17:27 dietpi systemd[1]: unbound.service: Failed with result 'exit-code'.
Mär 01 13:17:27 dietpi systemd[1]: Failed to start Unbound DNS server.

Joulinar, 2021-03-01:
Issue seems to be with local IP address.

```
Mär 01 13:17:26 dietpi unbound[2699]: [1614601046] unbound[2699:0] error: cannot parse ip address: '.0/24'
Mär 01 13:17:26 dietpi unbound[2699]: [1614601046] unbound[2699:0] error: cannot parse access control: .0/24 allow
Mär 01 13:17:26 dietpi unbound[2699]: [1614601046] unbound[2699:0] fatal error: Could not setup access control list
```

can you share the IP address of your DietPi device pls.

MichaIng, 2021-03-01:
I wonder how it worked before since we do not apply the IP that way anymore, which caused that `.0/24` issue if the LAN IP was no `192.168.*`. I checked the update log from the bug report upload and the patch that merged `dietpi-pihole.conf` into `dietpi.conf` worked successfully.

The following should fix it:
```
# Remove all existing private IP and access control settings
sed -Ei '/^[[:blank:]]*(access-control|private-address):/d' /etc/unbound/unbound.conf.d/dietpi.conf
# Add a generic set of those, which implies all IP ranges that are reserved for private networks
cat << '_EOF_' >> /etc/unbound/unbound.conf.d/dietpi.conf

	# Control IP ranges which should be able to use this Unbound instance.
	# The DietPi defaults permit access from official local network IP ranges only, hence requests from www are denied.
	access-control: 0.0.0.0/0 refuse
	access-control: 10.0.0.0/8 allow
	access-control: 127.0.0.1/8 allow
	access-control: 172.16.0.0/12 allow
	access-control: 192.168.0.0/16 allow
	access-control: ::/0 refuse
	access-control: ::1/128 allow
	access-control: fd00::/8 allow
	access-control: fe80::/10 allow

	# Private IP ranges, which shall never be returned or forwarded as public DNS response.
	# NB: 127.0.0.1/8 is sometimes used by adblock lists, hence DietPi by default allows those as response.
	private-address: 10.0.0.0/8
	private-address: 172.16.0.0/12
	private-address: 192.168.0.0/16
	private-address: 169.254.0.0/16
	private-address: fd00::/8
	private-address: fe80::/10
_EOF_
```

MichaIng, 2021-03-15:
Is this still an issue or can it be closed?

MichaIng, 2021-03-24:
Marking as closed. Feel free to reopen when required.

closed 2021-03-24 by MichaIng