MichaIng/DietPi #4160: Stuck on apt-get -q update, due to outdated key.

maxymkov, 2021-03-01:
Hi,
Sorry, I'm not familiar with programming and stuck with installing Dietpi with below error:
Would pls anyone help with that?
I did everything was able to find with google, however got in another issue: gnupg, gnupg2 and gnupg1 do not seem to be installed, but one of them is required for this operation

#### Details:
- Date           | Mon Mar  1 13:36:02 EET 2021
- DietPi version | v6.35.0 (MichaIng/master)
- Image creator  | 
- Pre-image      | 
- Hardware       |  (ID=22)
- Kernel version | Linux sr-imx6 4.9.150-imx6-sr #1 SMP Sun Sep 8 01:52:51 UTC 2019 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | apt-get -q update
- Exit code      | 100
- Software title | DietPi-PREP
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Hit:1 https://repo.solid-build.xyz/debian/buster/bsp-any ./ InRelease
Get:2 https://repo.solid-build.xyz/debian/buster/bsp-imx6 ./ InRelease [1511 B]
Err:2 https://repo.solid-build.xyz/debian/buster/bsp-imx6 ./ InRelease
  The following signatures were invalid: EXPKEYSIG A86C36D7E45C02CD BSP:IMX6 OBS Project <BSP:IMX6@mxobs>
Reading package lists...
W: GPG error: https://repo.solid-build.xyz/debian/buster/bsp-imx6 ./ InRelease: The following signatures were invalid: EXPKEYSIG A86C36D7E45C02CD BSP:IMX6 OBS Project <BSP:IMX6@mxobs>
E: The repository 'https://repo.solid-build.xyz/debian/buster/bsp-imx6 ./ InRelease' is no longer signed.
```

ravenclaw900, 2021-03-01:
Hello,
Does running `sudo apt-key adv --keyserver keys.gnupg.net --recv-keys A86C36D7E45C02CD` to update the key fix the problem?

Joulinar, 2021-03-01:
Hi,

many thanks for your message. Looks like one of the repositories is using an invalid key. What image you are using?

maxymkov, 2021-03-01:
> apt-key adv --keyserver keys.gnupg.net --recv-keys A86C36D7E45C02CD

 Hi,
Thanks for help. No this return error: gnupg, gnupg2 and gnupg1 do not seem to be installed, but one of them is required for this operation. 
And I can't install gnupg.

maxymkov, 2021-03-01:
> Hi,
> 
> many thanks for your message. Looks like one of the repositories is using an invalid key. What image you are using?

Hi, thanks for help.
I'm using sr-imx6-debian-buster-20190925-cli.img.xz from  [solid-run](https://images.solid-run.com/IMX6/Debian?_ga=2.150701238.1809309557.1614594373-1210214346.1614594373)

Joulinar, 2021-03-01:
interesting. I don't know these. @MichaIng do you?

Basically it is the SolidRun repository `https://repo.solid-build.xyz/` failing. Maybe you would need to check with these guys.

MichaIng, 2021-03-01:
Looks like the manufacturer repository shipped with the image. It contains the kernel packages as well, e.g.: https://repo.solid-build.xyz/debian/buster/bsp-imx8v2/arm64/

It has a keyring package, which should assure up-to-date key, when installed:
```
cd /tmp
wget https://repo.solid-build.xyz/debian/buster/bsp-any/all/solidrun-keyring_2020.11.01_all.deb
dpkg -i solidrun-keyring_2020.11.01_all.deb
rm solidrun-keyring_2020.11.01_all.deb
apt update
apt upgrade
```
I checked the package content, and it adds the keys the regular way, so all fine with that. If the errors are still thrown, you should inform SolidRun and ask them to update they keys.

maxymkov, 2021-03-01:
> Looks like the manufacturer repository shipped with the image. It contains the kernel packages as well, e.g.: https://repo.solid-build.xyz/debian/buster/bsp-imx8v2/arm64/
>
> It has a keyring package, [...]

Many thanks for your kind help! 
Ive got system installed and running!

closed 2021-03-01 by maxymkov