MichaIng/DietPi #4161: unbound fails to start : 'fail: the anchor is NOT ok..'

acyanic, 2021-03-01:
#### Required Information
- DietPi version | 7.0.2
- Distro version | bustercat /etc/debian_version
- Kernel version | Linux DietPi7 5.10.11+ #1399 Thu Jan 28 12:02:28 GMT 2021 armv6l GNU/Linux
- SBC model | RPi Zero W (armv6l) 
- Power supply used | USB port on laptop
- SDcard used | (SanDisk ultra

#### Additional Information (if applicable)
- unbound with pihole
- freshly installed
- Bug report ID | a5a09e6d-788f-4b5e-8bac-566378dcb6db

#### Steps to reproduce

1. install DietPi (over wifi)
2. set static IP
3. restart
4. install pihole and unbound
5. use defaults
6. stop and start unbound has zero impact

#### Expected behaviour
unbound should run

#### Actual behaviour
pihole returns SERVFAIL from unbound

#### Extra details
Similar issue experienced when installed on 6.35
Stopping and starting unbound solved that and is still working fine
(on a separate Pi Zero)

MichaIng, 2021-03-01:
Many thanks for your report.

Can you please show the output of `journalctl -u unbound` from around the time when Pi-hole failed to resolve?

acyanic, 2021-03-01:
I thought I had pasted that in already ! ..
I restarted a couple of times.

journalctl -u unbound
-- Logs begin at Mon 2021-03-01 15:30:39 GMT, end at Mon 2021-03-01 16:50:02 GMT. --
Mar 01 15:51:16 DietPi7 systemd[1]: Starting Unbound DNS server...
Mar 01 15:51:16 DietPi7 package-helper[12376]: /var/lib/unbound/root.key does not exist, copying from /usr/share/dns/root.key
Mar 01 15:52:08 DietPi7 package-helper[12376]: /var/lib/unbound/root.key has content
Mar 01 15:52:08 DietPi7 package-helper[12376]: fail: the anchor is NOT ok and could not be fixed
Mar 01 15:52:08 DietPi7 systemd[1]: Started Unbound DNS server.
Mar 01 15:52:08 DietPi7 unbound[12381]: [12381:0] info: start of service (unbound 1.9.0).
Mar 01 16:24:52 DietPi7 unbound[12381]: [12381:0] info: service stopped (unbound 1.9.0).
Mar 01 16:24:52 DietPi7 systemd[1]: Stopping Unbound DNS server...
Mar 01 16:24:52 DietPi7 systemd[1]: unbound.service: Succeeded.
Mar 01 16:24:52 DietPi7 systemd[1]: Stopped Unbound DNS server.
Mar 01 16:24:52 DietPi7 systemd[1]: Starting Unbound DNS server...
Mar 01 16:25:44 DietPi7 package-helper[17470]: /var/lib/unbound/root.key has content
Mar 01 16:25:44 DietPi7 package-helper[17470]: fail: the anchor is NOT ok and could not be fixed
Mar 01 16:25:44 DietPi7 systemd[1]: Started Unbound DNS server.
Mar 01 16:25:44 DietPi7 unbound[17480]: [17480:0] info: start of service (unbound 1.9.0).
Mar 01 16:26:01 DietPi7 unbound[17480]: [17480:0] info: service stopped (unbound 1.9.0).
Mar 01 16:26:01 DietPi7 systemd[1]: Stopping Unbound DNS server...
Mar 01 16:26:01 DietPi7 systemd[1]: unbound.service: Succeeded.
Mar 01 16:26:01 DietPi7 systemd[1]: Stopped Unbound DNS server.
Mar 01 16:26:26 DietPi7 systemd[1]: Starting Unbound DNS server...
Mar 01 16:27:17 DietPi7 package-helper[17494]: /var/lib/unbound/root.key has content
Mar 01 16:27:17 DietPi7 package-helper[17494]: fail: the anchor is NOT ok and could not be fixed
Mar 01 16:27:17 DietPi7 systemd[1]: Started Unbound DNS server.
Mar 01 16:27:17 DietPi7 unbound[17502]: [17502:0] info: start of service (unbound 1.9.0).

Joulinar, 2021-03-01:
In generell the service is starting. Even if `anchor`, it should not prevent the system from being able to use `unbound`.

To fix `anchor` during boot, do following:

```
mkdir -p /etc/systemd/system/ifup@.service.d
echo -e '[Service]\nType=oneshot' > /etc/systemd/system/ifup@.service.d/dietpi.conf
```

acyanic, 2021-03-01:
Thanks I'll try tomorrow.
I assumed anchor failing was causing the SERVFAIL message in pihole.
Also, is this different to 6.34 version, ie requiring an ifup service ?

Joulinar, 2021-03-01:
`anchor` was failing all the time on reboot if `unbound` was starting before a valid network connection could be established. The `ifup` services should handle this more gracefully. 

I hope it's explained correctly @MichaIng

acyanic, 2021-03-02:
sleeping on it helped... My apologies, I solved the main issue.
I have DNS based filtering turned on on my router, with custom DNS being the piholes.
I had to add the new pihole/unbound IP to 'no filtering'.

Still had the 'fail: anchor is NOT ok', but unbound was working like you said.

I ran your commands and that fixed the anchor issue.  Want thanks. This is closed for me.

journalctl -u unbound
-- Logs begin at Tue 2021-03-02 10:27:44 GMT, end at Tue 2021-03-02 10:28:51 GMT. --
Mar 02 10:27:56 DietPi systemd[1]: Starting Unbound DNS server...
Mar 02 10:27:58 DietPi package-helper[315]: /var/lib/unbound/root.key has content
Mar 02 10:27:58 DietPi package-helper[315]: success: the anchor is ok
Mar 02 10:27:58 DietPi unbound[341]: [341:0] info: start of service (unbound 1.9.0).
Mar 02 10:27:58 DietPi systemd[1]: Started Unbound DNS server.

closed 2021-03-02 by acyanic

MichaIng, 2021-03-02:
Great that it works now. The only thing I'm wondering is that the DietPi v7.0 update should have included the ifup@.service patch already 🤔: https://github.com/MichaIng/DietPi/blob/master/rootfs/etc/systemd/system/ifup@.service.d/dietpi.conf

Explanation for that patch is basically right: It is responsible to configure each defined network interface at boot, which includes starting the DHCP client and/or wpa_supplicant (for WiFi) in case, applying the IP address, setting up the routes etc.
By default Unbound could have started right after that ifup service has started, which means that it could be still in the process of waiting for the DHCP lease e.g., so that no actual network connectivity is available yet. With `Type=oneshot`, the service now needs to finish, before Unbound is started, so network connectivity is guaranteed (unless there was a failure of course).

Joulinar, 2021-03-02:
but is it introduced on exiting system/installations or on new systems/installations only? Just asking...

MichaIng, 2021-03-02:
> but is it introduced on exiting system/installations

Yes it is, all files from [here](https://github.com/MichaIng/DietPi/tree/master/rootfs) are copied as is to the systems root directory.

acyanic, 2021-03-02:
My bad again I think.  The image was not new but the 6.34 I already had, so maybe it didn't pull in all the 7 bits and pieces.
I'll try again, with the existing 6.34 image I have and download a new 7.

acyanic, 2021-03-02:
Well, the image is the same , dated 18 Jan 2021.
Of course you might say, DietPi is scripts and downloads the latest versions at run time.
And, you might also say the issue is with the DietPi-software install script for Unbound.
And you will say, that has been fixed already.
I see the beta thread conversation and commit @MichaInge 558f1d
With a bit more clever searching I could have found that sooner, but I still had to fix my DNS filter.

Thanks again for your time and patience.

Joulinar, 2021-03-02:
Yes there is no new v7 image as the final version has been release just one day ago. As this project is mainly maintained by a single person, it's nearly impossible to update all our images in such short time frame.

Anyway, I did a test installation of `unbound` on 6.34.3 and their `anchor` was failing as expected. Afterwards I did the update to v7.0.2 and ifup@.service was installed automatically. Means `anchor` is working as expected now.

acyanic, 2021-03-02:
Understood.  I feel a contribution coming ...
I did want to get on board for the beta test. I became interested in Unbound as I was discovering Raspberry Pi Zero W (it's a great little thing) and DietPi (another great thing), but I was too late.  Maybe next time.

MichaIng, 2021-03-02:
All fine, we'd be happy if you help testing the next beta, when it's ready 😃!

Even that the images are older DietPi versions, they are updated on first boot before any software installs can be done, so practically the issue is solved for those as well. But I'll update them soon, mainly to speed up the first boot and to not have the Microsoft VSCode repository added temporarily on RPi (#4083).