MichaIng/DietPi #4162: DietPi-Software | Nextcloud: Files High Performance Back-end

MichaIng, 2021-03-01:
as an optional software option like we have Nextcloud Talk.

Announcement: https://help.nextcloud.com/t/nextcloud-faster-than-ever-introducing-files-high-performance-back-end/108633
GitHub: https://github.com/nextcloud/notify_push

On small instances, the theoretical performance benefit is probably negligible and instead you have an additional daemon running on the server. But you get instant push notifications for clients and the browser, which makes it great in combination with Nextcloud Talk, e.g. you don't want to miss an incoming call by getting the notification 15 seconds later.

I'm currently setting it up for my home instance, let's see how it runs there.

MichaIng, 2021-03-04:
It seems to work fine. Can be configured to listen on a UNIX socket, so no additional network port it listened on. A reverse proxy needs and can be configured on any webserver easily.
Memory usage is ~5 MiB, not sure if that depends on usage as well.

One benefit is that client requests "can" be lowered, since they get push notifications from the server instead of needing to regularly send a request to check for changes. But I'm not sure whether clients somehow know or get informed by this by default, since the desktop client at least has no setting to set the check interval.

The other benefit is that clients also do not need to inform Nextcloud directly via the "slow" protocols and overhead, but do that faster via the slim dedicated push daemon, which communicates with the database directly. Same here, how do clients know that a push daemon is available?

And is there an automatic fallback when the push daemon is down?

I'll check back on the NC forum about this.

nicojx, 2025-03-31:
Your post is quite old now, but since v30 my Nextcloud is complaining about the lack of a high-performance backend, related to Nextcloud Talk:
`No High-performance backend configured - Running Nextcloud Talk without the High-performance backend only scales for very small calls (max. 2-3 participants). Please set up the High-performance backend to ensure calls with multiple participants work seamlessly. For more details see the [documentation ↗](https://nextcloud-talk.readthedocs.io/en/latest/quick-install/).`

Is this still something that is worth including as standard?

MichaIng, 2025-04-18:
The one for Nextcloud Talk is however a different thing/topic 😉. Here I mean the `notify_push` app for Nextcloud clients, which was called "High Performance Back-end" as well, confusingly, as it is really nothing more than a push notification server to inform clients about file changes, so clients don't need to regularly ask the server (which can imply quite a significant constant server load in case of large instances with many users/clients).