MichaIng/DietPi #4163: Specific issue with Node-RED, is this a DietPi thing or a Node-RED / StreamDeck thing?

MDAR, 2021-03-02:
### This may be a bug report for DietPi or it might be a bug report for Node-RED or the Node creator
Unfortunately my skills aren't good enough to take this any further without someone's assistance.

#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=34
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | 10.8
- Kernel version | Linux LiveStream 3.16.85+ #1 SMP PREEMPT Tue Jun 30 19:02:35 CEST 2020 aarch64 GNU/Linux
- SBC model | Odroid C2 (aarch64)
- Power supply used | 5V 3AMP
- SDcard used | eMMC 16GB

#### Additional Information (if applicable)
- Software title | Node-RED
- Software was added / updated using the dietpi-software script
- Can this issue be replicated on a fresh installation of DietPi? - Yes

#### Steps to reproduce

1. ... I have installed Node-RED as per the dietpi-software script and can get almost everything working perfectly, the issue is with a specific node set for Node-RED (now running as V1.2.9)
[StreamDeck for Node-RED](https://flows.nodered.org/node/node-red-contrib-streamdeck) V1.0.2
2. ... The author of this node is trying to be extremely helpful, but everything we try seems to bring us back to a very specific error when trying to load a dependency.
3. The issue can be reviewed here
[https://github.com/natcl/node-red-contrib-streamdeck/issues/4](https://github.com/natcl/node-red-contrib-streamdeck/issues/4)

#### Expected behaviour

- ... I was expecting the node to install as normal and be operational within Node-RED

#### Actual behaviour

- ... The node set or one of it's dependencies 
- The log as posted on Nat's GitHub page shows all the details of items that fail

MichaIng, 2021-03-02:
Many thanks for your report. I answered at the other issue, so we keep debugging and hopefully the solution at one place.

MDAR, 2021-03-02:
@MichaIng 

Thanks Michal

I'll follow your advice and see what happens.

closed 2021-03-02 by MDAR