MichaIng/DietPi #4165: Date format in dietpi-banner

math-gout, 2021-03-02:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
The dietpi-banner date format is mm/dd/yy, which I think can be confusing for people who are used to the dd/mm/yy format

#### Describe the solution you'd like:
I was wondering if it would be possible to add an option to select the date format in dietpi-banner. It would be better than patching /boot/dietpi/func/dietpi-banner by hand after each update

This is the line I edit :
```sh
local text_update_available_date=$(LC_ALL=${locale:-C.UTF-8} date +"%R - %a %x")
````
```sh
local text_update_available_date=$(LC_ALL=${locale:-C.UTF-8} date +"%R - %a %d/%m/%y")
```
(I don't know if there is a better way to do it)

Thanks for your time and for maintaining DietPi !

MichaIng, 2021-03-03:
Many thanks for your suggestion.

Actually the date representation is based on the locale already:
> %x   locale's date representation

Hence if you change the system locale via `dietpi-config` > `Language/Regional Options`, it should be presented the way that is common in your region. Setting now an explicit order, would destroy that behaviour which was added explicitly, including a lot of other common date formats like `%d.%m.%y` and such, a while ago (that `LC_ALL=${locale:-C.UTF-8}` part, which means to use the user chosen locale, and `C.UTF-8` only as fallback).

Does `date` (without command arguments) called from console show it the correct way for you? Ah, today 03/03 is a bad day to test that 😄.

Joulinar, 2021-03-03:
let's see how it looks tomorrow 😃

math-gout, 2021-03-03:
@MichaIng This is what I get after typing ```date```:
```
Wed Mar  3 16:19:18 CET 2021
```
I also tried ```LC_ALL=fr_FR.UTF-8 date``` and got the same result

Then I tried ```LC_ALL=fr_FR.UTF-8 date``` on my mac and got the expected result :
```
Mer  3 mar 2021 16:26:27 CET
```

I don't really understand why the ```LC_ALL=fr_FR.UTF-8``` part didn't have any effect on my DietPi installation

I haven't tried to change system locale via ```dietpi-config``` yet as I want to keep my system in english, and I just want to change date format to dd/mm/yy

MichaIng, 2021-03-03:
> I haven't tried to change system locale via dietpi-config yet as I want to keep my system in english, and I just want to change date format to dd/mm/yy

Ah I see that. So the French locales are not installed/compiled on your system. dietpi-config does an all or nothing selection and compiles only the one locale that was selected.

For now you can cheat the scripts that way:
```
# Compile the French locales
localedef -i fr_FR -f UTF-8 fr_FR.UTF-8
# The following command should show it now to be available
locale -a
# Apply it to dietpi.txt, since this is where the banner takes the info from
G_CONFIG_INJECT 'AUTO_SETUP_LOCALE=' 'AUTO_SETUP_LOCALE=fr_FR.UTF-8' /boot/dietpi.txt
```
All other DietPi scripts forcefully use `C.UTF-8` anyway, which is always available as static locale, so that any command output can reliably parsed.
You console stays in English, since this depends on `cat /etc/default/locale`. If you want the date in French generally, but keeping all other localisations in English, you can do the following:
```
echo -e 'LANG=C.UTF-8\nLC_TIME=fr_FR.UTF-8' > /etc/default/locale
unset LC_ALL
. /etc/default/locale
```
So the default locale will stay as it was, but for all time-related it will be French. `LC_ALL` btw is an override for those individual localisations, hence it needs to be unset and removed from that file. The effective individual ones can be checked via: `locale`

math-gout, 2021-03-03:
Ok thanks ! That dit the trick !
Just one last question, will this stick after updating DietPi ? Or will I have to re-run these commands every time ? And if so, is there a way to run user script automatically after updating ?

MichaIng, 2021-03-03:
This change is update-persistent. Only when using dietpi-config to re-set the locale, this would override your change. But also, I just made the script a bit less intrusive: https://github.com/MichaIng/DietPi/commit/c429cd76d864229277b44ef2229aecb4b0fd0774
It does not set the LC_ALL variable anymore so that at least one step less is required to override individual LC_s. I'll see if we re-add the ability to compile multiple locales at a later time.

math-gout, 2021-03-04:
Well I'm all good now ! Thanks a lot for your time and support !

MichaIng, 2021-03-04:
Okay great. I'll mark this issue as closed. Feel free to reopen if required.

closed 2021-03-04 by MichaIng

Joulinar, 2021-03-04:
for me it still looks incorrect, even if I have set `Timezone : [Europe/Berlin]` inside `dietpi-config`

`DietPi v7.0.2 : 20:42 - Thu 03/04/21`

MichaIng, 2021-03-04:
Not the timezone, but the locale/language is relevant. That works fine here:
```
# LC_ALL=de_DE.utf8 date '+%x'
04.03.2021
# dietpi-banner 0
 ─────────────────────────────────────────────────────
 DietPi v7.1.0 (dev) : 21:35 - Do 04.03.2021
 ─────────────────────────────────────────────────────
```

Joulinar, 2021-03-04:
got it, it's working now

```
root@DietPi3:~# dietpi-config
[ SUB1 ] DietPi-Set_software > locale (de_DE.UTF-8)
[  OK  ] DietPi-Set_software | dpkg-reconfigure -f noninteractive locales
[  OK  ] DietPi-Set_software | Setting in /boot/dietpi.txt adjusted: AUTO_SETUP_LOCALE=de_DE.UTF-8
[  OK  ] locale de_DE.UTF-8 | Completed
root@DietPi3:~# dietpi-banner 0
 ─────────────────────────────────────────────────────
 DietPi v7.0.2 : 22:07 - Do 04.03.2021
 ─────────────────────────────────────────────────────
 - LAN IP : 192.168.0.12 (eth0)
root@DietPi3:~#
```