MichaIng/DietPi #4167: [question] is there a way to dismiss system update?

k1r0s, 2021-03-02:
I get this message whenever I open a term promp: DietPi-Update | Latest version  : v6.35.0

Is there a way to dismiss it?

ravenclaw900, 2021-03-02:
There is no way to dismiss it, as it is reminding you to update your system, but you can use `dietpi-update` to run the update.

MichaIng, 2021-03-03:
Same with the APT update message. I was thinking to make the banner itself optional, or making the upper parts of it optional as well, but generally I like to remind users about any kind of updates since I find it important to update DietPi and APT packages, whenever there is some time to spare and in case a server downtime possible. Most problems appear when deferring such too long so that the update includes a huge number of APT packages and incremental DietPi patches.

k1r0s, 2021-03-10:
yep, i've already removed some banner parts although this one is the most annoying but i get the point. I'm planning to upgrade asap

closed 2021-03-10 by k1r0s