MichaIng/DietPi #4168: Mariadb error, can't install Nextcloud

manomaya90, 2021-03-03:
#### Details:
- Date           | Wed Mar  3 07:48:00 WIB 2021
- Bug report     | 2ffc5456-8e29-4f18-9b44-63362da29b5e
- DietPi version | v7.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | Linux DietPi 5.4.83-v7l+ #1379 SMP Mon Dec 14 13:11:54 GMT 2020 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | systemctl restart mariadb
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. Install Nextcloud
2. Wait a moment
#### Expected behaviour:

- Nextcloud is installed
#### Actual behaviour:

- The installation failed because of an error with Mariadb
#### Extra details:

- This is installation on a fresh DietPi
#### Additional logs:
```
Job for mariadb.service failed because the control process exited with error code.
See "systemctl status mariadb.service" and "journalctl -xe" for details.
```

Joulinar, 2021-03-03:
Hi,

many thanks for your message. Pls can you share following

```
systemctl restart mariadb.service 
journalctl -u mariadb 
cat /var/log/mysql/error.log 
readlink /var/lib/mysql 
readlink -f /var/lib/mysql 
```

manomaya90, 2021-03-03:
Here it is. Hope it can help.

systemctl restart mariadb.service
> root@DietPi:~# systemctl restart mariadb.service 
Job for mariadb.service failed because the control process exited with error code.
See "systemctl status mariadb.service" and "journalctl -xe" for details.

journalctl -u mariadb
> root@DietPi:~# journalctl -u mariadb
-- Logs begin at Wed 2021-03-03 14:15:03 WIB, end at Wed 2021-03-03 16:02:00 WIB. --
Mar 03 14:16:17 DietPi systemd[1]: Starting MariaDB 10.3.27 database server...
Mar 03 14:16:18 DietPi mysqld[966]: 2021-03-03 14:16:18 0 [Note] /usr/sbin/mysqld (mysqld
 10.3.27-MariaDB-0+deb10u1) starting as process 966 ...
Mar 03 14:16:18 DietPi mysqld[966]: 2021-03-03 14:16:18 0 [Warning] Could not increase nu
mber of max_open_files to more than 16384 (request: 32186)
Mar 03 14:16:18 DietPi systemd[1]: mariadb.service: Main process 
exited, code=exited, status=1/FAILURE
Mar 03 14:16:18 DietPi systemd[1]: mariadb.service: Failed with r
esult 'exit-code'.
Mar 03 14:16:18 DietPi systemd[1]: Failed to start MariaDB 10.3.2
7 database server.
Mar 03 16:02:00 DietPi systemd[1]: Starting MariaDB 10.3.27 database server...
Mar 03 16:02:00 DietPi mysqld[2200]: 2021-03-03 16:02:00 0 [Note] /usr/sbin/mysqld (mysql
d 10.3.27-MariaDB-0+deb10u1) starting as process 2200 ...
Mar 03 16:02:00 DietPi mysqld[2200]: 2021-03-03 16:02:00 0 [Warning] Could not increase n
umber of max_open_files to more than 16384 (request: 32186)
Mar 03 16:02:00 DietPi systemd[1]: mariadb.service: Main process 
exited, code=exited, status=1/FAILURE
Mar 03 16:02:00 DietPi systemd[1]: mariadb.service: Failed with r
esult 'exit-code'.
Mar 03 16:02:00 DietPi systemd[1]: Failed to start MariaDB 10.3.2
7 database server.

cat /var/log/mysql/error.log
> root@DietPi:~# cat /var/log/mysql/error.log 
2021-03-03 16:02:00 0 [Note] InnoDB: Using Linux native AIO
2021-03-03 16:02:00 0 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
2021-03-03 16:02:00 0 [Note] InnoDB: Uses event mutexes
2021-03-03 16:02:00 0 [Note] InnoDB: Compressed tables use zlib 1.2.11
2021-03-03 16:02:00 0 [Note] InnoDB: Number of pools: 1
2021-03-03 16:02:00 0 [Note] InnoDB: Using generic crc32 instructions
2021-03-03 16:02:00 0 [Note] InnoDB: Initializing buffer pool, total size = 128M, instances = 1, chunk size = 128M
2021-03-03 16:02:00 0 [Note] InnoDB: Completed initialization of buffer pool
2021-03-03 16:02:00 0 [Note] InnoDB: If the mysqld execution user is authorized, page cleaner thread priority can be changed. See the man page of setpriority().
2021-03-03 16:02:00 0 [Note] InnoDB: 128 out of 128 rollback segments are active.
2021-03-03 16:02:00 0 [Note] InnoDB: Removed temporary tablespace data file: "ibtmp1"
2021-03-03 16:02:00 0 [Note] InnoDB: Creating shared tablespace for temporary tables
2021-03-03 16:02:00 0 [Note] InnoDB: Setting file './ibtmp1' size to 12 MB. Physically writing the file full; Please wait ...
2021-03-03 16:02:00 0 [Note] InnoDB: File './ibtmp1' size is now 12 MB.
2021-03-03 16:02:00 0 [Note] InnoDB: 10.3.27 started; log sequence number 1602128; transaction id 8
2021-03-03 16:02:00 0 [Note] InnoDB: Loading buffer pool(s) from /mnt/dietpi_userdata/mysql/ib_buffer_pool
2021-03-03 16:02:00 0 [Note] Plugin 'FEEDBACK' is disabled.
2021-03-03 16:02:00 0 [ERROR] Could not open mysql.plugin table. Some plugins may be not loaded
2021-03-03 16:02:00 0 [Note] Recovering after a crash using tc.log
2021-03-03 16:02:00 0 [Note] Starting crash recovery...
2021-03-03 16:02:00 0 [Note] Crash recovery finished.
2021-03-03 16:02:00 0 [Note] InnoDB: Cannot open '/mnt/dietpi_userdata/mysql/ib_buffer_pool' for reading: No such file or directory
2021-03-03 16:02:00 0 [ERROR] Can't open and lock privilege tables: Table 'mysql.servers' doesn't exist
2021-03-03 16:02:00 0 [Note] Server socket created on IP: '127.0.0.1'.
2021-03-03 16:02:00 0 [ERROR] Fatal error: Can't open and lock privilege tables: Table 'mysql.user' doesn't exist

readlink /var/lib/mysql
>root@DietPi:~# readlink /var/lib/mysql
/mnt/dietpi_userdata/mysql

readlink -f /var/lib/mysql
>root@DietPi:~# readlink -f /var/lib/mysql
/mnt/dietpi_userdata/mysql

Joulinar, 2021-03-03:
can you check content of `ls -la /mnt/dietpi_userdata/mysql`

manomaya90, 2021-03-03:
> root@DietPi:~# ls -la /mnt/dietpi_userdata/mysql
total 122928
drwxr-xr-x  3 mysql  mysql      4096 Mar  3 16:02 .
drwxrwxr-x 10 dietpi dietpi     4096 Mar  3 08:58 ..
-rw-rw----  1 mysql  mysql      8192 Mar  3 16:02 aria_log.00000001
-rw-rw----  1 mysql  mysql        52 Mar  3 16:02 aria_log_control
-rw-r--r--  1 root   root          0 Mar  2 19:26 debian-10.3.flag
-rw-rw----  1 mysql  mysql  50331648 Mar  3 16:02 ib_logfile0
-rw-rw----  1 mysql  mysql  50331648 Mar  2 09:23 ib_logfile1
-rw-rw----  1 mysql  mysql  12582912 Mar  2 19:26 ibdata1
-rw-rw----  1 mysql  mysql  12582912 Mar  3 16:02 ibtmp1
drwx------  2 mysql  mysql      4096 Mar  2 19:26 mysql
-rw-rw----  1 mysql  mysql     24576 Mar  3 16:02 tc.log

Joulinar, 2021-03-03:
there seems to be something missing, can you share `ls -la /mnt/dietpi_userdata/mysql/mysql`

manomaya90, 2021-03-03:
> root@DietPi:~# ls -la /mnt/dietpi_userdata/mysql/mysql
total 12
drwx------ 2 mysql mysql 4096 Mar  2 19:26 .
drwxr-xr-x 3 mysql mysql 4096 Mar  3 18:03 ..
-rw-rw---- 1 mysql mysql   67 Mar  2 09:23 db.opt

Joulinar, 2021-03-03:
that doesn't looks healthy. I would recommend to remove MariaDB and try to intsall it again.

```
G_AGP mariadb-server
rm -Rf /{mnt/dietpi_userdata,var/lib,var/log,etc}/mysql
```

once done reboot your system and try to install MariaDB again `dietpi-software install 88`

manomaya90, 2021-03-03:
I have done all above. MariaDB has been installed but still failed to start. I tried installing Nextcloud, I believe it still has the same error.

Wonder whether this only happens to me? I have tried install it in the 64bit first but failed, so I thought maybe because it's in beta. But it's the same for this one. Beside unable to install Nextcloud, I also can't install Home Assistant (but that one just stuck for ages when installing pyhton and I have to end the ssh session). Though I can install Jellyfin, Rpimonitor, and Docker just fine.

Joulinar, 2021-03-03:
You don't need to install Nextcloud as along a MariaDB is not working

I did a test installation on my RPi4B running 32bit and MariaDB is working without issues. So it looks like dedicated to you. This we would need to find out, still. 

What is a little bit strange to me, you are running DietPi v7.0.2, however your kernel version is still `5.4.83`. However `dietpi-update` should have done an `apt` package update before. This include the new Raspberry OS kernel 5.10.11 as well 🤔

manomaya90, 2021-03-03:
Hmm.. I'm not sure either.  What should the kernel supposed to be?

Joulinar, 2021-03-03:
it should be `5.10.11`

```
root@DietPi4:~# uname -a
Linux DietPi4 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux
root@DietPi4:~#
```

What happen if you try `apt update && apt upgrade`

manomaya90, 2021-03-03:
Not much.
> 0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

Joulinar, 2021-03-03:
can you try to reboot and check `uname -a`. It is still given `5.4.83` ??

manomaya90, 2021-03-03:
Ah it's the most current kernel now.
>Linux DietPi 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux

Joulinar, 2021-03-03:
@MichaIng 
any ideas why MariaDB is not able to install?

MichaIng, 2021-03-03:
Please do the following:
```
G_AGP mariadb-server
G_AGA # This is important to purge the actual server packages, the above is only a dummy wrapper package
rm -Rf /{mnt/dietpi_userdata,var/lib,var/log,etc}/mysql
mkdir /mnt/dietpi_userdata/mysql
ln -s /mnt/dietpi_userdata/mysql /var/lib/mysql
apt install mariadb-server
systemctl status mariadb
```
And please paste the output of the `apt install` command if it fails, which is should when the service fails to start.

manomaya90, 2021-03-04:
Here it is:

```
root@DietPi:~# apt install mariadb-server
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  galera-3 libaio1 libconfig-inifiles-perl libdbi-perl libreadline5 lsof
  mariadb-client-10.3 mariadb-client-core-10.3 mariadb-common mariadb-server-10.3
  mariadb-server-core-10.3 mysql-common rsync socat
Suggested packages:
  libclone-perl libmldbm-perl libnet-daemon-perl libsql-statement-perl mailx
  mariadb-test netcat-openbsd tinyca openssh-client openssh-server
Recommended packages:
  libdbd-mysql-perl libterm-readkey-perl libhtml-template-perl
The following NEW packages will be installed:
  galera-3 libaio1 libconfig-inifiles-perl libdbi-perl libreadline5 lsof
  mariadb-client-10.3 mariadb-client-core-10.3 mariadb-common mariadb-server
  mariadb-server-10.3 mariadb-server-core-10.3 mysql-common rsync socat
0 upgraded, 15 newly installed, 0 to remove and 4 not upgraded.
Need to get 17.3 MB of archives.
After this operation, 147 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://kartolo.sby.datautama.net.id/raspbian/raspbian buster/main armhf mysql-common all 5.8+1.0.5 [7324 B]
Get:2 http://mirror.ossplanet.net/raspbian/raspbian buster/main armhf mariadb-common all 1:10.3.27-0+deb10u1 [32.2 kB]
Get:3 http://kartolo.sby.datautama.net.id/raspbian/raspbian buster/main armhf galera-3 armhf 25.3.25-2 [811 kB]
Get:4 http://mirror.ossplanet.net/raspbian/raspbian buster/main armhf libdbi-perl armhf 1.642-1+deb10u2 [767 kB]
Get:5 http://kartolo.sby.datautama.net.id/raspbian/raspbian buster/main armhf lsof armhf 4.91+dfsg-1 [307 kB]
Get:6 http://kartolo.sby.datautama.net.id/raspbian/raspbian buster/main armhf libconfig-inifiles-perl all 3.000001-1 [51.9 kB]
Get:7 http://kartolo.sby.datautama.net.id/raspbian/raspbian buster/main armhf libaio1 armhf 0.3.112-3 [10.8 kB]
Get:8 http://kartolo.sby.datautama.net.id/raspbian/raspbian buster/main armhf libreadline5 armhf 5.2+dfsg-3 [103 kB]
Get:9 http://mirror.ossplanet.net/raspbian/raspbian buster/main armhf mariadb-client-core-10.3 armhf 1:10.3.27-0+deb10u1 [4712 kB]
Get:12 http://kartolo.sby.datautama.net.id/raspbian/raspbian buster/main armhf rsync armhf 3.1.3-6 [356 kB]
Get:13 http://kartolo.sby.datautama.net.id/raspbian/raspbian buster/main armhf socat armhf 1.7.3.2-2 [324 kB]
Get:10 http://mirror.ossplanet.net/raspbian/raspbian buster/main armhf mariadb-client-10.3 armhf 1:10.3.27-0+deb10u1 [984 kB]
Get:11 http://mirror.ossplanet.net/raspbian/raspbian buster/main armhf mariadb-server-core-10.3 armhf 1:10.3.27-0+deb10u1 [5209 kB]
Get:14 http://mirror.ossplanet.net/raspbian/raspbian buster/main armhf mariadb-server-10.3 armhf 1:10.3.27-0+deb10u1 [3563 kB]
Get:15 http://mirror.ossplanet.net/raspbian/raspbian buster/main armhf mariadb-server all 1:10.3.27-0+deb10u1 [31.2 kB]
Fetched 17.3 MB in 29s (593 kB/s)                                                       
debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package mysql-common.
(Reading database ... 45689 files and directories currently installed.)
Preparing to unpack .../00-mysql-common_5.8+1.0.5_all.deb ...
Unpacking mysql-common (5.8+1.0.5) ...
Selecting previously unselected package mariadb-common.
Preparing to unpack .../01-mariadb-common_1%3a10.3.27-0+deb10u1_all.deb ...
Unpacking mariadb-common (1:10.3.27-0+deb10u1) ...
Selecting previously unselected package galera-3.
Preparing to unpack .../02-galera-3_25.3.25-2_armhf.deb ...
Unpacking galera-3 (25.3.25-2) ...
Selecting previously unselected package libdbi-perl:armhf.
Preparing to unpack .../03-libdbi-perl_1.642-1+deb10u2_armhf.deb ...
Unpacking libdbi-perl:armhf (1.642-1+deb10u2) ...
Selecting previously unselected package lsof.
Preparing to unpack .../04-lsof_4.91+dfsg-1_armhf.deb ...
Unpacking lsof (4.91+dfsg-1) ...
Selecting previously unselected package libconfig-inifiles-perl.
Preparing to unpack .../05-libconfig-inifiles-perl_3.000001-1_all.deb ...
Unpacking libconfig-inifiles-perl (3.000001-1) ...
Selecting previously unselected package libaio1:armhf.
Preparing to unpack .../06-libaio1_0.3.112-3_armhf.deb ...
Unpacking libaio1:armhf (0.3.112-3) ...
Selecting previously unselected package libreadline5:armhf.
Preparing to unpack .../07-libreadline5_5.2+dfsg-3_armhf.deb ...
Unpacking libreadline5:armhf (5.2+dfsg-3) ...
Selecting previously unselected package mariadb-client-core-10.3.
Preparing to unpack .../08-mariadb-client-core-10.3_1%3a10.3.27-0+deb10u1_armhf.deb ...
Unpacking mariadb-client-core-10.3 (1:10.3.27-0+deb10u1) ...
Selecting previously unselected package mariadb-client-10.3.
[... 45 lines cut ...]
o
     Docs: man:mysqld(8)
           https://mariadb.com/kb/en/library/systemd/
 Main PID: 2838 (code=exited, status=1/FAILURE)
   Status: "MariaDB server is down"

Mar 03 20:19:48 DietPi mysqld[2838]: 2021-03-03 20:19:48 0 [ERROR] Plugin 'InnoDB' init f
unction returned error.
Mar 03 20:19:48 DietPi mysqld[2838]: 2021-03-03 20:19:48 0 [ERROR] Plugin 'InnoDB' regist
ration as a STORAGE ENGINE failed.
Mar 03 20:19:48 DietPi mysqld[2838]: 2021-03-03 20:19:48 0 [Note] Plugin 'FEEDBACK' is di
sabled.
Mar 03 20:19:48 DietPi mysqld[2838]: 2021-03-03 20:19:48 0 [ERROR] Could not open mysql.p
lugin table. Some plugins may be not loaded
Mar 03 20:19:48 DietPi mysqld[2838]: 2021-03-03 20:19:48 0 [ERROR] Unknown/unsupported st
orage engine: InnoDB
Mar 03 20:19:48 DietPi mysqld[2838]: 2021-03-03 20:19:48 0 [ERROR] Aborting
Mar 03 20:19:48 DietPi mysqld[2838]: Warning: Memory not freed: 392
Mar 03 20:19:48 DietPi systemd[1]: mariadb.service: Main process 
exited, code=exited, status=1/FAILURE
Mar 03 20:19:48 DietPi systemd[1]: mariadb.service: Failed with r
esult 'exit-code'.
Mar 03 20:19:48 DietPi systemd[1]: Failed to start MariaDB 10.3.2
7 database server.
```

manomaya90, 2021-03-04:
I am reflashing my ssd right now and I notice that in the first boot the APT upgrade is stuck without any data being transfered (I look at my router). My connection is good though, I can stream videos just fine. Perhaps this is the reason why my kernel isn't up to date before?
```
 ─────────────────────────────────────────────────────
 DietPi v6.34.3 : 00:40 - Thu 03/04/21
 ─────────────────────────────────────────────────────
 - LAN IP : 10.10.10.10 (eth0)
[  OK  ] DietPi-Login | Checking network connectivity
[  OK  ] DietPi-Login | Checking DNS resolver
[  OK  ] DietPi-Run_NTPD | systemctl restart systemd-timesyncd
[ INFO ] DietPi-Run_NTPD | Waiting for completion of systemd-timesyncd (1/60)
[  OK  ] DietPi-Run_NTPD | systemd-timesyncd synced
[  OK  ] Network time sync | Completed

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for available DietPi update

[ INFO ] DietPi-Update | Checking mirror: https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi/server_version-6
[  OK  ] DietPi-Update | Using update server: https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi/server_version-6
[  OK  ] DietPi-Update | Update available:
[ INFO ] DietPi-Update | Current version : v6.34.3
[ INFO ] DietPi-Update | Latest version  : v6.35.0

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for update pre-requirements

[  OK  ] DietPi-Update | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Update | Free space check: path=/ | available=114444 MiB | required=100 MiB
[ SUB1 ] DietPi-Services > stop 
[ INFO ] DietPi-Services | skip : cron (due to mask)

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Applying pre-patches

[  OK  ] DietPi-Update | Downloading pre-patch file
[  OK  ] DietPi-Update | Applying execute permission
[ INFO ] Pre-patch 25 | Running DietPi v7 pre-patches
[  OK  ] DietPi-Update | Successfully applied pre-patches

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Upgrading APT packages

[ INFO ] DietPi-Update | APT update, please wait...
Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:2 https://archive.raspberrypi.org/debian buster InRelease [32.8 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian buster/non-free armhf Packages [104 kB]
Get:4 https://archive.raspberrypi.org/debian buster/main armhf Packages [364 kB]
Get:5 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Get:6 http://raspbian.raspberrypi.org/raspbian buster/contrib armhf Packages [58.7 kB]
Fetched 13.6 MB in 35s (392 kB/s)
Reading package lists...
[  OK  ] DietPi-Update | APT update
[ INFO ] DietPi-Update | APT upgrade, please wait...
```

Joulinar, 2021-03-04:
well it's loading the packages in background now. Speed depends on the raspberrypi.org server hosting the files. Sometimes they are extremely slow.

manomaya90, 2021-03-04:
Ok, but not even one packet is received or send. I'll wait. Isn't it better for the server to automatically be changed to the nearest one? Maybe that can help with the speed.

Joulinar, 2021-03-04:
it should but this is something out of out control.

MichaIng, 2021-03-04:
Also extracting the lists to parse them takes a while when many packages are involved. I'm thinking to reduce the G_AGUP calls we do, using a flag file to tell all our scripts if/that lists are up-to-date already. Then change or disable the default list compression or enable the APT cache by default again, to speed up this tasks.

However, if it takes more than 5 minutes, it sounds like something went wrong.

manomaya90, 2021-03-05:
I got it working!!!

I think the problem wasn't with MariaDB afterall, but with the APT upgrade in the first boot. As I said before, it stuck forever (I have left it for more than 1 hour). I tried changing the repo server to the nearest one, but it didn't help much. So I bypassed it and did the APT update & upgrade manually, reboot, and the first run started. Check the kernel and it's up to date. Then Nextcloud was installed just fine without any problem.

The weird thing was when I tried to reflash the 64bit once more, I just realized that the first boot run without issue.

I got an off topic question, if I install non supported Dietpi software that utilize let's say MariaDB too, will there be any conflict?

Joulinar, 2021-03-05:
Usually it shouldn't be a problem if 2 or more applications use the same database

MichaIng, 2021-03-05:
That's the whole idea behind a database server 😃. Also from Debian Buster on we do not change the MariaDB setting away from the Debian package defaults so nothing to worry about when following any instructions or installers for Debian.

Joulinar, 2021-03-07:
I'm going to close the issue as this is working now. Feel free to reopen if needed.

closed 2021-03-07 by Joulinar