MichaIng/DietPi #4170: [rockrpro64] [7.0.2] fresh  installation terminates with botched kernel update?

St3S0f, 2021-03-04:
#### Required Information
- DietPi version | 7.0.2
- Distro version | buster
- Kernel version | 5.9.16-rockchip64
- SBC model | ROCKPro64 (aarch64)
- Power supply used | pine64 12V 5A

---

On a freshly installed 7.0.2, it seems that the initial update process leaves the machine with a botched attempt to upgrade to 5.10.16-rockchip64.

Here are the results of my analysis so far:

- dietpi-driver_manager warns of "autofs4 module not available in the kernel"
- installing and starting autofs service fails

- installing docker via optimized software fails, with a message hinting at this problem [image]

- this is the results i get on the box after a fresh install
[image]

- same hardware, same commands, running on another box (DietPi v6.34.3) installed ~1month ago, which has none of these issues
[image]

If you have any idea and don't have any HW available, I can restage / test at will so I am happy to apply and test any workarounds.

Thanks

Joulinar, 2021-03-04:
Hi,

many thanks for your message. Looks like your system did a kernel update. To take effect a reboot is required. That's the reason why Docker is not going to be installed atm. Did you have done a reboot already?

St3S0f, 2021-03-04:
Hi Joulinar, yes reboot was tried but did not solve the issue.

St3S0f, 2021-03-04:
As far as you know, Is there a way to prevent this from triggering during the initial installation?

Joulinar, 2021-03-04:
just to answer a question above

> I tried flashing with the 6.34.3 image as well but same result, looks like the upgrade happens during the initial auto installation process, not sure if I can prevent that from happening. My dietpi.txt is set with:
CONFIG_CHECK_DIETPI_UPDATES=1
CONFIG_CHECK_APT_UPDATES=

Yes correct, DietPi will update to latest version on a new installation, always

But the issue seems with the kernel installed. This happen as part of the installation process because DietPi will perform an `apt update && apt upgrade` to push all packages to latest version.

what happen if you run `apt update && apt upgrade`

St3S0f, 2021-03-04:
Thanks for the feedback, your statement pointed me into the right direction, we can close the issue, the error was on my end.
I will detail my resolution for anyone who can benefit from it.

I have flashed the SPI of my SBCs with U-Boot to be able to boot from usb.
https://github.com/ayufan-rock64/linux-build/blob/master/recipes/flash-spi.md
As rockpro64 are very picky on usb booting and are not able to boot directly from usb3 ports, in my devices the booting start from a thumb drive on usb2 that switches over mid boot to a usb3 hd, where the filesystem resides.

The resolution of my problem was to make sure the "bridge" ubs2 thumb drive was also fully upgraded to the 5.10.16 kernel, to be in sync to the ubs3 drive version. With that everything boots correctly.

closed 2021-03-04 by St3S0f

Joulinar, 2021-03-04:
Thanks for sharing your solution. Interesting setup you have 😀

St3S0f, 2021-03-04:
no sd card = peace of mind ☺️

MichaIng, 2021-03-04:
Indeed, we got just another case of likely dying SD card in the forum. However, also USB booting obviously has it's difficulties depending on SBC/kernel/bootloader 😄.