MichaIng/DietPi #4173: Config backup

cocoflan, 2021-03-05:
Config backup system would be very handy.

After installing a new system there should be the option to back up the system settings to an external usb device. if there are problems and you need to really do a reinstall, it is possible to restore this file with the settings via an external device. 

So if there is nothing more to it than a complete reinstallation (wifi, Lan, external disks mounts, ..) can be restored. 

Greetings

Joulinar, 2021-03-05:
Hi,

Many thanks for your question.  You can backup your entire system using `dietpi-backup` to an external device. As well is it possible to restore such a backup from an external device during first boot process automatically.

MichaIng, 2021-05-07:
I mark this as closed. Let's discuss additional backup and restore ideas in #4293.

closed 2021-05-07 by MichaIng