MichaIng/DietPi #4175: Chromium Kiosk Mode no longer working

xthedakmanx, 2021-03-06:
### Creating a bug report/issue

#### Required Information
- DietPi version | G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=0
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'`
- Distro version | `10.8`
- Kernel version | `Linux DietPi 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux`
- SBC model | `RPi 4 Model B (armv7l)`
- Power supply used | Offical RPI USB C PSU
- SDcard used |  SanDisk

#### Additional Information (if applicable)
- Software title | Chromium
- Was the software title installed freshly or updated/migrated?   Fresh
- Can this issue be replicated on a fresh installation of DietPi?   Yes

#### Steps to reproduce

1. Flash Dietpi
2. Install Chromium
3. Set up autoboot to a url in dietpi-config/autostart options
4. reboot

#### Expected behaviour
- Webpage displays on screen attached to PI

#### Actual behaviour
- Errors out
- 
#### Extra details
X.Org X Server logs:

`[    14.983] 
X.Org X Server 1.20.4
X Protocol Version 11, Revision 0
[    14.983] Build Operating System: Linux 5.4.0-54-generic armv8l Raspbian
[    14.984] Current Operating System: Linux DietPi 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l
[    14.984] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=1920 bcm2708_fb.fbheight=1080 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:13:E1:3D vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=tty1 root=PARTUUID=e8af6eb2-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet net.ifnames=0
[    14.984] Build Date: 15 January 2021  02:03:57PM
[    14.984] xorg-server 2:1.20.4-1+rpt2+deb10u3 (https://www.debian.org/support) 
[    14.984] Current version of pixman: 0.36.0
[    14.984] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[    14.984] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    14.985] (==) Log file: "/var/log/Xorg.0.log", Time: Fri Mar  5 18:25:38 2021
[    14.996] (==) Using config directory: "/etc/X11/xorg.conf.d"
[    14.996] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[    14.997] (==) No Layout section.  Using the first Screen section.
[    14.997] (==) No screen section available. Using defaults.
[    14.997] (**) |-->Screen "Default Screen Section" (0)
[    14.997] (**) |   |-->Monitor "<default monitor>"
[    15.001] (==) No monitor specified for screen "Default Screen Section".
	Using a default monitor configuration.
[    15.001] (**) Option "BlankTime" "0"
[    15.001] (==) Automatically adding devices
[    15.001] (==) Automatically enabling devices
[    15.001] (==) Automatically adding GPU devices
[    15.005] (==) Max clients allowed: 256, resource mask: 0x1fffff
[    15.006] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[    15.006] 	Entry deleted from font path.
[    15.006] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[    15.006] 	Entry deleted from font path.
[    15.006] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[    15.006] 	Entry deleted from font path.
[    15.006] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[    15.006] 	Entry deleted from font path.
[    15.006] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[    15.006] 	Entry deleted from font path.
[    15.006] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[    15.006] 	Entry deleted from font path.
[    15.006] (==) FontPath set to:
	/usr/share/fonts/X11/misc,
	built-ins
[    15.006] (==) ModulePath set to "/usr/lib/xorg/modules"
[    15.006] (**) Extension "DPMS" is disabled
[    15.006] (II) The server relies on udev to provide the list of input devices.
	If no devices become available, reconfigure udev or disable AutoAddDevices.
[    15.006] (II) Loader magic: 0x1fdf80
[    15.006] (II) Module ABI versions:
[    15.006] 	X.Org ANSI C Emulation: 0.4
[    15.006] 	X.Org Video Driver: 24.0
[    15.006] 	X.Org XInput driver : 24.1
[    15.006] 	X.Org Server Extension : 10.0
[    15.007] (--) using VT number 2

[    15.007] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[    15.009] (II) no primary bus or device found
[    15.009] (II) LoadModule: "glx"
[    15.010] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[    15.045] (II) Module glx: vendor="X.Org Foundation"
[    15.045] 	compiled for 1.20.4, module version = 1.0.0
[    15.046] 	ABI class: X.Org Server Extension, version 10.0
[    15.046] (==) Matched modesetting as autoconfigured driver 0
[    15.046] (==) Matched fbdev as autoconfigured driver 1
[    15.046] (==) Assigned the driver to the xf86ConfigLayout
[    15.046] (II) LoadModule: "modesetting"
[    15.047] (II) Loading /usr/lib/xorg/modules/drivers/modesetting_drv.so
[    15.050] (II) Module modesetting: vendor="X.Org Foundation"
[    15.050] 	compiled for 1.20.4, module version = 1.20.4
[    15.050] 	Module class: X.Org Video Driver
[    15.050] 	ABI class: X.Org Video Driver, version 24.0
[    15.050] (II) LoadModule: "fbdev"
[    15.050] (WW) Warning, couldn't open module fbdev
[    15.050] (EE) Failed to load module "fbdev" (module does not exist, 0)
[    15.050] (II) modesetting: Driver for Modesetting Kernel Drivers: kms
[    15.059] (WW) Falling back to old probe method for modesetting
[    15.059] (EE) open /dev/dri/card0: No such file or directory
[    15.059] (WW) Falling back to old probe method for modesetting
[    15.059] (EE) open /dev/dri/card0: No such file or directory
[    15.059] (EE) No devices detected.
[    15.059] (EE) 
Fatal server error:
[    15.059] (EE) no screens found(EE) 
[    15.059] (EE) 
Please consult the The X.Org Foundation support 
	 at http://wiki.x.org
 for help. 
[    15.059] (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
[    15.060] (EE) 
[    15.071] (EE) Server terminated with error (1). Closing log file.`

MichaIng, 2021-03-06:
Many thanks for your report.

That directory is not present at all?
```
ls -l /dev/dri
```

Can you try the following:
```
/boot/dietpi/func/dietpi-set_hardware rpi-opengl vc4-kms-v3d-pi4
reboot
```

If that does not work:
```
apt install xserver-xorg-video-fbdev
```

And if you're in mood to compare how well the alternative behaves:
```
apt puge xserver-xorg-video-fbdev
apt install xserver-xorg-video-fbturbo
```

Loader23, 2021-03-07:
Can confirm. Its broken on a fresh installation and existing setups are broken after Update 😢.
/dev/dri is also not present on 6.34.3.

Joulinar, 2021-03-07:
Hi,

I did a test an RPi4B 64bit and there Chromium is still working on kiosk mode after updating to 7.0.2. Will test 32bit now

MichaIng, 2021-03-08:
Actually the DRI device should be always available, if I'm not mistaken, so I think it's a kernel error. But with last release, we slimmed down the X server install so that no additional GPU driver is installed automatically, while before we installed most of them, including AMD, legacy ATI, VMware and fallback VESA drivers on all SBCs, which does not harm but is pretty non-required. On Odroids, dedicated fbdev or fbturbo drivers are installed, on many other SBCs we ship an xorg.conf that explicitly uses the regular modesetting driver, which is always installed as part of the X.org core package, and, compared to fallback drivers, provides a basic GPU acceleration.

On RPi, `/dev/fb0` is always available, so `xserver-xorg-video-fbdev` and `xserver-xorg-video-fbturbo` should work as well, if the DRI devices are not available. But actually they should regardless of legacy/fake KMS or full KMS GPU mode. It sounds somehow logic that the X server KMS driver requires the kernel KMS device/driver enabled, but at least on my tests the X server started up pretty without without it.

I wonder what might be the difference between 32-bit and 64-bit. @Joulinar since there is still not 64-bit/arm64 Chromium package available, does it mean the 32-bit/armhf package just installed and runs fine on RPi 64-bit (now)? There are two binaries, in `/usr/lib/chromium-browser/`, one with `-v7` suffix and one without. End of last year, the launcher script got an addition so that arm64 systems will use the one without `-v7`, suffix, which was previously only the case for ARMv6 ones. Probably that fixed it for 64-bit images, while it's still a 32-bit binary of course. Or did you install the generic `chromium` package from the Debian repository?

Joulinar, 2021-03-08:
>  @Joulinar since there is still not 64-bit/arm64 Chromium package available, does it mean the 32-bit/armhf package just installed and runs fine on RPi 64-bit (now)?

Nope, I changed to `chromium` package as `chromium-browser` is still not available on RPi OS 64bit

MichaIng, 2021-03-08:
Okay, I guess the RPi Chromium invokes the X server or GPU features internally differently. Dies `/dev/dri/card0` exist on RPi 4 with 64-bit kernel?

Since we add armhf as foreign DPkg architecture on the 64-bit image, it could be actually possible to install `chromium-browser`, but that would pull or require a bunch of 32-bit libraries.

I'll do some tests the next days, also to compare the different X/GPU drivers.

Joulinar, 2021-03-08:
not sure why, but out of curiosity I did another test on an older 64bit backup and this time it's failing as well. Not sure what is the difference to my first test now. Anyway same fix is working on 64bit also.

MichaIng, 2021-03-08:
Does starting a desktop work (without the fix)?

Joulinar, 2021-03-09:
staring the desktop is quite a challenge. If I install `LXDE` (autostart still set to  `Manual login`), the system is going to hang right after `dietpi-boot.service` completed. It just give a blinking cursor 🤔 

Same as this https://dietpi.com/phpbb/viewtopic.php?f=11&t=8766

Will need to play with it more to get more details...

MichaIng, 2021-03-09:
Okay, to not force the use of the OpenGL driver, I'll re-add the `xserver-xorg-video-fbdev` driver install for RPis.

The autostart setting is strange, as nothing is changed when cancelling the dialog, and if it's not cancelled, the autostart menu is shown but nothing automatically selected 🤔.

Joulinar, 2021-03-09:
nice behaviour of `LightDM`, good found

MichaIng, 2021-03-09:
Verified failing X server without KMS overlay enabled. I can swear I tested it before, but either I did something wrong or something changed kernel/firmware-wise. However it sounds pretty logic that fbdev is for /dev/fb0 and modesetting/KMS is for /dev/dri/card0 and that KMS X display driver requires the KMS firmware driver enabled. Fixed both issues with: https://github.com/MichaIng/DietPi/commit/7e8046c2404f37d1c03ad55729747a57ef85458f

I'm currently running tests about which driver combination works best. Sadly with full KMS, probably the best performing, the monitor blanks after a few minutes and it's not possible to wake it up again. X server seems to have crashed 😢.

kmplngj, 2021-03-25:
> Hi,
> 
> I did a test on RPi4B 64bit and there Chromium is still working on kiosk mode after updating to 7.0.2. Will test 32bit now
> 
> **EDIT:**
> Yep it's failing on 32bit, however this one fixed it
> 
> ```
> /boot/dietpi/func/dietpi-set_hardware rpi-opengl vc4-kms-v3d-pi4
> reboot
> ```

Could this also be used for a Pi Zero?

Thanks

MichaIng, 2021-03-25:
Please do instead:
```
apt install xorg-xserver-video-fbdev
```

closed 2021-04-24 by MichaIng

Joulinar, 2021-07-05:
there was a similar issue on a Odroid N+ running DietPi v7.3.2 https://dietpi.com/phpbb/viewtopic.php?f=11&t=9206

running `apt install xserver-xorg-video-fbdev` fixed it

kmplngj, 2021-10-02:
I run into the issue again with updates to 7.6.2 but I could not fix it again.

`root@DietPi:~# apt install xorg-xserver-video-fbdev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package xorg-xserver-video-fbdev`

Joulinar, 2021-10-02:
You used wrong package name `xorg-xserver-video-fbdev.` It should be `xserver-xorg-video-fbdev` instead