MichaIng/DietPi #4176: The installating for MineOs crash during installation

champymarty, 2021-03-06:
#### Details:
- Date           | Fri Mar  5 22:44:10 EST 2021
- DietPi version | v7.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | Linux DietPi 5.10.17-v7+ #1403 SMP Mon Feb 22 11:29:51 GMT 2021 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | ./update_webui.sh
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. Install mineOs with dietpi-software
2. ...
#### Expected behaviour:

-  Install MineOs
#### Actual behaviour:

- The code is crashing at npm update dependencies
#### Extra details:

- ...
#### Additional logs:
```
Script execution started on: Fri Mar  5 22:43:14 EST 2021
Updating mineos-node repository...OK
Pulling master branch...From https://github.com/hexparrot/mineos-node
 * branch            master     -> FETCH_HEAD
OK
Updating npm dependencies...npm ERR! code 1
npm ERR! path /mnt/dietpi_userdata/mineos/minecraft/node_modules/userid
npm ERR! command failed
npm ERR! command sh -c node-gyp rebuild
npm ERR! make: Entering directory '/mnt/dietpi_userdata/mineos/minecraft/node_modules/userid/build'
npm ERR!   CXX(target) Release/obj.target/userid/src/userid.o
npm ERR! make: Leaving directory '/mnt/dietpi_userdata/mineos/minecraft/node_modules/userid/build'
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@7.1.2
npm ERR! gyp info using node@15.11.0 | linux | arm
npm ERR! gyp info find Python using Python version 3.7.3 found at "/usr/bin/python3"
npm ERR! gyp info spawn /usr/bin/python3
npm ERR! gyp info spawn args [
npm ERR! gyp info spawn args   '/usr/local/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
npm ERR! gyp info spawn args   'binding.gyp',
npm ERR! gyp info spawn args   '-f',
npm ERR! gyp info spawn args   'make',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/mnt/dietpi_userdata/mineos/minecraft/node_modules/userid/build/config.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/usr/local/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/root/.cache/node-gyp/15.11.0/include/node/common.gypi',
npm ERR! gyp info spawn args   '-Dlibrary=shared_library',
npm ERR! gyp info spawn args   '-Dvisibility=default',
npm ERR! gyp info spawn args   '-Dnode_root_dir=/root/.cache/node-gyp/15.11.0',
npm ERR! gyp info spawn args   '-Dnode_gyp_dir=/usr/local/lib/node_modules/npm/node_modules/node-gyp',
npm ERR! gyp info spawn args   '-Dnode_lib_file=/root/.cache/node-gyp/15.11.0/<(target_arch)/node.lib',
npm ERR! gyp info spawn args   '-Dmodule_root_dir=/mnt/dietpi_userdata/mineos/minecraft/node_modules/userid',
npm ERR! gyp info spawn args   '-Dnode_engine=v8',
npm ERR! gyp info spawn args   '--depth=.',
npm ERR! gyp info spawn args   '--no-parallel',
npm ERR! gyp info spawn args   '--generator-output',
npm ERR! gyp info spawn args   'build',
npm ERR! gyp info spawn args   '-Goutput_dir=.'
npm ERR! gyp info spawn args ]
npm ERR! gyp info spawn make
npm ERR! gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
npm ERR! In file included from ../src/userid.cc:11:
[... 63 lines cut ...]
npm ERR!        (node::addon_register_func) (regfunc),                          \
npm ERR!                                            ^
npm ERR! /root/.cache/node-gyp/15.11.0/include/node/node.h:872:3: note: in expansion of macro ‘NODE_MODULE_X’
npm ERR!    NODE_MODULE_X(modname, regfunc, NULL, 0)  // NOLINT (readability/null_usage)
npm ERR!    ^~~~~~~~~~~~~
npm ERR! ../src/userid.cc:174:1: note: in expansion of macro ‘NODE_MODULE’
npm ERR!  NODE_MODULE(userid, Init);
npm ERR!  ^~~~~~~~~~~
npm ERR! make: *** [userid.target.mk:109: Release/obj.target/userid/src/userid.o] Error 1
npm ERR! gyp ERR! build error
npm ERR! gyp ERR! stack Error: `make` failed with exit code: 2
npm ERR! gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
npm ERR! gyp ERR! stack     at ChildProcess.emit (node:events:378:20)
npm ERR! gyp ERR! stack     at Process.ChildProcess._handle.onexit (node:internal/child_process:290:12)
npm ERR! gyp ERR! System Linux 5.10.17-v7+
npm ERR! gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
npm ERR! gyp ERR! cwd /mnt/dietpi_userdata/mineos/minecraft/node_modules/userid
npm ERR! gyp ERR! node -v v15.11.0
npm ERR! gyp ERR! node-gyp -v v7.1.2
npm ERR! gyp ERR! not ok

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-03-06T03_44_09_838Z-debug.log
FAILED
```

champymarty, 2021-03-06:
The dietpi is from a fresh install, there is only pivpn and pihole that I install before installing MineOs the same day.

Joulinar, 2021-03-06:
Hi,

many thanks for your message. I did a test installation in a RPi3B+ and MineOS is working without issues. 🤔 

are you able to attach the log file `/root/.npm/_logs/2021-03-06T03_44_09_838Z-debug.log`
Maybe best to try to attach it as full file

MichaIng, 2021-03-06:
There seem to be an empty variable that causes all these errors. I wonder if the following has to do with it:
```
warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
```
as a bug in the source code. But that all is pretty cryptic to me, so not sure if the full log helps, but doesn't hurt.

At least, I'd give it another try today where Joulinar succeeded on the exact same machine.

champymarty, 2021-03-06:
I just tried again today with no succes. Here is the log you requested. If it does not work I will probaly go with cuberite anyway :D

0 verbose cli [
0 verbose cli   '/usr/local/bin/node',
0 verbose cli   '/usr/local/bin/npm',
0 verbose cli   'install',
0 verbose cli   '--no-spin',
0 verbose cli   '--unsafe-perm'
0 verbose cli ]
1 info using npm@7.6.0
2 info using node@v15.11.0
3 timing config:load:defaults Completed in 6ms
4 timing config:load:file:/usr/local/lib/node_modules/npm/npmrc Completed in 3ms
5 timing config:load:builtin Completed in 3ms
6 timing config:load:cli Completed in 10ms
7 timing config:load:env Completed in 1ms
8 timing config:load:file:/mnt/dietpi_userdata/mineos/minecraft/.npmrc Completed in 0ms
9 timing config:load:project Completed in 3ms
10 timing config:load:file:/root/.npmrc Completed in 0ms
11 timing config:load:user Completed in 0ms
12 timing config:load:file:/usr/local/etc/npmrc Completed in 0ms
13 timing config:load:global Completed in 1ms
14 timing config:load:cafile Completed in 1ms
15 timing config:load:validate Completed in 0ms
16 timing config:load:setUserAgent Completed in 2ms
17 timing config:load:setEnvs Completed in 3ms
18 timing config:load Completed in 33ms
19 verbose npm-session bd202cd8be3128b1
20 timing npm:load Completed in 64ms
21 timing arborist:ctor Completed in 3ms
22 timing idealTree:init Completed in 1804ms
23 timing idealTree:userRequests Completed in 0ms
24 silly idealTree buildDeps
25 timing idealTree:#root Completed in 2ms
26 timing idealTree:buildDeps Completed in 6ms
27 timing idealTree:fixDepFlags Completed in 2ms
28 timing idealTree Completed in 1820ms
29 timing reify:loadTrees Completed in 1942ms
30 timing reify:diffTrees Completed in 106ms
31 silly reify moves {}
32 timing reify:retireShallow Completed in 3ms
33 timing reify:createSparse Completed in 1007ms
34 timing reify:loadBundles Completed in 0ms
35 http fetch POST 200 https://registry.npmjs.org/-/npm/v1/security/advisories/bulk 16843ms
36 timing auditReport:getReport Completed in 16876ms
37 timing reifyNode:node_modules/setimmediate Completed in 17444ms
38 timing reifyNode:node_modules/natives Completed in 17420ms
39 timing reifyNode:node_modules/cycle Completed in 17361ms
40 timing reifyNode:node_modules/copy-to Completed in 17362ms
41 timing reifyNode:node_modules/buffercursor Completed in 17349ms
42 timing reifyNode:node_modules/brace-expansion Completed in 17350ms
43 timing reifyNode:node_modules/yeast Completed in 17661ms
44 timing reifyNode:node_modules/wtf-8 Completed in 17654ms
45 timing reifyNode:node_modules/wrappy Completed in 17656ms
46 timing reifyNode:node_modules/wrap-ansi Completed in 17658ms
47 timing reifyNode:node_modules/win-release Completed in 17659ms
48 timing reifyNode:node_modules/whoami Completed in 17660ms
49 timing reifyNode:node_modules/validate-npm-package-license Completed in 17658ms
50 timing reifyNode:node_modules/unicode-length/node_modules/strip-ansi Completed in 17652ms
51 timing reifyNode:node_modules/universalify Completed in 17655ms
52 timing reifyNode:node_modules/unicode-length Completed in 17656ms
53 timing reifyNode:node_modules/unicode-length/node_modules/ansi-regex Completed in 17655ms
54 timing reifyNode:node_modules/unescape Completed in 17656ms
55 timing reifyNode:node_modules/tsame Completed in 17649ms
56 timing reifyNode:node_modules/toidentifier Completed in 17647ms
57 timing reifyNode:node_modules/to-fast-properties Completed in 17647ms
58 timing reifyNode:node_modules/tmatch Completed in 17646ms
59 timing reifyNode:node_modules/tarball-extract/node_modules/chownr Completed in 17712ms
60 timing reifyNode:node_modules/strip-bom Completed in 17740ms
61 timing reifyNode:node_modules/string-width Completed in 17735ms
[... 834 lines cut ...]
799 error In file included from ../../nan/nan.h:54,
799 error                  from ../src/userid.cc:11:
799 error ../src/userid.cc: At global scope:
799 error /root/.cache/node-gyp/15.11.0/include/node/node.h:838:43: warning: cast between incompatible function types from ‘void (*)(Nan::ADDON_REGISTER_FUNCTION_ARGS_TYPE)’ {aka ‘void (*)(v8::Local<v8::Object>)’} to ‘node::addon_register_func’ {aka ‘void (*)(v8::Local<v8::Object>, v8::Local<v8::Value>, void*)’} [-Wcast-function-type]
799 error        (node::addon_register_func) (regfunc),                          \
799 error                                            ^
799 error /root/.cache/node-gyp/15.11.0/include/node/node.h:872:3: note: in expansion of macro ‘NODE_MODULE_X’
799 error    NODE_MODULE_X(modname, regfunc, NULL, 0)  // NOLINT (readability/null_usage)
799 error    ^~~~~~~~~~~~~
799 error ../src/userid.cc:174:1: note: in expansion of macro ‘NODE_MODULE’
799 error  NODE_MODULE(userid, Init);
799 error  ^~~~~~~~~~~
799 error make: *** [userid.target.mk:109: Release/obj.target/userid/src/userid.o] Error 1
799 error gyp ERR! build error
799 error gyp ERR! stack Error: `make` failed with exit code: 2
799 error gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
799 error gyp ERR! stack     at ChildProcess.emit (node:events:378:20)
799 error gyp ERR! stack     at Process.ChildProcess._handle.onexit (node:internal/child_process:290:12)
799 error gyp ERR! System Linux 5.10.17-v7+
799 error gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
799 error gyp ERR! cwd /mnt/dietpi_userdata/mineos/minecraft/node_modules/userid
799 error gyp ERR! node -v v15.11.0
799 error gyp ERR! node-gyp -v v7.1.2
799 error gyp ERR! not ok
800 verbose exit 1

MichaIng, 2021-03-06:
It seems to be the [userid](https://www.npmjs.com/package/userid) module. Could you try to install it globally:
```
npm i -g userid
```
And if that works, manually locally:
```
cd /mnt/dietpi_userdata/mineos/minecraft
npm i --unsafe-perm userid
```

If that works, do you have a swap file enabled?
```
free
```
Probably insufficient memory.

champymarty, 2021-03-06:
installing the module glabally worked. (But mineOs i still crashing on install) . For the manually locally it just did not install userid, it crash with the same error that I get when trying to install mineOs.
0 verbose cli [
0 verbose cli   '/usr/local/bin/node',
0 verbose cli   '/usr/local/bin/npm',
0 verbose cli   'i',
0 verbose cli   '--unsafe-perm',
0 verbose cli   'userid'
0 verbose cli ]
1 info using npm@7.6.0
2 info using node@v15.11.0
3 timing config:load:defaults Completed in 6ms
4 timing config:load:file:/usr/local/lib/node_modules/npm/npmrc Completed in 2ms
5 timing config:load:builtin Completed in 3ms
6 timing config:load:cli Completed in 10ms
7 timing config:load:env Completed in 1ms
8 timing config:load:file:/mnt/dietpi_userdata/mineos/minecraft/.npmrc Completed in 1ms
9 timing config:load:project Completed in 2ms
10 timing config:load:file:/root/.npmrc Completed in 0ms
11 timing config:load:user Completed in 0ms
12 timing config:load:file:/usr/local/etc/npmrc Completed in 0ms
13 timing config:load:global Completed in 0ms
14 timing config:load:cafile Completed in 1ms
15 timing config:load:validate Completed in 0ms
16 timing config:load:setUserAgent Completed in 2ms
17 timing config:load:setEnvs Completed in 4ms
18 timing config:load Completed in 32ms
19 verbose npm-session 98072de8eb26a1f1
20 timing npm:load Completed in 64ms
21 timing arborist:ctor Completed in 3ms
22 timing idealTree:init Completed in 2734ms
23 timing idealTree:userRequests Completed in 70ms
24 silly idealTree buildDeps
25 silly fetch manifest userid@^0.3.1
26 http fetch GET 304 https://registry.npmjs.org/userid 1075ms (from cache)
27 silly placeDep ROOT userid@0.3.1 KEEP for: mineos-node@1.1.1 want: ^0.3.1
28 timing idealTree:#root Completed in 1155ms
29 timing idealTree:buildDeps Completed in 1165ms
30 timing idealTree:fixDepFlags Completed in 127ms
31 timing idealTree Completed in 4115ms
32 timing reify:loadTrees Completed in 4119ms
33 timing reify:diffTrees Completed in 85ms
34 silly reify moves {}
35 timing reify:retireShallow Completed in 1ms
36 timing reify:createSparse Completed in 451ms
37 timing reify:loadBundles Completed in 0ms
38 http fetch POST 200 https://registry.npmjs.org/-/npm/v1/security/advisories/bulk 13289ms
39 timing auditReport:getReport Completed in 13327ms
40 http fetch GET 200 https://registry.npmjs.org/diff 4259ms (from cache)
41 timing metavuln:packument:diff Completed in 4269ms
42 timing metavuln:cache:get:security-advisory:hoek:qPgThZvvjHHtWa7NcVWYnodOY6URJw+eilXmZLf03+oIUkZKnGqvzkxlz5zwS3FYp3fg711ixdgtj9xbcQZEWw== Completed in 4498ms
43 timing metavuln:cache:get:security-advisory:debug:c/EpWYcLqpggMauDEqU7tsEO/tHyEILe760kXcHD/9HAZjVLK249B6/U6G+d9ROTtaLwElI+G0NkpJJoLNd3sQ== Completed in 4470ms
44 timing metavuln:cache:get:security-advisory:fstream:WlvA7Kz/d36oaERYo+bgP0nK63YpsjHNT5O1EYOj7PyVzYxaSuBzDaGc4uEuA6fpyeV7J0PVJF3CQtjP3brzQA== Completed in 4466ms
45 timing metavuln:cache:get:security-advisory:parsejson:G3AW2RwCugvXOS2cwvQufj0oSNteT3HpwxoBPobrS367lRlQLWgut6W9TaO8Y3t6d+qUb8M9q5aDTLPUCS97PA== Completed in 4463ms
46 timing metavuln:cache:get:security-advisory:angular:Osx0otUkii9Ym2tEoyKQorqnJWJxtTFnzDAYG6qN0JVouWT5V30VbopdtLR6McJx1v7mCmQC9JCBDdgnaFWvyw== Completed in 4453ms
47 timing metavuln:cache:get:security-advisory:ini:Ycq69/gaCzr/G/fbN43rDLIsHS7IdZHzW5bh0ju4uwHwvENyyzefC75bnxzsKQ1M+jh/75af+MnibiJLXd61Iw== Completed in 4444ms
48 timing metavuln:cache:get:security-advisory:angular:Z8xmzCdHACzAzHPJRUILOZkayfSnN1XjzANGlZ6cazuL8WtCVYldDrWX+4MoRazvBozBmRSOy/koEw5D5twHPg== Completed in 4446ms
49 timing metavuln:cache:get:security-advisory:socket.io:2WgpQhoJaxuyRBUPkqlZPU0ipBVCKm9S3IkR2kSHIleGRJIxAABTsaywJwkMG0EL1t9NkMTM38adGigY7dQcyQ== Completed in 4431ms
50 timing metavuln:cache:get:security-advisory:diff:jyaxcXiG/Si8JbPB2uQQNPirCQZoIkPG8XC7ZRlFapI+98qR4TIu+wxwa3doVrcC7BqSmr7bQ27QSCxD85CNag== Completed in 4419ms
51 timing metavuln:load:security-advisory:diff:1631 Completed in 6ms
52 timing metavuln:calculate:security-advisory:diff:1631 Completed in 4425ms
53 timing reifyNode:node_modules/setimmediate Completed in 17846ms
54 timing reifyNode:node_modules/natives Completed in 17814ms
55 timing reifyNode:node_modules/cycle Completed in 17737ms
56 timing reifyNode:node_modules/copy-to Completed in 17737ms
57 timing reifyNode:node_modules/buffercursor Completed in 17721ms
58 timing reifyNode:node_modules/brace-expansion Completed in 17727ms
59 timing reifyNode:node_modules/yeast Completed in 18086ms
60 timing reifyNode:node_modules/wtf-8 Completed in 18080ms
61 timing reifyNode:node_modules/wrappy Completed in 18079ms
62 timing reifyNode:node_modules/wrap-ansi Completed in 18082ms
[... 841 lines cut ...]
802 error                                            ^
802 error /root/.cache/node-gyp/15.11.0/include/node/node.h:872:3: note: in expansion of macro ‘NODE_MODULE_X’
802 error    NODE_MODULE_X(modname, regfunc, NULL, 0)  // NOLINT (readability/null_usage)
802 error    ^~~~~~~~~~~~~
802 error ../src/userid.cc:174:1: note: in expansion of macro ‘NODE_MODULE’
802 error  NODE_MODULE(userid, Init);
802 error  ^~~~~~~~~~~
802 error make: *** [userid.target.mk:109: Release/obj.target/userid/src/userid.o] Error 1
802 error gyp ERR! build error
802 error gyp ERR! stack Error: `make` failed with exit code: 2
802 error gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
[... cut ...]
802 error gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
802 error gyp ERR! cwd /mnt/dietpi_userdata/mineos/minecraft/node_modules/userid
802 error gyp ERR! node -v v15.11.0
802 error gyp ERR! node-gyp -v v7.1.2
802 error gyp ERR! not ok
803 verbose exit 1

With the command free I get this:
              total        used        free      shared  buff/cache   available
Mem:         997124       42476      505128       19312      449520      872604
Swap:       1099772           0     1099772

MichaIng, 2021-03-06:
Very strange. Could you try again after reinstalling Node:
```
dietpi-software reinstall 9
```

champymarty, 2021-03-06:
still the same :(

MichaIng, 2021-03-06:
Strange. Can you check `dmesg -l emerg,alert,crit,err` for kernel errors? It's a new install of MineOS, right? Probably something gotten messed with that local Node modules, so let's try it freshly by removing all related files:
```
sed -i '/aSOFTWARE_INSTALL_STATE\[53\]=/c\aSOFTWARE_INSTALL_STATE\[53\]=2' /boot/dietpi/.installed
dietpi-software uninstall 53 # Cheat that it's installed to allow uninstalling it, which removes everything cleanly
dietpi-software install 53
```

champymarty, 2021-03-07:
It worked ! Thanks you !

MichaIng, 2021-03-07:
Great! Good to keep in in mind, if a Node build fails, to remove everything cleanly before retrying 🙂.

Joulinar, 2021-03-10:
Guys I'm going to close this topic, feel free to reopen of needed

closed 2021-03-10 by Joulinar