MichaIng/DietPi #4177: NTP server not updating anymore after recent update

burstina, 2021-03-06:
Hello guys,
Yesterday I did a dietpi-update because the tty login banner told me there was a brand new version of DietPi available.
It completed pretty fine. I've to say I use this little device as a 4G LTE "modem" for my home network and  and main functionality has not been compromised, that made me happy.

Nevertheless, I noticed in journalctl some warning about ntp sync

```
Mar 06 09:00:07 DietPi systemd-timesyncd[15500]: Timed out waiting for reply from 194.0.5.123:123 (3.pool.ntp.org).
Mar 06 09:00:18 DietPi systemd-timesyncd[15500]: Timed out waiting for reply from 37.247.53.178:123 (3.pool.ntp.org).
Mar 06 09:00:28 DietPi systemd-timesyncd[15500]: Timed out waiting for reply from 188.213.165.209:123 (3.pool.ntp.org).
Mar 06 09:00:38 DietPi systemd-timesyncd[15500]: Timed out waiting for reply from 62.94.33.173:123 (3.pool.ntp.org).
Mar 06 09:05:05 DietPi systemd-timesyncd[15500]: Timed out waiting for reply from 212.45.144.3:123 (0.pool.ntp.org).
Mar 06 09:05:15 DietPi systemd-timesyncd[15500]: Timed out waiting for reply from 185.19.184.35:123 (0.pool.ntp.org).
Mar 06 09:05:25 DietPi systemd-timesyncd[15500]: Timed out waiting for reply from 31.14.133.122:123 (0.pool.ntp.org).
Mar 06 09:05:35 DietPi systemd-timesyncd[15500]: Timed out waiting for reply from 80.182.170.49:123 (0.pool.ntp.org).
```

So I tried to change NTP server url using Webmin web interface, but webpage crashed and webmin wepage doesn't is not working anymore (connection timeout)

I tried then DIETPI-CONFIG, again trying to change NTP server, but looks like each ntp server I try cannot be reached.
It feels wrong, because ping to those servers works fine.

Is it a bug or something I can somehow work out?

thanks
Graziano

### Creating a bug report/issue
[  OK  ] DietPi-Config | Setting in /boot/dietpi.txt adjusted: CONFIG_NTP_MIRROR=pool.ntp.org
[ SUB1 ] DietPi-Set_software > ntpd-mode (2)
[  OK  ] DietPi-Set_software | Setting in /boot/dietpi.txt adjusted: CONFIG_NTP_MODE=2
[  OK  ] ntpd-mode 2 | Completed
[  OK  ] DietPi-Run_NTPD | systemctl restart systemd-timesyncd
[ INFO ] DietPi-Run_NTPD | Waiting for completion of systemd-timesyncd (1/10)
[ INFO ] DietPi-Run_NTPD | Waiting for completion of systemd-timesyncd (2/10)
[ INFO ] DietPi-Run_NTPD | Waiting for completion of systemd-timesyncd (3/10)
[ INFO ] DietPi-Run_NTPD | Waiting for completion of systemd-timesyncd (4/10)
[ INFO ] DietPi-Run_NTPD | Waiting for completion of systemd-timesyncd (5/10)
[ INFO ] DietPi-Run_NTPD | Waiting for completion of systemd-timesyncd (6/10)
[ INFO ] DietPi-Run_NTPD | Waiting for completion of systemd-timesyncd (7/10)
[ INFO ] DietPi-Run_NTPD | Waiting for completion of systemd-timesyncd (8/10)
[ I...............

#### Required Information
```
root@DietPi:~#  dietpi-bugreport
[  OK  ] DietPi-Bugreport | Packing upload archive
[  OK  ] DietPi-Bugreport | Sending bug report
[ INFO ] DietPi-Bugreport | Reference code: 47c5fb60-f112-41fa-af6f-9b1a3de4942d
root@DietPi:~# cat /boot/dietpi/.version
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=0
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
root@DietPi:~# echo $G_DISTRO_NAME` or `cat /etc/debian_version
bash: or: command not found
bustercat /etc/debian_version
root@DietPi:~# uname -a
Linux DietPi 5.10.16-rockchip64 #21.02.2 SMP PREEMPT Sun Feb 14 21:35:01 CET 2021 aarch64 GNU/Linux
root@DietPi:~# echo $G_HW_MODEL_NAME
NanoPi NEO3 (aarch64)

```

Joulinar, 2021-03-06:
Hi, 

Many thanks for your message. Do you have an internet router that is providing NTP server functionality as well? Maybe you can use this one. As well `ping` is using a different port that network time protocol. Maybe NTP get blocked somewhere. Do you have some firewall or package filter in your local network?

burstina, 2021-03-06:
hello,
Network client machines, such as Windows 10 pc, do update date and time thru internet perfectly. 
On dietpi machine, Webmin system time interface gives me this error : `NTP time synchronization failed : Missing ntpdate and sntp commands`
Since network PC cliend does perfectly update via ntp, it becomes unnecessary to specify that firewall, via iptables, doesn't have any rule to block outgoing packets.

MichaIng, 2021-03-06:
Note that DietPi uses the always installed [systemd-timesyncd](https://manpages.debian.org/systemd-timesyncd) for network time sync. ntp, ntpdate, sntp, chrony and all those are optional other time sync daemons, required only if you have specific needs, require high precision etc. Would be worth the send a request to webmin for systemd-timesyncd support. It's really an overhead if users are forced to install an additional daemon while there is already one. It's if you install an additional program on Windows to do that, while it's internal clock has an internal time sync included, although in case of Windows a bad one since it forcefully syncs to time.windows.com 😄.

However, if `systemd-timesyncd` does not work well for whatever reason, a different client could be tested. [ntp](https://packages.debian.org/ntp) is a larger overhead, since it is not only a client, but a local NTP server as well. [ntpdate](https://packages.debian.org/ntpdate) is not maintained and officially marked as deprecated by Debian. [htpdate](https://packages.debian.org/htpdate) syncs via HTTP from HTTP servers, which means protocol overhead and less precision naturally, so the only use case for it is if NTP/port 123 is blocked by the firewall but HTTP/port 80 is not. Then there is [chrony](https://packages.debian.org/chrony) which syncs time with other systems in the network. And many more. To use e.g. `ntp` you need to disable DietPi's time sync feature, which would otherwise always time out, since systemd-timesyncd automatically denies startup when it detects any other installed NTP daemon:
```
/boot/dietpi/func/dietpi-set_software ntpd-mode 0
apt install ntp
```
______________
Lately in my case it takes exactly 30 seconds on all VMs and even on a WiFi-attached laptop:
```
2021-03-06 14:51:51 root@VM-Buster:~# journalctl -u dietpi-boot -u systemd-timesyncd
-- Logs begin at Sat 2021-03-06 14:50:28 CET, end at Sat 2021-03-06 14:51:20 CET. --
Mär 06 14:50:29 VM-Buster systemd[1]: Starting DietPi-Boot...
Mär 06 14:50:29 VM-Buster DietPi-Boot[190]: [ .... ] DietPi-Run_NTPD | systemctl restart systemd-timesyncd
Mär 06 14:50:29 VM-Buster systemd[1]: Starting Network Time Synchronization...
Mär 06 14:50:37 VM-Buster systemd[1]: Started Network Time Synchronization.
Mär 06 14:50:37 VM-Buster DietPi-Boot[190]: [  OK  ] DietPi-Run_NTPD | systemctl restart systemd-timesyncd
Mär 06 14:50:37 VM-Buster DietPi-Boot[190]: [ INFO ] DietPi-Run_NTPD | Waiting for completion of systemd-timesyncd (1/60)
Mär 06 14:50:38 VM-Buster DietPi-Boot[190]: [ INFO ] DietPi-Run_NTPD | Waiting for completion of systemd-timesyncd (2/60)
...
Mär 06 14:51:06 VM-Buster DietPi-Boot[190]: [ INFO ] DietPi-Run_NTPD | Waiting for completion of systemd-timesyncd (29/60)
Mär 06 14:51:07 VM-Buster DietPi-Boot[190]: [ INFO ] DietPi-Run_NTPD | Waiting for completion of systemd-timesyncd (30/60)
Mär 06 14:51:07 VM-Buster systemd-timesyncd[199]: Synchronized to time server for the first time [2a01:4f8:200:3296::2]:123 (2.debian.pool.ntp.org).
Mär 06 14:51:08 VM-Buster DietPi-Boot[190]: [  OK  ] DietPi-Run_NTPD | systemd-timesyncd synced
Mär 06 14:51:08 VM-Buster systemd[1]: Stopping Network Time Synchronization...
Mär 06 14:51:08 VM-Buster systemd[1]: systemd-timesyncd.service: Succeeded.
Mär 06 14:51:08 VM-Buster systemd[1]: Stopped Network Time Synchronization.
Mär 06 14:51:08 VM-Buster DietPi-Boot[190]: [  OK  ] Network time sync | Completed
Mär 06 14:51:08 VM-Buster systemd[1]: Started DietPi-Boot.
```
```
2021-03-06 14:55:45 root@emachines:~# journalctl -u dietpi-boot -u systemd-timesyncd
Journal file /var/log/journal/8d6ad3de532c43bd9a4c9519df1bae58/system.journal is truncated, ignoring file.
-- Journal begins at Sat 2021-03-06 14:52:50 CET, ends at Sat 2021-03-06 14:55:44 CET. --
Mar 06 14:52:52 emachines systemd[1]: Starting DietPi-Boot...
Mar 06 14:52:52 emachines DietPi-Boot[221]: [ .... ] DietPi-Run_NTPD | systemctl restart systemd-timesyncd
Mar 06 14:52:52 emachines systemd[1]: Starting Network Time Synchronization...
Mar 06 14:52:53 emachines systemd[1]: Started Network Time Synchronization.
Mar 06 14:52:53 emachines DietPi-Boot[221]: [  OK  ] DietPi-Run_NTPD | systemctl restart systemd-timesyncd
Mar 06 14:52:53 emachines DietPi-Boot[221]: [ INFO ] DietPi-Run_NTPD | Waiting for completion of systemd-timesyncd (1/60)
Mar 06 14:52:54 emachines DietPi-Boot[221]: [ INFO ] DietPi-Run_NTPD | Waiting for completion of systemd-timesyncd (2/60)
...
Mar 06 14:53:21 emachines DietPi-Boot[221]: [ INFO ] DietPi-Run_NTPD | Waiting for completion of systemd-timesyncd (28/60)
Mar 06 14:53:22 emachines DietPi-Boot[221]: [ INFO ] DietPi-Run_NTPD | Waiting for completion of systemd-timesyncd (29/60)
Mar 06 14:53:24 emachines systemd-timesyncd[247]: Initial synchronization to time server [2a01:4f8:110:12c8::123]:123 (2.debian.pool.ntp.org).
Mar 06 14:53:24 emachines DietPi-Boot[221]: [  OK  ] DietPi-Run_NTPD | systemd-timesyncd synced
Mar 06 14:53:24 emachines systemd[1]: Stopping Network Time Synchronization...
Mar 06 14:53:24 emachines systemd[1]: systemd-timesyncd.service: Succeeded.
Mar 06 14:53:24 emachines systemd[1]: Stopped Network Time Synchronization.
Mar 06 14:53:24 emachines DietPi-Boot[221]: [  OK  ] Network time sync | Completed
Mar 06 14:53:24 emachines systemd[1]: Finished DietPi-Boot.
```
[... cut ...] ssue.

I tried `ntp`, hacked it into our time sync script and it took always less than 15 seconds, with and without that egoist burst/iburst options on debian.pool.ntp.org.

Strange, as not long ago the sync finished in mostly under 10 seconds via systemd-timesyncd as well. To rule it out, I tested it on a Stretch system where no package upgrade happened for two years, but it exactly too `(30/60)` as well. The emachines notebook btw is a Bullseye system, so it's on all supported Debian versions.

burstina, 2021-03-06:
```
mar 06 17:12:15 graz-ubuntu systemd[1]: Stopping Network Time Synchronization...
mar 06 17:12:15 graz-ubuntu systemd[1]: systemd-timesyncd.service: Succeeded.
mar 06 17:12:15 graz-ubuntu systemd[1]: Stopped Network Time Synchronization.
mar 06 17:12:15 graz-ubuntu systemd[1]: Starting Network Time Synchronization...
mar 06 17:12:15 graz-ubuntu systemd[1]: Started Network Time Synchronization.
mar 06 17:12:15 graz-ubuntu systemd-timesyncd[46440]: Initial synchronization to time server 91.189.94.4:123 (ntp.ubuntu.com).
```

Perfectly fine on Ubuntu machine in the same network as Dietpi machine.
No 30/60 secs deley

MichaIng, 2021-03-06:
On a service restart it syncs much faster as well on DietPi, mostly in a second, the issue is only on the very first start at boot. However, which Ubuntu and package version is it?
```
dpkg -l | grep systemd
```
I'll test `ntp.ubuntu.com`. Interesting that it's a single server only, no pool.

burstina, 2021-03-06:
>  
> On a service restart it syncs much faster as well on DietPi, mostly in a second, the issue is only on the very first start at boot [...]

Not really, I just send `systemctl restart systemd-timesyncd` manually and `journalctl `shows:

```
Mar 06 19:14:34 DietPi systemd[1]: Stopping Network Time Synchronization...
Mar 06 19:14:34 DietPi systemd[1]: systemd-timesyncd.service: Succeeded.
Mar 06 19:14:35 DietPi systemd[1]: Stopped Network Time Synchronization.
Mar 06 19:14:35 DietPi systemd[1]: Starting Network Time Synchronization...
Mar 06 19:14:35 DietPi systemd[1]: Started Network Time Synchronization.
Mar 06 19:14:45 DietPi systemd-timesyncd[10790]: Timed out waiting for reply from 51.38.27.129:123 (0.pool.ntp.org).
Mar 06 19:14:55 DietPi systemd-timesyncd[10790]: Timed out waiting for reply from 95.110.248.206:123 (0.pool.ntp.org).
Mar 06 19:15:05 DietPi systemd-timesyncd[10790]: Timed out waiting for reply from 31.14.133.122:123 (0.pool.ntp.org).
Mar 06 19:15:15 DietPi systemd-timesyncd[10790]: Timed out waiting for reply from 80.211.178.99:123 (0.pool.ntp.org).
...

```

MichaIng, 2021-03-06:
On your DietPi, there is another issue. Could you try to use either "default" or "ntp.ubuntu.com" as NTP mirror on DietPi? All I see is `[0123].pool.ntp.org`.

And did you try to use `ntp`?

I'll install the Ubuntu systemd-timesyncd version to see if that makes a difference 👍.

Joulinar, 2021-03-06:
for me this looks like ntp request somewhere blocked on network level. My test system doesn't have an issue at all. Doesn't matter if it is a reboot or a service restart

```
root@DietPi4:~# journalctl -u systemd-timesyncd.service
-- Logs begin at Sat 2021-03-06 19:35:56 CET, end at Sat 2021-03-06 19:37:09 CET. --
Mar 06 19:36:03 DietPi4 systemd[1]: Starting Network Time Synchronization...
Mar 06 19:36:04 DietPi4 systemd[1]: Started Network Time Synchronization.
Mar 06 19:36:18 DietPi4 systemd-timesyncd[452]: Synchronized to time server for the first time 195.201.19.162:123 (0.pool.ntp.org).
Mar 06 19:36:19 DietPi4 systemd[1]: Stopping Network Time Synchronization...
Mar 06 19:36:19 DietPi4 systemd[1]: systemd-timesyncd.service: Succeeded.
Mar 06 19:36:19 DietPi4 systemd[1]: Stopped Network Time Synchronization.
Mar 06 19:37:09 DietPi4 systemd[1]: Starting Network Time Synchronization...
Mar 06 19:37:09 DietPi4 systemd[1]: Started Network Time Synchronization.
Mar 06 19:37:09 DietPi4 systemd-timesyncd[1115]: Synchronized to time server for the first time 78.47.158.133:123 (0.pool.ntp.org).
root@DietPi4:~#
```

MichaIng, 2021-03-06:
Dammit, none of the Ubuntu packages can work on Debian, either the systemd version itself does not match, or the libc6 version. Overriding the dependencies leads to internal function failures.

Outstanding the test if NTP/port 123 packages are somehow lost on that system in general (test [ntp](https://github.com/MichaIng/DietPi/issues/4177#issuecomment-791969576)) or if it's something related with the daemon itself, similar like the mysterious 30 seconds delay on all my machines.
_______
Btw, I disabled time sync at boot and ran it manually after login, and it synced in the first loop second. I disabled our check and enabled systemd-timesyncd natively to see how long it takes when it is not manually (re)started by our script:
```
Mar 06 20:00:25 emachines systemd[1]: Starting Network Time Synchronization...
Mar 06 20:00:25 emachines systemd[1]: Started Network Time Synchronization.
Mar 06 20:00:58 emachines systemd-timesyncd[163]: Initial synchronization to time server 213.209.109.45:123 (2.debian.pool.ntp.org).
```
Same 30 seconds (the additional 3 seconds are systemctl call finishing, entering the loop and the additional 1 second after the last "Waiting for" print). So our scripts are unrelated.

burstina, 2021-03-06:
> 
> 
> On your DietPi, there is another issue. Could you try to use either "default" or "ntp.ubuntu.com" as NTP mirror on DietPi? All I see is `[0123].pool.ntp.org`.

```
 OK  ] DietPi-Config | Setting in /boot/dietpi.txt adjusted: CONFIG_NTP_MIRROR=Default
[ SUB1 ] DietPi-Set_software > ntpd-mode (2)
[  OK  ] DietPi-Set_software | Setting in /boot/dietpi.txt adjusted: CONFIG_NTP_MODE=2
[  OK  ] ntpd-mode 2 | Completed
[  OK  ] DietPi-Run_NTPD | systemctl restart systemd-timesyncd
[ INFO ] DietPi-Run_NTPD | Waiting for completion of systemd-timesyncd (1/10)
[ INFO ] DietPi-Run_NTPD | Waiting for completion of systemd-timesyncd (2/10)
[ INFO ] DietPi-Run_NTPD | Waiting for completion of systemd-timesyncd (3/10)
[ INFO ] DietPi-Run_NTPD | Waiting for completion of systemd-timesyncd (4/10)
[ INFO ] DietPi-Run_NTPD | Waiting for completion of systemd-timesyncd (5/10)
[ INFO ] DietPi-Run_NTPD | Waiting for completion of systemd-timesyncd (6/10)
[ INFO ] DietPi-Run_NTPD | Waiting for completion of systemd-timesyncd (7/10)

```
and JOURNALCTL

```
Mar 06 20:32:00 DietPi systemd-fstab-generator[12378]: Failed to create unit file /run/systemd/generator/tmp.mount, as it already exists. Duplicate entry in /etc/fstab?
Mar 06 20:32:00 DietPi systemd[1]: /lib/systemd/system/xrdp.service:8: PIDFile= references path below legacy directory /var/run/, updating /var/run/xrdp/xrdp.pid → /run/xrdp/xrdp.pid; please update the unit file accordingly.
Mar 06 20:32:00 DietPi systemd[1]: /lib/systemd/system/xrdp-sesman.service:9: PIDFile= references path below legacy directory /var/run/, updating /var/run/xrdp/xrdp-sesman.pid → /run/xrdp/xrdp-sesman.pid; please update the unit file accordingly.
Mar 06 20:32:00 DietPi systemd[1]: /lib/systemd/system/haveged.service:19: Unknown lvalue 'ProtectHostname' in section 'Service', ignoring
Mar 06 20:32:00 DietPi systemd[1]: /lib/systemd/system/haveged.service:20: Unknown lvalue 'ProtectKernelLogs' in section 'Service', ignoring
Mar 06 20:32:00 DietPi systemd[1]: Starting Network Time Synchronization...
Mar 06 20:32:01 DietPi systemd[1]: Started Network Time Synchronization.
Mar 06 20:32:11 DietPi systemd-timesyncd[12412]: Timed out waiting for reply from 162.159.200.123:123 (0.debian.pool.ntp.org).
Mar 06 20:32:21 DietPi systemd-timesyncd[12412]: Timed out waiting for reply from 85.199.214.99:123 (0.debian.pool.ntp.org).
Mar 06 20:32:31 DietPi systemd-timesyncd[12412]: Timed out waiting for reply from 194.0.5.123:123 (0.debian.pool.ntp.org).
Mar 06 20:32:41 DietPi systemd-timesyncd[12412]: Timed out waiting for reply from 213.251.52.250:123 (0.debian.pool.ntp.org).
Mar 06 20:32:52 DietPi systemd-timesyncd[12412]: Timed out waiting for reply from 162.159.200.123:123 (1.debian.pool.ntp.org).

```

> 
> And did you try to use `ntp`?

Not yet, I'd prefere stick to default as much as possible

burstina, 2021-03-06:
> 
> 
> for me this looks like ntp request somewhere blocked on network level. My test system doesn't have an issue at all. Doesn't matter if it is a reboot or a service restart

Dietpi is the network router, as it connects my home network to WAN.
So, if Ubuntu and Windows, who are clients, are able to sync time via NTP, it is unthinkable that Dietpi, who is the router, is blocked by some (outgoing) firewall rules. Am I wrong?

MichaIng, 2021-03-06:
> Dietpi is the network router,
> ...
> it is unthinkable that Dietpi, who is the router, is blocked by some (outgoing) firewall rules

True then, but probably some routing of those requests goes wrong. Actually, being the router, `ntp` would have the benefit that your network clients could use it as NTP server, in theory, Windows e.g. has time.windows.com hardcoded. However, I can only encourage you to test it so that we can narrow down the issue.

Another idea would be to downgrade the systemd packages or upgrade them to Buster backports. I'll tell you how in 30 minutes.

Joulinar, 2021-03-06:
Maybe we could use `tcpdump` to check where packages are routed to.

MichaIng, 2021-03-06:
Good idea: `apt install tcpdump; tcpdump 'port 123'` on a second SSH session could do it.

> For all other clients, Pihole is distributing the NTP server option via DHCP 😉

Not sure if this is working on macOS as well, but that NTP mirrors are taken via DHCP is how it should be 👍.

Joulinar, 2021-03-06:
> Not sure if this is working on macOS as well, but that NTP mirrors are taken via DHCP is how it should be

Honestly I hove no clue about macOS 🤣 

That's how an DHCP request looks like on my network

```
root@DietPi4:~# nmap --script broadcast-dhcp-discover
Starting Nmap 7.70 ( https://nmap.org ) at 2021-03-06 21:22 CET
Pre-scan script results:
| broadcast-dhcp-discover:
|   Response 1 of 1:
|     IP Offered: 192.168.0.119
|     DHCP Message Type: DHCPOFFER
|     Server Identifier: 192.168.0.11
|     IP Address Lease Time: 2m00s
|     Renewal Time Value: 1m00s
|     Rebinding Time Value: 1m45s
|     Subnet Mask: 255.255.255.0
|     Broadcast Address: 192.168.0.255
|     Domain Name Server: 192.168.0.11
|     Domain Name: lan
|     NTP Servers: 192.168.0.1
|_    Router: 192.168.0.1
WARNING: No targets were specified, so 0 hosts scanned.
Nmap done: 0 IP addresses (0 hosts up) scanned in 4.03 seconds
root@DietPi4:~#
```

> apt install tcpdump;

Uhh that's where we have `tcpdump` available via `dietpi-software install 15` 😉

MichaIng, 2021-03-06:
> Uhh that's where we have tcpdump available via dietpi-software install 15 😉

Ah I forgot about that 😄, however it does nothing else than `apt install tcpdump` anyway.

Okay, not trivial to downgrade systemd, as the old packages are not available anymore, so the alternative would be to upgrade to backports:
```
apt install -t buster-backports systemd
```
This will install a dedicated systemd-timesyncd package, which has been split off.

burstina, 2021-03-06:
> Good idea: `apt install tcpdump; tcpdump 'port 123'` on a second SSH session could do it.
> 
> > For all other clients, Pihole is distributing the NTP server option via DHCP 😉
> 
> Not sure if this is working on macOS as well, but that NTP mirrors are taken via DHCP is how it should be 👍.
```

oot@DietPi:~# tcpdump port 123
tcpdump: verbose output suppressed, use -v or -
vv for full protocol decode
listening on eth0, link-type EN10MB (Ethernet),
 capture size 262144 bytes
23:26:38.418734 IP 192.168.0.209.56921 > pugot.canonical.com.ntp: NTPv4, Client, length 48
23:26:38.473999 IP pugot.canonical.com.ntp > 192.168.0.209.56921: NTPv4, Server, length 48
```
sbouldn't listen on WWAN0?

Joulinar, 2021-03-06:
use this one, to listen on all interfaces. It will stop after 500 lines captured

`tcpdump -i any -c500 -nn port 123`

burstina, 2021-03-07:
```
root@DietPi:~# tcpdump -i any -c500 -nn port 123
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on any, link-type LINUX_SLL (Linux cooked), capture size 262144 bytes
12:53:12.426698 IP 10.10.31.13.36524 > 213.251.52.250.123: NTPv4, Client, length 48
12:53:12.463393 IP 213.251.52.250.123 > 10.10.31.13.36524: NTPv4, Server, length 48
12:53:22.501460 IP 10.10.31.13.43734 > 212.237.31.130.123: NTPv4, Client, length 48
12:53:22.553987 IP 212.237.31.130.123 > 10.10.31.13.43734: NTPv4, Server, length 48
12:53:32.748776 IP 10.10.31.13.49277 > 93.41.196.243.123: NTPv4, Client, length 48
12:53:32.788797 IP 93.41.196.243.123 > 10.10.31.13.49277: NTPv4, Server, length 48
12:53:42.999171 IP 10.10.31.13.39129 > 31.207.113.74.123: NTPv4, Client, length 48
12:53:43.051526 IP 31.207.113.74.123 > 10.10.31.13.39129: NTPv4, Server, length 48
12:53:53.294875 IP 10.10.31.13.56485 > 162.159.200.1.123: NTPv4, Client, length 48
12:53:53.353288 IP 162.159.200.1.123 > 10.10.31.13.56485: NTPv4, Server, length 48
12:54:03.499622 IP 10.10.31.13.39005 > 95.110.248.206.123: NTPv4, Client, length 48
12:54:03.535141 IP 95.110.248.206.123 > 10.10.31.13.39005: NTPv4, Server, length 48
12:54:13.749245 IP 10.10.31.13.37821 > 185.19.184.35.123: NTPv4, Client, length 48
12:54:13.789789 IP 185.19.184.35.123 > 10.10.31.13.37821: NTPv4, Server, length 48
```

by the way, if of any help:

```
root@DietPi:~# iptables -L
Chain INPUT (policy DROP)
target     prot opt source               destination
ACCEPT     all  --  anywhere             anywhere
ACCEPT     tcp  --  anywhere             anywhere             tcp flags:ACK/ACK
ACCEPT     udp  --  anywhere             anywhere             udp spt:domain dpts:1024:65535
ACCEPT     icmp --  anywhere             anywhere             icmp echo-reply
ACCEPT     icmp --  anywhere             anywhere             icmp destination-unreachable
ACCEPT     icmp --  anywhere             anywhere             icmp source-quench
ACCEPT     icmp --  anywhere             anywhere             icmp time-exceeded
ACCEPT     icmp --  anywhere             anywhere             icmp parameter-problem

Chain FORWARD (policy ACCEPT)
target     prot opt source               destination

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination
root@DietPi:~#

```

Joulinar, 2021-03-07:
hmm looks like communication with NTP server is working as the request get's answered quite quickly

```
12:53:12.426698 IP 10.10.31.13.36524 > 213.251.52.250.123: NTPv4, Client, length 48
12:53:12.463393 IP 213.251.52.250.123 > 10.10.31.13.36524: NTPv4, Server, length 48
```

IP `10.10.31.13` is your WAN interface?

burstina, 2021-03-08:
here I am.
```
root@DietPi:~# tcpdump -i any -c500 -nn port 123
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on any, link-type LINUX_SLL (Linux cooked), capture size 262144 bytes
09:58:44.429155 IP 192.168.0.209.123 > 17.253.52.253.123: NTPv4, Client, length 48
09:58:44.429212 IP 10.18.47.221.123 > 17.253.52.253.123: NTPv4, Client, length 48
09:58:44.478867 IP 17.253.52.253.123 > 10.18.47.221.123: NTPv4, Server, length 48
09:58:44.478959 IP 17.253.52.253.123 > 192.168.0.209.123: NTPv4, Server, length 48
09:58:44.482111 IP 192.168.0.209.123 > 17.253.54.253.123: NTPv4, Client, length 48
09:58:44.482197 IP 10.18.47.221.123 > 17.253.54.253.123: NTPv4, Client, length 48
09:58:44.522865 IP 17.253.54.253.123 > 10.18.47.221.123: NTPv4, Server, length 48
09:58:44.522924 IP 17.253.54.253.123 > 192.168.0.209.123: NTPv4, Server, length 48
09:58:44.526094 IP 192.168.0.209.123 > 17.253.54.123.123: NTPv4, Client, length 48
09:58:44.526173 IP 10.18.47.221.123 > 17.253.54.123.123: NTPv4, Client, length 48
09:58:44.562877 IP 17.253.54.123.123 > 10.18.47.221.123: NTPv4, Server, length 48
09:58:44.562940 IP 17.253.54.123.123 > 192.168.0.209.123: NTPv4, Server, length 48
```

and 

```
root@DietPi:~# ip link
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP mode DEFAULT group default qlen 1000
    link/ether 1e:46:36:b5:46:27 brd ff:ff:ff:ff:ff:ff
3: wwan0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UNKNOWN mode DEFAULT group default qlen 1000
    link/none
root@DietPi:~# ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 1e:46:36:b5:46:27 brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.1/24 brd 192.168.0.255 scope global eth0
       valid_lft forever preferred_lft forever
3: wwan0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UNKNOWN group default qlen 1000
    link/none
    inet 10.18.47.221/30 brd 10.18.47.223 scope global noprefixroute wwan0
       valid_lft forever preferred_lft forever
```

burstina, 2021-03-08:
> IP `10.10.31.13` is your WAN interface?

Yes, it is

MichaIng, 2021-03-08:
I'm wondering if it's an issue and why requests are sent and received by both, the LAN (eth0) and WAN (wwan0) interface:
```
09:58:44.429155 IP 192.168.0.209.123 > 17.253.52.253.123: NTPv4, Client, length 48
09:58:44.429212 IP 10.18.47.221.123 > 17.253.52.253.123: NTPv4, Client, length 48
09:58:44.478867 IP 17.253.52.253.123 > 10.18.47.221.123: NTPv4, Server, length 48
09:58:44.478959 IP 17.253.52.253.123 > 192.168.0.209.123: NTPv4, Server, length 48
```
The `route` command shows the WAN interface as gateway, but is probably attached to the LAN inteface? What does this show?
```
ip r
```
If it's `default via 10.18.47.222 dev eth0 onlink`, then probably it can be changed to `default dev wwan0 onlink` or via the actual WAN side gateway. Obviously I'm not having much experience with setting up a router manually 😉.

With the `/etc/network/interfaces` that dietpi-config sets up, the LAN interface is given either via DHCP or statically a gateway, which makes ifup creating that default route entry. And that should be wrong, or at least to an obsolete back and forst, when a WAN device is available where packets should by default be send through directly.

Joulinar, 2021-03-08:
yes looks like request is sent on both interfaces, but as well it seems to be received on both. 🤔 

```
09:58:44.429155 IP 192.168.0.209.123 > 17.253.52.253.123: NTPv4, Client, length 48
09:58:44.429212 IP 10.18.47.221.123 > 17.253.52.253.123: NTPv4, Client, length 48
09:58:44.478867 IP 17.253.52.253.123 > 10.18.47.221.123: NTPv4, Server, length 48
09:58:44.478959 IP 17.253.52.253.123 > 192.168.0.209.123: NTPv4, Server, length 48
```

burstina, 2021-03-09:
192.168.0.209 is another LAN client.
WAN interface as gateway is correct, it is a 4G LTE modem ("cdc-wdm0") directly connected to USB port

```
root@DietPi:~# ip r
default via 10.11.31.88 dev wwan0 proto static metric 700
10.11.31.80/28 dev wwan0 proto kernel scope link src 10.11.31.87 metric 700
192.168.0.0/24 dev eth0 proto kernel scope link src 192.168.0.1
```

yesterday I did see similar behaviour using for the first time OpenVPN via Network Manager:  tcpdump showed packets sent to server and received from Dietpi, but openvpn client went timeout. It seems packets  did not not arrive to application.

MichaIng, 2021-03-09:
Ah, now I see it also on the route output, showing the interface name as well and that wwan0 IP and gateway IP are actually different 10.11.31.87 vs 10.11.31.88 and before 10.18.47.220 vs 10.18.47.222. So yes that looks all correct as far as I can think of it.

Another idea I had was a rate limit hit or such as multiple clients from the same network (WAN IP) contact the same NTP server concurrently but your first logs don't show that second client and there are even NTP settings that send out a whole burst of requests at a time.

MichaIng, 2021-03-09:
Interesting, on my RPis now systemd-timesyncd succeeds in 1-2 loops at boot, regardless if DHCP or static IP is chosen. I turned on the notebook, first boot did not count since I had that ntpd hack inside, reverted it, upgraded all packages and since now it as well finishes time sync in 1-2 loops, that one WiFi-attached btw.

The VMs, whatever I do, remain on that 30 loops.

burstina, 2021-03-09:
my notebook, old machine running Ubuntu, systemd-timesyncd syncronize NTP pretty instant, 2 seconds at most. Same home network btw.

MichaIng, 2021-03-09:
Yes, now all machines do it aside of the VMs. Also the VMs sync in a second when doing it after boot btw, it's only during boot that it takes that long.
Two other reboots on Bullseye with 53 and 57 loops. While and even before systemd-timesyncd is waiting, the same server can be pinged successfully. Tomorrow I'll let tcpdump run to see whether it's similar that NTP packets are actually received. But I don't think that our issues are the same, actually.

Sounds like something for https://superuser.com/ or https://serverfault.com/, we need some network experts here. [tshark](https://packages.debian.org/tshark) can btw way be used to see the content of the received packages, to check whether they're valid NTP or not. It works a bit similar than tcpdump, showing all packets that match the filters, but it shows the whole content as well, hence filters should be strict enough 😉.

Joulinar, 2021-03-09:
I tent to agree that both issues are separate one. Because for @burstina it is not working at all.

Maybe issue of @burstina is related to the Gateway/Router functionality? But yeah a network expert would be welcome.

@MichaIng 
this long time sync during boot we had in past already. It was quite an intensive discussion already (without solution) https://dietpi.com/phpbb/viewtopic.php?f=11&t=7192

At least on new install this doesn't happen anymore. Just on my old live system it's still present

burstina, 2021-03-09:
While hope still lasts :) ,I finally tried `ntpd` .
I see pool 
```
DNS lookup 3.debian.pool.ntp.org started
pool DNS 3.debian.pool.ntp.org completed

```
Should it mean did it work?

 ```
ntpd -I wwan0 -d 99
 9 Mar 23:17:49 ntpd[23318]: ntpd 4.2.8p12@1.3728-o (1): Starting
 9 Mar 23:17:49 ntpd[23318]: Command line: ntpd -I wwan0 -d 99
 9 Mar 23:17:49 ntpd[23318]: proto: precision = 0.291 usec (-22)
Finished Parsing!!
config_access: top-level node 0xaaaaf4dd8700: ippeerlimit -1
hack_restrict: op RESTRICT_FLAGS addr 0.0.0.0 mask 0.0.0.0 ippeerlimit -1 mflags 00000000 rflags 00000bd0
config_access: top-level node 0xaaaaf4dd88b0: ippeerlimit -1
hack_restrict: op RESTRICT_FLAGS addr :: mask :: ippeerlimit -1 mflags 00000000 rflags 00000bd0
config_access: top-level node 0xaaaaf4dd8920: ippeerlimit -1
hack_restrict: op RESTRICT_FLAGS addr 127.0.0.1 mask 255.255.255.255 ippeerlimit -1 mflags 00000000 rflags 00000000
config_access: top-level node 0xaaaaf4dd8990: ippeerlimit -1
hack_restrict: op RESTRICT_FLAGS addr ::1 mask ffff:ffff:ffff:ffff:ffff:ffff:ffff:ffff ippeerlimit -1 mflags 00000000 rflags 00000000
config_access: top-level node 0xaaaaf4dd8a60: ippeerlimit -1
restrict source template ippeerlimit -1 mflags 4000 rflags 380
hack_restrict: op RESTRICT_FLAGS addr (null) mask (null) ippeerlimit -1 mflags 00004000 rflags 00000380
 9 Mar 23:17:49 ntpd[23318]: leapsecond file ('/usr/share/zoneinfo/leap-seconds.list'): good hash signature
 9 Mar 23:17:49 ntpd[23318]: leapsecond file ('/usr/share/zoneinfo/leap-seconds.list'): loaded, expire=2021-12-28T00:00:00Z last=2017-01-01T00:00:00Z ofs=37
event at 0 0.0.0.0 c01e 0e TAI 37 leap 201701010000 expires 202112280000
move_fd: estimated max descriptors: 1024, initial socket boundary: 16
 9 Mar 23:17:49 ntpd[23318]: Listen and drop on 0 v6wildcard [::]:123
 9 Mar 23:17:49 ntpd[23318]: Listen and drop on 1 v4wildcard 0.0.0.0:123
 9 Mar 23:17:49 ntpd[23318]: Listen normally on 2 lo 127.0.0.1:123
hack_restrict: op RESTRICT_FLAGS addr 127.0.0.1 mask 255.255.255.255 ippeerlimit -4 mflags 00003000 rflags 00000001
 9 Mar 23:17:49 ntpd[23318]: Listen normally on 3 wwan0 10.14.7.55:123
hack_restrict: op RESTRICT_FLAGS addr 10.14.7.55 mask 255.255.255.255 ippeerlimit -4 mflags 00003000 rflags 00000001
 9 Mar 23:17:49 ntpd[23318]: Listening on routing socket on fd #20 for interface updates
newpeer(0.debian.pool.ntp.org) found no existing and 0 other associations
newpeer(0.debian.pool.ntp.org) - ippeerlimit -1 ignored
key_expire: at 0 associd 19535
peer_clear: at 0 next 1 associd 19535 refid POOL
event at 0 0.0.0.0 8811 81 mobilize assoc 19535
newpeer: <null>->0.0.0.0 mode 3 vers 4 poll 6 10 flags 0x101 0x8 ttl 0 key 00000000
newpeer(1.debian.pool.ntp.org) found no existing and 0 other associations
newpeer(1.debian.pool.ntp.org) - ippeerlimit -1 ignored
key_expire: at 0 associd 19536
peer_clear: at 0 next 2 associd 19536 refid POOL
event at 0 0.0.0.0 8811 81 mobilize assoc 19536
newpeer: <null>->0.0.0.0 mode 3 vers 4 poll 6 10 flags 0x101 0x8 ttl 0 key 00000000
newpeer(2.debian.pool.ntp.org) found no existing and 0 other associations
newpeer(2.debian.pool.ntp.org) - ippeerlimit -1 ignored
key_expire: at 0 associd 19537
peer_clear: at 0 next 3 associd 19537 refid POOL
event at 0 0.0.0.0 8811 81 mobilize assoc 19537
newpeer: <null>->0.0.0.0 mode 3 vers 4 poll 6 10 flags 0x101 0x8 ttl 0 key 00000000
newpeer(3.debian.pool.ntp.org) found no existing and 0 other associations
newpeer(3.debian.pool.ntp.org) - ippeerlimit -1 ignored
key_expire: at 0 associd 19538
peer_clear: at 0 next 4 associd 19538 refid POOL
event at 0 0.0.0.0 8811 81 mobilize assoc 19538
newpeer: <null>->0.0.0.0 mode 3 vers 4 poll 6 10 flags 0x101 0x8 ttl 0 key 00000000
kernel loop status 0x41 (Clock Unsynchronized)
 9 Mar 23:17:49 ntpd[23318]: kernel reports TIME_ERROR: 0x41: Clock Unsynchronized
event at 0 0.0.0.0 c01d 0d kern kernel time sync enabled
kernel loop status 0x41 (Clock Unsynchronized)
 9 Mar 23:17:49 ntpd[23318]: kernel reports TIME_ERROR: 0x41: Clock Unsynchronized
event at 0 0.0.0.0 c012 02 freq_set kernel 0.000 PPM
event at 0 0.0.0.0 c011 01 freq_not_set
event at 0 0.0.0.0 c016 06 restart
peer_name_resolved(99) rescode 0
[... 103 lines cut ...]
pool_xmit: at 266 10.14.7.55->31.14.133.122 pool
 9 Mar 23:22:16 ntpd[23318]: Soliciting pool server 31.14.133.122
peer_xmit: at 268 10.14.7.55->0.0.0.99 mode 3 len 48 xmt 0xe3f27399.fcc0b9ae
pool DNS lookup 3.debian.pool.ntp.org started
pool DNS 3.debian.pool.ntp.org completed
restrictions: looking up 95.110.248.206
hack_restrict: op RESTRICT_FLAGS addr 95.110.248.206 mask 255.255.255.255 ippeerlimit -1 mflags 00004000 rflags 00000380
restrict_source: 95.110.248.206 host restriction added
pool_xmit: at 269 10.14.7.55->95.110.248.206 pool
 9 Mar 23:22:19 ntpd[23318]: Soliciting pool server 95.110.248.206
restrictions: looking up 185.19.184.35
pool_xmit: at 327 10.14.7.55->185.19.184.35 pool
 9 Mar 23:23:16 ntpd[23318]: Soliciting pool server 185.19.184.35
restrictions: looking up 80.211.137.82
pool_xmit: at 331 10.14.7.55->80.211.137.82 pool
 9 Mar 23:23:20 ntpd[23318]: Soliciting pool server 80.211.137.82
restrictions: looking up 2606:4700:f1::1
pool_xmit: at 332 ::->2606:4700:f1::1 pool
 9 Mar 23:23:21 ntpd[23318]: Soliciting pool server 2606:4700:f1::1
restrictions: looking up 162.159.200.123
pool_xmit: at 334 10.14.7.55->162.159.200.123 pool
 9 Mar 23:23:23 ntpd[23318]: Soliciting pool server 162.159.200.123
peer_xmit: at 335 10.14.7.55->0.0.0.99 mode 3 len 48 xmt 0xe3f273dc.fcbefa2b

```

Joulinar, 2021-03-09:
there are a couple of lines conatining the time out

`9 Mar 23:17:49 ntpd[23318]: kernel reports TIME_ERROR: 0x41: Clock Unsynchronized`

MichaIng, 2021-03-09:
Ah yes, thanks for the reminder, that indeed looks pretty similar. One difference is, that at least on the VM I can see the systemd-timesyncd sending several requests immediately. A concurrent tcpdump or even tshark might show more.

I wonder on ntpd how a successful sync log would look like. I just see successful DNS lookups for the pool servers. Not sure what the following means, which is mostly the last line before a new timestamp with new soliciting starts:
```
pool_xmit: at 262 10.14.7.55->85.199.214.99 pool
```
`ntpd -q` (q=quit) btw can be used to have ntpd quitting automatically once it successfully synced. So that would show for sure how long it takes or if it even succeeds. To measure the time: `time ntpd -q`

Joulinar, 2021-03-09:
can you share `iptables -L`

Joulinar, 2021-03-10:
quite different to what was set before https://github.com/MichaIng/DietPi/issues/4177#issuecomment-792265573

```
root@DietPi:~# iptables -L
Chain INPUT (policy DROP)
target     prot opt source               destination
ACCEPT     all  --  anywhere             anywhere
ACCEPT     tcp  --  anywhere             anywhere             tcp flags:ACK/ACK
ACCEPT     udp  --  anywhere             anywhere             udp spt:domain dpts:1024:65535
ACCEPT     icmp --  anywhere             anywhere             icmp echo-reply
ACCEPT     icmp --  anywhere             anywhere             icmp destination-unreachable
ACCEPT     icmp --  anywhere             anywhere             icmp source-quench
ACCEPT     icmp --  anywhere             anywhere             icmp time-exceeded
ACCEPT     icmp --  anywhere             anywhere             icmp parameter-problem
```

the whole input was set to `DROP`. 

But I'm not sure how DietPi scripts should have influenced this. @MichaIng can confirm it better than I, but DietPi is not doing anything with `iptables`

burstina, 2021-03-10:
> quite different to what was set before [#4177 (comment)](https://github.com/MichaIng/DietPi/issues/4177#issuecomment-792265573)

true, perhaps it was too late the night and Dietpi was telling me to sleep 😄 

So a did again this morning `iptables -L `this morning after a `reboot ` command and it looks more like the same before.

```
root@DietPi:~# date
Wed Mar 10 08:23:42 CET 2021
root@DietPi:~# iptables -L
Chain INPUT (policy DROP)
target     prot opt source               destination
ACCEPT     all  --  anywhere             anywhere
ACCEPT     tcp  --  anywhere             anywhere             tcp flags:ACK/ACK
ACCEPT     udp  --  anywhere             anywhere             udp spt:domain dpts:1024:65535
ACCEPT     icmp --  anywhere             anywhere             icmp echo-reply
ACCEPT     icmp --  anywhere             anywhere             icmp destination-unreachable
ACCEPT     icmp --  anywhere             anywhere             icmp source-quench
ACCEPT     icmp --  anywhere             anywhere             icmp time-exceeded
ACCEPT     icmp --  anywhere             anywhere             icmp parameter-problem
ACCEPT     all  --  anywhere             anywhere

Chain FORWARD (policy ACCEPT)
target     prot opt source               destination

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination
```

I restarted systemd-timesyncd and time had synced.
I tried openvpn client and it worked fine.

```
Mar 10 08:22:01 DietPi systemd[1]: packagekit.service: Succeeded.
Mar 10 08:23:07 DietPi systemd[1]: Stopping Network Time Synchronization...
Mar 10 08:23:07 DietPi systemd[1]: systemd-timesyncd.service: Succeeded.
Mar 10 08:23:07 DietPi systemd[1]: Stopped Network Time Synchronization.
Mar 10 08:23:07 DietPi systemd[1]: Starting Network Time Synchronization...
Mar 10 08:23:07 DietPi systemd[1]: Started Network Time Synchronization.
Mar 10 08:23:07 DietPi systemd-timesyncd[3701]: Synchronized to time server for the first time 95.110.248.206:123 (0.pool.ntp.org).
```

I executed `reboot ` command a couple of times since this issue has been opened, but it never solved the problem until I did a firewall reset via Webmin.

Joulinar, 2021-03-10:
maybe there was more in the hidden darkness of `iptables` leading to packages being dropped. Maybe `iptables` or `webmin` software package has been updated during execution of `dietpi-update` causing some kind of inconsistency?? Quite some guessing but at least it could explain it a little bit. Probably we never find out...

burstina, 2021-03-10:
> ... Maybe `iptables` or `webmin` software package has been updated during execution of `dietpi-update` causing some kind of inconsistency?? ...

This looks right. Before submitting this issue, I already went to Webmin to commit rules configuration as they were (the Apply Configuration button). I did not Revert Configuration do default, because "reset" seems always a drastic measure to me.

Well, next system updates I'd better remember to check this

MichaIng, 2021-03-10:
> pretty simple configuration

😄, zero rules, all accepted, so yes as simple as no firewall.

It looks like Webmin produces doubled rules, the
```
ACCEPT     all  --  anywhere             anywhere
```
being present two times. Does that, without a further filters, not accept simply everything, e.g. overriding
```
ACCEPT     icmp --  anywhere             anywhere             icmp echo-reply
```
which is to accept "icmp echo-reply" packets only?

And yes, I can assure that DietPi (updates) do never touch firewall rules and I'm 99% sure that even iptables package updates do never affect the effective rules. Also there was no iptables update recently: https://metadata.ftp-master.debian.org/changelogs//main/i/iptables/iptables_1.8.2-4_changelog
But there were two Webmin updates recently, which I hence suspect being the culprit here: https://webmin.com/changes.html

The only case where we touch iptables is to apply IP forwarding rules for WiFi HotSpot and VPN server installs, to share the internet connection to WiFi/VPN clients. In case of WireGuard this is only done on service start, I'll review the OpenVPN and WiFi HotSpot rules to assure that they only touch the FORWARD chain and not erase any existing INPUT/OUTPUT rules or policy.

burstina, 2021-03-10:
> 
> 
> > pretty simple configuration
> 
> 😄, zero rules, all accepted, so yes as simple as no firewall.
> 

well... look at `Webmin `and `iptables -L`, screenshot has been taken after a fresh Webmin login and iptables -L command just few second after. Is it only my eyes, or do they show different rules?

[image]

burstina, 2021-03-11:
well, I'm going to have an headache..
This morning the situation was back to "not working again", time not syncing again.

So went through Webmin interface and put in input rule list a new one on port 123 for NTP and a new other one on 4443 (for vpn)

It worked.

I'm really confused. Input rules? Shouldn't connection be already up from `systemd-timesync `itself?

Perhaps it could be good to do a deeper look to iptables rules file, the one Webmin imports by `iptables-restore `:

```
root@DietPi:~# cat /etc/iptables.up.rules
# Generated by iptables-save v1.8.2 on Tue Mar  9 22:42:43 2021
*filter
:INPUT DROP [0:0]
:FORWARD ACCEPT [0:0]
:OUTPUT ACCEPT [0:0]
-A INPUT ! -i wwan0 -j ACCEPT
-A INPUT -p tcp -m tcp --tcp-flags ACK ACK -j ACCEPT
-A INPUT -p udp -m udp --dport 1024:65535 --sport 53 -j ACCEPT
-A INPUT -p icmp -m icmp --icmp-type 0 -j ACCEPT
-A INPUT -p icmp -m icmp --icmp-type 3 -j ACCEPT
-A INPUT -p icmp -m icmp --icmp-type 4 -j ACCEPT
-A INPUT -p icmp -m icmp --icmp-type 11 -j ACCEPT
-A INPUT -p icmp -m icmp --icmp-type 12 -j ACCEPT
-A INPUT -p udp -m udp -m multiport --ports 123 -j ACCEPT
-A INPUT -p udp -m udp --dport 4443 -j ACCEPT
COMMIT
# Completed on Tue Mar  9 22:42:43 2021
# Generated by iptables-save v1.8.2 on Tue Mar  9 22:42:43 2021
*mangle
:PREROUTING ACCEPT [10882344:9768269898]
:INPUT ACCEPT [184432:31813275]
:FORWARD ACCEPT [10697740:9736418163]
:OUTPUT ACCEPT [229992:41381675]
:POSTROUTING ACCEPT [10927732:9777799838]
COMMIT
# Completed on Tue Mar  9 22:42:43 2021
# Generated by iptables-save v1.8.2 on Tue Mar  9 22:42:43 2021
*nat
:INPUT ACCEPT [0:0]
:PREROUTING ACCEPT [0:0]
:OUTPUT ACCEPT [0:0]
:POSTROUTING ACCEPT [0:0]
-A POSTROUTING -o tun0 -j MASQUERADE
-A POSTROUTING -o wwan0 -j MASQUERADE
COMMIT
# Completed on Tue Mar  9 22:42:43 2021

```

Joulinar, 2021-03-11:
I'm far away from being a iptables expert but input is set to drop

`:INPUT DROP [0:0]`

probably this will drop all packages not being allowed specifically.

MichaIng, 2021-03-11:
> probably this will drop all packages not being allowed specifically.

Exactly. It's not about who initiated the connection, but it affects all incoming packets. Hence it makes sense the an accept port 123 rule is required to make time sync work, same as port 80 and 443 rules are required to make APT work.

For rules about who initiated the connection, connection marks need to be used: It's e.g. possible to mark a connection when an INPUT packet is from a new connection, which must then be an inbound connection. Based on that connection mark, packets can then be dropped or filtered. But I'd need to search the exact options/commands for that 😅.

About the mismatch between `iptables -L`, the Webmin UI and restore file:
- Note that only the upper part below `*filter` is relevant to compare as this is the default table shown when using iptables without the `-t`/`--table` option.
- That matches the Webmin UI pretty fine. 
- According to the iptables manpage the interface names are suppressed until you use `-v`: `iptables -v -L INPUT` to limit it to the INPUT chain but show the complete rules. This explains the missing "if interface is not wwan0" part.

burstina, 2021-03-13:
> `:INPUT DROP [0:0]`
> 
> probably this will drop all packages not being allowed specifically.

My opinion is this rule to be ok. I Could obviously be wrong, I'm really far for being an `iptables `expert!
As far as I know, but it is possible I lack the most, firewalls don't block packets on estabilished connections.
Roughly, Initiating a connection means to request a comunication channel to a destination host on a defined port.
When a machine request a connection, it is estabilished after firewalls check rules compatibility: from now on, outgoing and incoming packets exchanged thru this connection are free to pass.
That's the reason why usually all outoging connections are allowed and incoming connections are always dropped

So if Dietpi open a connection on port 123 and rules for outgoing packets doesn't block port 123, after connection has been estabilished, all packets from NTP servers passing thru this connection should be allowed.

That's the reason why usually online applications with simple network management, little games for example, doens't require firewall configuration: they open an outbound connection.
Thus, games far more heavy also needs lo listen to incoming connection from a undefined game server on a different port: that need sfirewalls to be setup accordingly.

I did see a lot of easier firewall to configure, beginning with my ASUSWRT home router :)

Well, at least we have the proof this issue is related to firewall configuration.

MichaIng, 2021-03-13:
> My opinion is this rule to be ok.

Of course it is, it's a very secure approach to only allow a known range of ports, especially on Linux distributions in general or DietPi specifically since there is only very specific transparent network traffic. Consequently you could use DROP policy and add related rules on all chains, as those chains are related to single packets, not to connections, so allowing outgoing packets on a certain port but not allowing the answer to pass does not make that much sense 😉.
I once tried to go with that strategy on Windows... impossible without breaking half of the system down as there is an unbelievable amount of various ports used by various system components, and most of them do not behave well when being delayed by a firewall prompt and me to try and giving up to find detailed information why this or that executable is needing this or that connection for something that looks like a complete offline feature 😄.

> firewalls don't block packets on estabilished connections.

Every firewall is different, but iptables sorts every single packet into those chains and applies policies and rules accordingly on a packet level, not a connection level. As mentioned above, you can mark a connection as well and create rules based on those connection marks, but `:INPUT DROP [0:0]` does block every single incoming packet, regardless if it belongs to a new or open connection, as long as no rule allows it. This diagram shows quite nicely which chains and rule tables are passed by each packet:

[image]

Ah nice, the same website explains the packets vs connection block issue as well 🥳: https://www.booleanworld.com/depth-guide-iptables-linux-firewall/#The_connection_tracking_module

burstina, 2021-03-15:
> Ah nice, the same website explains the packets vs connection block issue as well, including the module to address this via connection marks (conntrack) 🥳: https://www.booleanworld.com/depth-guide-iptables-linux-firewall/#The_connection_tracking_module

A more stupid-user friendly firewall? :)

MichaIng, 2021-03-15:
[ufw](https://manpages.debian.org/ufw) is a frontend for iptables with a simplified interface: https://www.linux.com/training-tutorials/introduction-uncomplicated-firewall-ufw/

It has the `in`/`out` selector, but honestly I'm not sure whether this is for connection level or packet level as well. Could be given a try at least and the created rules reviewed with `iptables -L`.

burstina, 2021-03-31:
Now that the problem has been clarified, I think we can close this issue.
thanks

closed 2021-03-31 by burstina

tavuntu, 2021-04-01:
Sorry guys, I'm not sure I understand how this was solved, can some please tell me? I'm having the same issues when I restart (after connecting to wifi) and it won't connect anymore, showing lots of these:
```
[ INFO ] DietPi-Run_NTPD | Waiting for completion of systemd-timesyncd (2/10)
[ INFO ] DietPi-Run_NTPD | Waiting for completion of systemd-timesyncd (3/10)
[ INFO ] DietPi-Run_NTPD | Waiting for completion of systemd-timesyncd (4/10)
[ INFO ] DietPi-Run_NTPD | Waiting for completion of systemd-timesyncd (5/10)
[ INFO ] DietPi-Run_NTPD | Waiting for completion of systemd-timesyncd (6/10)
[ INFO ] DietPi-Run_NTPD | Waiting for completion of systemd-timesyncd (7/10)
...
```

ravenclaw900, 2021-04-01:
Hi,
Could you post `iptables -L`.

tavuntu, 2021-04-01:
Not at home right now but will do in a few hrs.

tavuntu, 2021-04-01:
So it seems this was my mistake since I set up wifi but not the Timezone, it connects now after a few seconds when rebooting. Now, is there a way to not wait for wifi before login and do that in the background? I'm guessing is something around Network Options: Misc

Joulinar, 2021-04-02:
you can deactivate the time sync on boot, but this will leave your system with incorrect time settings. There is an option inside `/boot/dietpi.txt`

```
# Network time sync: 0=disabled | 1=boot only | 2=boot + daily | 3=boot + hourly | 4=Daemon + Drift
CONFIG_NTP_MODE=2
```

Or you go to test option 4

tavuntu, 2021-04-02:
Using option 0 there + Boot Net Wait = OFF seems to do the trick, however, I still feel it's taking more time than necessary when booting, I remember I installed DietPi a couple of months ago and it used to go from boot to login faster, not sure what I'm doing differently now 🤔

Joulinar, 2021-04-02:
Can you do a reboot and check following right after `systemd-analyze blame`

tavuntu, 2021-04-02:
sure, this is the output:
```
11.525s ifup@eth0.service
9.518s dietpi-boot.service
8.730s ifup@wlan0.service
1.168s dev-mmcblk0p2.device
 490ms systemd-fsck@dev-disk-by\x2dpartuuid-e8af6eb2\x2d01.service
 420ms keyboard-setup.service
 308ms systemd-udev-trigger.service
 301ms networking.service
 265ms dietpi-preboot.service
 195ms systemd-fsck-root.service
 176ms ssh.service
 169ms systemd-journald.service
 156ms dietpi-ramlog.service
 142ms fake-hwclock.service
 127ms systemd-remount-fs.service
 124ms sys-kernel-debug.mount
 115ms systemd-udevd.service
  95ms systemd-tmpfiles-setup.service
  88ms dev-mqueue.mount
  84ms var-swap.swap
  77ms boot.mount
  71ms systemd-journal-flush.service
  56ms systemd-modules-load.service
  53ms systemd-sysctl.service
  52ms systemd-update-utmp.service
  52ms systemd-sysusers.service
  52ms systemd-tmpfiles-setup-dev.service
  51ms systemd-random-seed.service
  47ms kmod-static-nodes.service
  46ms sys-kernel-config.mount
  45ms systemd-update-utmp-runlevel.service
  40ms systemd-rfkill.service
  33ms systemd-user-sessions.service
  25ms console-setup.service
  20ms alsa-restore.service
  17ms ifupdown-pre.service
  13ms var-log.mount
  13ms tmp.mount
```

Joulinar, 2021-04-02:
interesting, most of the time is spend on trying to bring up Ethernet interface. Can you check status in `dietpi-config`? Usually it should be disabled if there is no cable plug in

tavuntu, 2021-04-03:
Doing that (Wifi on and Ethernet off) + Disabling Boot Net Wait + "Custom" in Time sync mode, seems to do the trick at least for me (I've installed RetroPie, retroarch and other related software so I'm not %100 sure if that somehow caused this).

Now it boots a logins a lot faster. Thanks a lot!

MichaIng, 2021-04-04:
Switching time sync mode to "Custom" btw will earlier or later break network connections, as your system time is not synchronised anymore. We sync at boot by default so that e.g. DietPi and APT update checks, as well as the banner prints that involve a network request (WAN IP, MOTD, weather, ...) do not fail as of future or past HTTPS certificate valid time. While these are probably not essential to you, generally keeping your system time in sync should be. So when you switch to "Custom", assure that a time sync daemon is active, e.g. by enabling the `systemd-timesyncd` service:
```
systemctl enable systemd-timesyncd
```

The network time sync btw should succeed within 1-2 seconds usually (though e.g. my VMs here also require much longer for that during boot, and I don't know why). When network is configured via DHCP, and your router an serve as NTP server as well, usually (Linux) clients with with your router, instead of with public NTP providers, which should be very fast. You can also force it via `dietpi-config` > `Network Options: Misc` > `NTP Mirror` > `Gateway`.

tavuntu, 2021-04-04:
Alright, I've done those adjustments, thanks for the heads up!