MichaIng/DietPi #4178: Install of Bitwarden_RS stalling indefinitely at "compiling walkdir v2.3.1"

Vizban, 2021-03-06:
#### Required Information
- DietPi version | Fresh DL from dietpi.com (https://dietpi.com/downloads/images/testing/DietPi_RPi-ARMv8-Buster.7z)
- SBC model | RPi4 2GiB
- SDcard used | SanDisk ultra 16gb

#### Additional Information (if applicable)
- Software title | Bitwarden_RS
- Fresh install during initial setup
- Still occurs after a fresh flash of the SDcard

#### Steps to reproduce
1. Download https://dietpi.com/downloads/images/testing/DietPi_RPi-ARMv8-Buster.7z
2. Flash to SDcard
3. SSH into root
4. Start Setup with selecting Bitwarden_RS from optimized software pool

#### Expected behaviour
- Initial setup completes and installs the selected software

#### Actual behaviour
- Installing seems to stall indefinitely at the "compiling walkdir v2.3.1" (no change after 2+ hours)
[image]
- Opening a second terminal and trying to ssh into root does not yield a response after entering the password

#### Extra details
- I am fascinated by the complexity, completeness and longevity of this project and humbled by the immense support that you provide to newbies like myself. 
- This is my first github issue bug report. Please do point out if I failed to provide required information or seem to have made a rookie mistake.

Thank you!

MichaIng, 2021-03-06:
Many thanks for your report.

Yeah, it's a heavy compilation, however an RPi 4 should be able to deal with it without issues. However, did you install Bitwarden_RS along or together with other software titles? If so could you try install it along? Probably some other install consumes already too much memory (temporary files or such).

Another thing, do you use a fan for that Pi, or could you check whether too high CPU temperature during compilation could be the issue? You could do a stress test via dietpi-config > Tools or have a second SSH session open running:
```
while sleep 1; do G_OBTAIN_CPU_TEMP; done
```

We're likely switching from binary compilation to using pre-compiled binaries from the Docker images that are provided for all supported architectures. That would eliminate the need for that time consuming install: https://github.com/MichaIng/DietPi/issues/3724#issuecomment-789709945

Vizban, 2021-03-07:
Thank you for your quick & detailed reply, MichaIng! I now got around to testing your suggestions.

#**tl;dr** It appears that 2 GiB RAM and the default 147 MiB swap file size were not enough for the heavy Bitwarden_RS compilation. Temporarily increasing the swap file size appears to have solved the issue.

**1. CPU temperature // case & fan**
I use a FLIRC case without fan. The temperature during the compilation procedure stayed at around 40 but dropped when arriving at walkdir. This led me to your memory theory:

**2. Memory usage**
Indeed, a second terminal showed with free -m:
```
              total        used        free      shared  buff/cache   available
Mem:           1900         662          23        1178        1213          13
Swap:           147         147           0
```

So I flashed the SDcard anew, turned logging off and let it run the initial setup without any additional software installs.
This left me after a reboot with:
```
              total        used        free      shared  buff/cache   available
Mem:           1900          40        1751          16         107        1786
Swap:           147           0         147
```

I then started the Bitwarden_RS installation via dietpi-software.
When arriving at compiling walkdir, I had roughly:

```
              total        used        free      shared  buff/cache   available
Mem:           1900         191         362        1285        1345         364
Swap:           147           5         142
```

It stayed at compiling walkdir until:
```
              total        used        free      shared  buff/cache   available
Mem:           1900         689          24        1147        1185          14
Swap:           147         147           0
```

**3. Increase swap file size and repeat**
Although I currenlty only use an SDcard, I increased the swap file size to 2048 (super easy with die dietpi tool!), rebooted and started the installation anew.

The maximum memory/swap usage during the compilation process was:
              total        used        free      shared  buff/cache   available
Mem:           1900        1471          25         361         403          17
Swap:          2047        1040        1007

After that, the installation successfully finished and I was able to access the Bitwarden_RS web interface.

Thank you for directly pointing me into the right direction!

closed 2021-03-07 by Vizban

MichaIng, 2021-03-08:
Great that it finally worked. Strange is that it does not simply error out when reaching max, at least this is what I faced when testing and playing with min memory requirements. Obviously that min was too low to be failsafe. With less than 3 GiB memory, concurrent build jobs are limited to two, but some jobs require only 100 MiB, others more than 600 MiB, so when by chance two high memory jobs overlap, that might cause this issues, and that build time for each dependency varies from device to device.

However, I'm looking forward for providing a pre-compiled package-based install with next release 😃.