MichaIng/DietPi #4179: Remot3.it not working after installed from Software Optimized

shaungt1, 2021-03-06:
### Creating a bug report/issue
remot3.it Not working after installed from Software Optimized

#### Required Information
- DietPi version | `latest`
- Distro version | `Amiberry image:`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra 128GB)

#### Additional Information (if applicable)
- Software title | remote.it
- Was the software title installed freshly or updated/migrated? **Yes**
- Can this issue be replicated on a fresh installation of DietPi? **Yes**

- Bug report ID | `echo $G_HW_UUID` **n/a**

#### Steps to reproduce

1. I am building a cluster of raspi 4 8Gb x 6 and installed the following programs below on each before installing remote.it
2. After install, update, and upgrade I installed ONLY remote.it service >> did a reboot and could not acquire a connection to my local IP and port 2999 to access remote.it service. **Ex: http://192.168.1.243:29999** 
3. was meet with a default page error.

**I ran these software packages 1st:**
`LXDE: ultra lightweight desktop
Chromium: web browser for desktop or autostart
XRDP: remote desktop protocol (rdp) server 
RealVNC Server: desktop for remote connection 
Amiberry: Optimised Amiga emulator for ARM-based SoCs 
LEMP: nginx    | mariadb | php
phpMyAdmin: optional mysql admin tools
Redis: optional non-sql database store
Certbot: Obtain and renew Let's Encrypt SSL certs for HTTPS 
Samba Server: feature-rich file server `

I then ran the dietpi update and no connection was available as if it where not installed correctly

#### Expected behaviour

- Remote.It would not acquire a connection or work.

#### Actual behaviour

- No Connection as if it where not installed

#### Extra details

-  In order to get it working I did a manual install according to remot3.it website as follows:
- I ran this and it worked fine across all 6 of my raspis

sudo apt update 
sudo apt install remoteit
Copy the following commands in my browser 
=========================================================================
Continue your device configuration at 
 http://find.remote.it
 or http://raspberrypi.local:29999
 or http://192.168.1.243:29999
=========================================================================

Note: I'm not sure if after my reboot before installing Remote.it via software optimized  if the `update`  might have had something to do with it? or if the optimized software was not installing the packages correctly? Following the the manual commands above seem to work fine though.

I also tried on a fresh build with only installing  `LXDE: ultra lightweight desktop` then going into install remote.it and the same problem occurred after systems setup.

MichaIng, 2021-03-06:
Did you run the installer after installing the installer package (which is what is done 😄) via `dietpi-software`?
```
connectd_installer
```
I wonder why we do not call it as part of the install process. It requires manual inputs, but at least on attended installs we can do that.

MichaIng, 2021-03-24:
I'm turning this into an enhancement, to have the connectd_installer called directly during the install process, when it is an interactive installs. E.g. when Remot3.it is installed as part of automated first boot installs, it still needs to be called manually, as it requires interactive console input.

MichaIng, 2021-04-21:
Configuration is now offered directly after the installs have finished. It fit's well there between other interactive prompts about GPU memory split and autostart choices: https://github.com/MichaIng/DietPi/commit/1c4003be7fb0a1a404605027bb10171e7e1aa151

I think you mixed up the two possible Remot3.it setups, once via the connectd package (that we install) and one via the remoteit package. The second one provides the web interface, which makes it heavier as well, of course, the first one is much slimmer, more established and mass-production ready: https://support.remote.it/hc/en-us/articles/360044787712-Create-a-remote-it-account
However, using connectd requires you to setup application connections from the command line while with the remoteit package the web interface seems to be more convenient.

Earlier I thought that the remoteit package is available for Raspberry Pi only, but it is available now for other Linux architectures/platforms as well. They still have the RPi/non-RPi separation, but the archives are actually the same: https://remote.it/downloads/

So we could implement the alternative remoteit-package based Remot3.it variant as dedicated install option.

closed 2021-04-24 by MichaIng