MichaIng/DietPi #4181: v7.0 on NanoPi ZeroPi: no network, can't install

f00b4r0, 2021-03-07:
Freshly downloaded 7.0 image, flashed on SD card, I can see that the device boots, the heartbeat led starts flashing, the ethernet LEDs blink, but the device never sends a DHCP request, and is unreachable: impossible to install or debug (no serial OTG on this device).
Fetched current Armbian Buster to check that everything is otherwise all right: it is. Installer boots and device becomes reachable.

Joulinar, 2021-03-07:
Hi,

many thanks for your message. But there is no DietPi v7 that can be download because DietPi v7.0.2 has been released end of Feb 2021, while the NanoPi ZeroPi was created Jun 2020 already. Probably there is a misunderstanding due to the file name `DietPi_NanoPiZeroPi-ARMv7-Buster.7z`. `ARMv7` is the architecture and not the DietPi version 😉 

But this doesn't ix your issue. As you don't have a monitor attached, it will be hard to investigate. Are you able to connect the SD card to another Linux based computer. This way you could have a look to the first boot log files located on the ext4 partition. Log files are located at `/var/tmp/dietpi/logs`.

MichaIng, 2021-03-08:
Probably worth the build a new image, e.g. if there was a new hardware revision that requires a new kernel version.

f00b4r0, 2021-03-10:
> Hi,
> 
> many thanks for your message. But there is no DietPi v7 that can be download because DietPi v7.0.2 has been released end of Feb 2021, while the NanoPi ZeroPi image was created Jun 2020 already. Probably there is a misunderstanding due to the file name `DietPi_NanoPiZeroPi-ARMv7-Buster.7z`. `ARMv7` is the architecture and not the DietPi version 😉

😑
And here I thought ".7z" was the version :P

The website says "7.0" so I report "7.0". 

BTW thank you for pointing out the problem that is the lack of image versioning.

> But this doesn't fix your issue.

Indeed.

> As you don't have a monitor attached, it will be hard to investigate. Are you able to connect the SD card to another Linux based computer. This way you could have a look to the first boot log files located on the ext4 partition. Log files are located at `/var/tmp/dietpi/logs`.

I'm afraid I was in urgent need of a working system so I installed armbian, which I subsequently stripped down to what I needed, a close enough approximation of dietpi as far as I'm concerned. I guess this will be for someone else to debug.

MichaIng, 2021-03-10:
New image uploaded, let's see if it helps: https://dietpi.com/downloads/images/DietPi_ZeroPi-ARMv7-Buster.7z

f00b4r0, 2021-03-11:
> New image uploaded, let's see if it helps: https://dietpi.com/downloads/images/DietPi_ZeroPi-ARMv7-Buster.7z

I tried booting that and it was reachable. I guess this is fixed.

Side note feature request: a simple knob/option to disable automatic resize would be very nice.

closed 2021-03-11 by f00b4r0

Joulinar, 2021-03-11:
well if there would be no resize at the first boot, your installation most probably will fail as RootFS would be to small.

f00b4r0, 2021-03-11:
> well if there would be no resize at the first boot, your installation most probably will fail as RootFS would be to small.

I don't see why if you don't install anything after first boot. The point is to be able to decide partitioning (e.g. split out `/home`) without having to shrink `/` after the fact.

MichaIng, 2021-03-11:
Probably we can add an option to define the desired root file system size, or the desired remaining free space on the drive, or both.