MichaIng/DietPi #4182: LanCache.NET

Camry2731, 2021-03-07:
### Creating a software request

#### Formal software information
- Software title | LanCache.NET
- Short description | Make the most of your network. Get more play for your gamers. Download your games once and serve them out to many people at your LAN.
- Official URL (if available) | https://lancache.net/
- GitHub URL (if available) | https://github.com/lancachenet
- (Official) install documentation | https://lancache.net/docs/

#### Are there similar/alternative software titles available with DietPi-Software?

- There is currently not

#### What makes your requested software better than the above solutions, if available?

- N/A

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?

- DietPi can make the install easier by making it work with services such as PiHole and/or Unbound, which would require additional configuration since these programs are ones that change how your DNS is used. Additionally, the program does require Docker and Docker-Compose, which isn't something that's on a Linux install by default, but DietPi has support for already

#### Can you provide the installation steps that you would suggest DietPi-Software to do?

1. Check for the needed prerequisites and see if there installed already. (Docker, Docker-Compose, etc.)
2. If there not present, install them
3. Install LanCache.NET using the commands listed on the docs page (Or a variation of it to make it more DietPi friendly)
4. Detect the use of programs that change how the DNS is used or changed (PiHole, Unbound, etc.)
5. Apply the needed config changes in order for these types of programs to still work while also allowing LanCache.NET to work (Not too sure of the exact order needed)
6. ???
7. Profit?

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.

- Yes, although I'm not 100% familiar with everything, I'd be willing to try and help out where I can.

Joulinar, 2021-03-07:
Hi,

many thanks for your request. Looks like a simple docker container. But not sure if there is a real benefit as PiHole as well as Unbound already do DNS caching.

ravenclaw900, 2021-03-07:
From my quick read, this doesn't just do DNS caching, but also physically caches the games on disk.

Camry2731, 2021-03-07:
> Hi,
> 
> many thanks for your request. Looks like a simple docker container. But not sure if there is a real benefit as PiHole as well as Unbound already do DNS caching. 

Appreciate the reply. Although both PiHole and Unbound do DNS caching, wouldn't the caching of LanCache.NET be different? As this would allow games to be pulled from a storage device on the local network rather than from the provider servers? Unless that same functionality can be accomplished in PiHole?

Joulinar, 2021-03-07:
ah ok it's caching games as well.

MichaIng, 2021-03-08:
If I understood it right, the DNS feature is used to redirect requests to game platform update/download servers to the central cache as well. So without it, the game or platform can be started from all participants via a single actual game/platform installed on the cache server. But those each participants instance would all still contact the manufacturer/platform provider servers for updates, new, patches, whatever. The additional DNS service is basically a hosts list to redirect those requests to the cache server as well, to reduce redundant and probably incompatible network traffic.

In conjunction with Pi-hole and/or Unbound this would mean an additional DNS server in the row. But when not putting that LanCache DNS service as the first server in the row, and if all LAN clients resolve via Pi-hole already, setup effort could be reduced by changing the Pi-hole upstream DNS to that LanCache DNS so that clients or router settings don't need to be changed. Also when Unbound is installed, LanCache DNS upstream could be connected to Unbound.

If the involved host list is easily available and can be easily updated, it could be also integrated into Pi-hole and/or Unbound directly to eliminate the need for the dedicated LanCache DNS instance.

So generally I see that possibility that we could do some helpful setup steps aside of installing the Docker container only. And personally I'm a friend of all efforts that aim to reduce network traffic, disk usage etc where applicable 😄.

ravenclaw900, 2021-03-08:
There's a list of domain names at https://github.com/uklans/cache-domains.

Joulinar, 2021-03-08:
Basically if you have Pihole as well as Unbound installed, apps order would be: LAN client > Pihole > LanCache > Unbound > upstream DNS. On container settings for LanCache, port settings would need to be adjusted away from port 53 and Unbound would need to be upstream DNS. Means we would need to adjust compose config and env file before.

MichaIng, 2021-03-08:
Exactly. Although probably it should be an option whether to have **Client > Pi-hole > LanCache > Unbound** or **Client > LanCache > Pi-hole > Unbound**, since depending on probably existing DNS settings sometimes the first and sometimes the seconds would cause less issues.

But with https://github.com/uklans/cache-domains, LanCache DNS is not necessarily required anymore. There are even two scripts to create an Unbound or dnsmasq (Pi-hole) config based on chosen service: https://github.com/uklans/cache-domains/tree/master/scripts

Joulinar, 2021-03-08:
looks like an interesting alternative, but it would require a `cache server`??

MichaIng, 2021-03-08:
That is the core part of LanCache 😄, what they call "Monolithic", LanCache DNS is a dedicated second container and the SNI proxy to serve HTTPS traffic as a third. And the Docker Compose image runs all three of them, if I get it right. But should be possible to keep using the compose image and disable the DNS container when not required.

Joulinar, 2021-03-08:
ok found a big downside and a possible nogo. Both container for LanCache are not able to run on ARM. According DockerHub, they are compiled for AMD64 only. 

https://hub.docker.com/r/lancachenet/monolithic/tags?page=1&ordering=last_updated
https://hub.docker.com/r/lancachenet/lancache-dns/tags?page=1&ordering=last_updated

MichaIng, 2021-03-08:
Although I think on a LAN you don't want to serve a bunch of games to a bunch of PCs from a little SBC anyway 😄. For the use case I think it's fine currently. But also larger ARM servers are raising, so soon ARM builds will make sense.

Joulinar, 2021-03-08:
at least first container I was able to build on arm. however the 2nd is more tricky 🤔

Joulinar, 2021-03-08:
I'm able to start both container now. But `monolithic` container is failing right after start. Probably I messed something up as the container is build on other container and it was needed to compile all of them in a row / one by one.

```
nginx: [emerg] "proxy_max_temp_file_size" directive invalid value in /etc/nginx/sites-available/generic.conf.d/root/20_cache.conf:29
nginx: configuration file /etc/nginx/nginx.conf test failed
```

I had some issues on the required `ubuntu-nginx` container and it was needed to manipulate the `Dockerfile` 🙄 https://github.com/lancachenet/ubuntu-nginx

Joulinar, 2021-03-08:
I should have checked further before trying myself 🤣 

That's basically same explanation for not being available on ARM, as you already have given @MichaIng , https://github.com/lancachenet/monolithic/issues/112

However there is a fork created for RPi and it seems to be working https://github.com/jrcichra/lancache-rpi

[image]

MichaIng, 2021-03-08:
Nice, and it should work on all ARMv7 boards and if I'm not mistaken ARMv7 containers can run on ARMv8 Docker/systems as well?

Joulinar, 2021-03-08:
it should, according DockerHub, container is available for ARMv7 and ARM64. I did the test on 32bit https://hub.docker.com/r/jrcichra/lancachenet-lancache-dns/tags?page=1&ordering=last_updated