MichaIng/DietPi #4183: Python3 Module lost after Rebooting Dietpi

Gill-Bates, 2021-03-08:
When I install a Python3 module with `pip` (in my case `psutils`), the module will be "lost", after rebooting the Raspberry Pi.

Executing my Python-Script will throw "Missing Module". Reinstalling the Module will solve this issue (until the next Reboot).

Bug or Feature? :-)

MichaIng, 2021-03-08:
Sounds like a read-only feature, lol.

Did you check whether the files have been created at /usr/local/lib|bin? And any kernel errors after installing the module?
```
dmesg -l emerg,alert,crit,err
```

Gill-Bates, 2021-03-08:
My Userdata is outsourced to an external drive. It that maybe the issue?

[image]

MichaIng, 2021-03-08:
Not as long as it's not a local Python module installed to the userdata directory. Global/root pip installs are done to /usr/local/lib and /usr/local/bin, which is located on the root filesystem.

Gill-Bates, 2021-03-09:
Issue solved. Root-Cause: OSI-Layer 8 :-)
I have called the script in a wrong manner.

closed 2021-03-09 by Gill-Bates