MichaIng/DietPi #4184: External hard drive not being recognized, stuck in apt update

usmanmehmood55, 2021-03-09:
#### Details:
- Date           | Mon Mar  8 15:48:45 PKT 2021
- DietPi version | v7.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite 64-bit
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | Linux DietPi 5.10.17-v8+ #1403 SMP PREEMPT Mon Feb 22 11:37:54 GMT 2021 aarch64 GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=0)
- Command        | apt-get -q update
- Exit code      | 100
- Software title | DietPi-Drive_Manager
#### Steps to reproduce:

1. Attach external Hard drive on RPi 4 2GB
2. Launch drive manager
#### Expected behaviour:

- It should have showed me the hard drive being mounted
#### Actual behaviour:

- it starts apt update and then fails

Hit:1 http://packages.microsoft.com/repos/code stable InRelease
Hit:2 https://deb.debian.org/debian buster InRelease
Hit:3 https://archive.raspberrypi.org/debian buster InRelease
Get:4 https://deb.debian.org/debian buster-updates InRelease [51.9 kB]
Get:5 https://deb.debian.org/debian-security buster/updates InRelease [65.4 kB]
Get:6 https://deb.debian.org/debian buster-backports InRelease [46.7 kB]
Reading package lists...
E: Release file for https://deb.debian.org/debian/dists/buster-updates/InRelease is not valid yet (invalid for another 15h 23min 52s). Updates for this repository will not be applied.
E: Release file for https://deb.debian.org/debian-security/dists/buster/updates/InRelease is not valid yet (invalid for another 17h 12min 53s). Updates for this repository will not be applied.
E: Release file for https://deb.debian.org/debian/dists/buster-backports/InRelease is not valid yet (invalid for another 15h 23min 52s). Updates for this repository will not be applied.
```

Joulinar, 2021-03-09:
Hi,

Looks like your system time is not correctly set. Please can you check to have an up-to-date system time.

MichaIng, 2021-03-09:
You have an NTFS, HFS or exFAT file system attached, right? Those require additional packages for basic I/O features, respectively for POSIX permission support in case of NTFS.

But yeah, the error is due to system clock being out of sync. The following should fix it: `/boot/dietpi/func/run_ntpd 1`

usmanmehmood55, 2021-03-10:
Hi. Thank you so much for the clarification. I fixed the time and the error has gone away.