MichaIng/DietPi #4185: Bitwarden_RS fresh install error message.

frontalgaming, 2021-03-09:
### Creating a bug report/issue

#### Required Information
- DietPi version | 7 0 2
- Distro version | buster
- Kernel version | Linux Manager 5.10.17-v7+ #1403 SMP Mon Feb 22 11:29:51 GMT 2021 armv7l GNU/Linux
- SBC model | RPI3
- Power supply used | Raspberry official PSU 5V 2A
- SDcard used | SAMSUNG FIT 64 GB

#### Additional Information (if applicable)
- Software title | Bitwarden_RS
- Was the software title installed freshly or updated/migrated? Fresh install.
- Can this issue be replicated on a fresh installation of DietPi? Didn't re-try it but this is a fresh install.

-No.

#### Steps to reproduce

1. Selected Bitwarden_RS on the software list
2. Message appeared at the start of installation.
3. [FAILED] DietPi-Software | Automatic latest Bitwarden_RS version detection failed. Version "1.19.0" will be installed as fallback, but a newer version might be available. Please report this at: https://github.com/MichaIng/DietPi/issues

#### Expected behaviour

- Find the latest version and install it.

#### Actual behaviour

-Defaults to an older version because it can't find the newest one of Bitwarden_RS
- Error message:
- [FAILED] DietPi-Software | Automatic latest Bitwarden_RS version detection failed. Version "1.19.0" will be installed as fallback, but a newer version might be available. Please report this at: https://github.com/MichaIng/DietPi/issues

#### Extra details

- Error message:
- [FAILED] DietPi-Software | Automatic latest Bitwarden_RS version detection failed. Version "1.19.0" will be installed as fallback, but a newer version might be available. Please report this at: https://github.com/MichaIng/DietPi/issues

Joulinar, 2021-03-09:
Hi,

Many thanks for your message. Does the installation failed completely or does it finished but with an older version?

frontalgaming, 2021-03-09:
I'm installing right now. It's taking quite a while. I'll report back once it's done.
Currently it's doing a bunch of compiling for over half an hour but isn't locked up or anything.
It's running hot and heavy.

frontalgaming, 2021-03-09:
After 77 minutes everything seems to be working properly. The installation finished and I can login to the webserver page and create an account.
Thanks again guys for this amazing project that helps people streamline these things.
Hope you find the bug.

frontalgaming, 2021-03-09:
dietpi@Manager:~$ curl -sSfL 'https://api.github.com/repos/dani-garcia/bitwarden_rs/releases/latest' | mawk -F\" '/"tag_name": /{print $4}'
1.19.0

Yep, it's working. The only reason I posted is because the installation process asked me to report it. Thanks for the help!

Joulinar, 2021-03-09:
no problem. Maybe there was a temp issue to detect the version and that's why you fall back to the hardcoded download link. If you are fine I'm going to close this issue. Feel free to reopen if needed.

closed 2021-03-09 by Joulinar

MichaIng, 2021-03-14:
Thanks for reporting btw. I added that warning to better keep track of broken GitHub API links or changed file names. Previously the fallback was used silently which means that it could have stayed undetected until someone verified it manually.

If this causes too many false alarms, we can remove it again. I'm anyway going through all GitHub API URLs before a new release since a while, and it can also be automated relatively easy.