MichaIng/DietPi #4186: No bluetooth audio anymore

TASSDevon, 2021-03-09:
### Creating a bug report/issue

#### Required Information
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=0
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | buster
- Kernel version | Linux skyspeaker 5.10.17+ #1403 Mon Feb 22 11:26:13 GMT 2021 armv6l GNU/Linux
- SBC model | RPi B+ (armv6l)

#### Additional Information (if applicable)
Somewhat related to #3611
That setup was running fine, I just updated to the latest dietpi and now I don't get audio on the bt speaker anymore

`root@skyspeaker:~# systemctl status bluealsa
● bluealsa.service - Bluealsa daemon
   Loaded: loaded (/etc/systemd/system/bluealsa.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2021-03-09 20:10:21 GMT; 11min ago
     Docs: https://github.com/Arkq/bluez-alsa/
 Main PID: 234 (bluealsa)
    Tasks: 6 (limit: 991)
   CGroup: /system.slice/bluealsa.service
           └─234 /usr/bin/bluealsa

Mar 09 20:12:12 skyspeaker bluealsa[234]: /usr/bin/bluealsa: D: ../../src/at.c:161: AT messa
ge: SET: command:+CSRSF, value:0,0,0,1,0,0,0
Mar 09 20:12:12 skyspeaker bluealsa[234]: /usr/bin/bluealsa: W: Unsupported AT message: SET:
 command:+CSRSF, value:0,0,0,1,0,0,0
Mar 09 20:12:12 skyspeaker bluealsa[234]: /usr/bin/bluealsa: D: ../../src/ba-rfcomm.c:108: S
ending AT message: RESP: command:(null), value:ERROR
Mar 09 20:12:12 skyspeaker bluealsa[234]: /usr/bin/bluealsa: D: ../../src/dbus.c:57: Called:
 org.bluez.MediaEndpoint1.SetConfiguration() on /org/bluez/hci0/A2DP/SBC/Source/1
Mar 09 20:12:12 skyspeaker bluealsa[234]: /usr/bin/bluealsa: D: ../../src/bluez.c:585: Confi
guration: Selected A2DP SBC bit-pool range: [2, 53]
Mar 09 20:12:12 skyspeaker bluealsa[234]: /usr/bin/bluealsa: D: ../../src/bluez.c:735: A2DP
Source (SBC) configured for device 10:B7:F6:28:24:7E
Mar 09 20:12:12 skyspeaker bluealsa[234]: /usr/bin/bluealsa: D: ../../src/bluez.c:737: Confi
guration: channels: 2, sampling: 48000
Mar 09 20:12:12 skyspeaker bluealsa[234]: /usr/bin/bluealsa: D: ../../src/ba-transport.c:721
: State transition: 0 -> 0
Mar 09 20:12:12 skyspeaker bluealsa[234]: /usr/bin/bluealsa: D: ../../src/bluez.c:931: Creat
ing media endpoint object: /org/bluez/hci0/A2DP/SBC/Source/3
Mar 09 20:12:12 skyspeaker bluealsa[234]: /usr/bin/bluealsa: D: ../../src/bluez.c:849: Regis
tering media endpoint: /org/bluez/hci0/A2DP/SBC/Source/3`

`root@skyspeaker:~# bluetoothctl info 10:B7:F6:28:24:7E
Device 10:B7:F6:28:24:7E (public)
	Name: Jabra Solemate Mini v8.0.0
	Alias: Jabra Solemate Mini v8.0.0
	Class: 0x00240404
	Icon: audio-card
	Paired: yes
	Trusted: no
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)`

#### Expected behaviour

Audio used to play through the bluetooth speaker

#### Actual behaviour
No audio can be heard through the bt speaker

MichaIng, 2021-03-09:
Many thanks for your report.

Can you show the update log that we especially see which packages were upgraded: `cat /var/tmp/dietpi/logs/dietpi-update.log`

> Paired: yes Trusted: no

Probably something can be done to trust the Bluetooth device.

Also please show the ALSA config so that we see that all is still as it should be:
```
for i in /etc/asound.conf /etc/alsa/conf.d/*; do [[ -f $i ]] && echo "$i: $(<$i)"; done
```

TASSDevon, 2021-03-10:
```
root@skyspeaker:~# cat /var/tmp/dietpi/logs/dietpi-update.log
[ SUB2 ] DietPi-Update > Applying pre-patches
[  OK  ] DietPi-Update | Downloading pre-patches
[  OK  ] DietPi-Update | Applying execute permission
[ INFO ] Pre-patch 25 | Running DietPi v7 pre-patches
[  OK  ] DietPi-Update | Successfully applied pre-patches
[ SUB2 ] DietPi-Update > Upgrading APT packages
[ INFO ] DietPi-Update | APT update, please wait...
Get:1 http://packages.microsoft.com/repos/code stable InRelease [10.4 kB]
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:3 https://archive.raspberrypi.org/debian buster InRelease
Get:4 http://packages.microsoft.com/repos/code stable/main amd64 Packages [16.3 kB]
Get:5 http://packages.microsoft.com/repos/code stable/main armhf Packages [16.8 kB]
Get:6 http://packages.microsoft.com/repos/code stable/main arm64 Packages [16.8 kB]
Fetched 60.3 kB in 8s (7205 B/s)
Reading package lists...
[  OK  ] DietPi-Update | APT update
[ INFO ] DietPi-Update | APT upgrade, please wait...
[  OK  ] DietPi-Update | APT upgrade
[ SUB2 ] DietPi-Update > Installing new DietPi code
[  OK  ] DietPi-Update | Downloading update archive
[  OK  ] DietPi-Update | Unpacking update archive
[  OK  ] DietPi-Update | Installing new DietPi scripts
[  OK  ] DietPi-Update | Installing new DietPi system files
[  OK  ] DietPi-Update | Setting execute permissions for DietPi scripts
[ SUB3 ] DietPi-Set_software > verify_dietpi.txt ()
[  OK  ] DietPi-Set_software | Checking URL: https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi.txt
[  OK  ] DietPi-Set_software | Downloading current dietpi.txt
[  OK  ] DietPi-Set_software | dietpi.txt verification completed
[  OK  ] verify_dietpi.txt  | Completed
[ SUB2 ] DietPi-Update > Applying incremental patches
[ INFO ] DietPi-Update | Current version : v6.30.0
[ INFO ] DietPi-Update | Latest version  : v7.0.2
[ INFO ] DietPi-Patch | Patching 6.30 to 6.31
[  OK  ] DietPi-Patch | chmod a+x /mnt/dietpi_userdata
[  OK  ] DietPi-Patch | mkdir /media
vm.swappiness = 1
net.ipv4.ping_group_range = 0 2147483647
[ INFO ] DietPi-Patch | Preparing SqueezeLite update...
removed '/usr/bin/squeezelite'
removed '/usr/bin/squeezelite_amd64'
removed '/usr/bin/squeezelite_arm64'
removed '/usr/bin/squeezelite_armv6'
removed '/usr/bin/squeezelite_armv7'
[  OK  ] DietPi-Patch | Added setting User=squeezelite to /etc/systemd/system/squeezelite.service after line [Service]
[ INFO ] DietPi-Patch | Patching 6.31 to 6.32
removed '/etc/apt/apt.conf.d/99-dietpi-norecommends'
removed '/etc/apt/apt.conf.d/98-dietpi-no_translations'
removed '/etc/apt/apt.conf.d/99-dietpi-forceconf'
[  OK  ] DietPi-Patch | Removing deprecated "max_usb_current" from config.txt
[ INFO ] DietPi-Patch | Patching 6.32 to 6.33
[ INFO ] DietPi-Patch | Patching 6.33 to 6.34
[ SUB4 ] DietPi-Set_hardware > gpumemsplit (16)
[  OK  ] DietPi-Set_hardware | Desired setting in /boot/config.txt was already set: gpu_mem_1024=16
[  OK  ] DietPi-Set_hardware | Desired setting in /boot/config.txt was already set: gpu_mem_512=16
[  OK  ] DietPi-Set_hardware | Desired setting in /boot/config.txt was already set: gpu_mem_256=16
[  OK  ] DietPi-Set_hardware | eval echo 'blacklist vc_sm_cma' > /etc/modprobe.d/dietpi-disable_vcsm.conf
[  OK  ] DietPi-Set_hardware | Compiling device tree overlay: /boot/overlays/dietpi-disable_vcsm.dtbo
[  OK  ] DietPi-Set_hardware | Added setting dtoverlay=dietpi-disable_vcsm to end of file /boot/config.txt
[  OK  ] gpumemsplit 16 | Completed
[ INFO ] DietPi-Patch | Patching 6.34 to 6.35
[  OK  ] DietPi-Patch | rm -f /etc/apt/sources.list.d/vscode.list /etc/apt/trusted.gpg.d/microsoft.gpg /etc/apt/preferences.d/3rd_parties.pref
[  OK  ] DietPi-Patch | rmdir --ignore-fail-on-non-empty /var/lib/dietpi/dietpi-software/services
[  OK  ] DietPi-Software | Initialised database
[  OK  ] DietPi-Software | Reading database
[ SUB4 ] DietPi-Software > Automated reinstall
[  OK  ] DietPi-Software | Reinstalling Squeezelite: audio player for lms & squeezebox
[ INFO ] DietPi-Software | 108: Amiberry is not currently installed
[ INFO ] DietPi-Software | Use "dietpi-software install 108" to install Amiberry.
[... 451 lines cut ...]
ctl.!default {
    type pulse
}
/etc/alsa/conf.d/99-pulse.conf: # PulseAudio alsa plugin configuration file to set the pulseaudio plugin as
# default output for applications using alsa when pulseaudio is running.
hook_func.pulse_load_if_running {
	lib "libasound_module_conf_pulse.so"
	func "conf_pulse_hook_load_if_running"
}

@hooks [
	{
		func pulse_load_if_running
		files [
			"/usr/share/alsa/pulse-alsa.conf"
		]
		errors false
	}
]

```

As for the speaker being labeled trusted, I have no idea if it was like this in the past or not. the whole setup was up and running for months (without downtime of either the pi or the speaker).

Anyway, hope this info helps!

MichaIng, 2021-03-10:
Ah, the Squeezelite reinstall re-created `/etc/asound.conf` which should be removed: `rm /etc/asound.conf`

TASSDevon, 2021-03-12:
I removed the config file and rebooted, still no BT audio though. Anywhere I need to specify which audio device to send output to?

MichaIng, 2021-03-12:
Just searched through the old pages. Indeed not no but the following config should do it:
```
echo -e 'pcm.!default bluealsa\nctl.!default bluealsa' > /etc/asound.conf
```

TASSDevon, 2021-03-13:
Thanks, I tried and Squeezelite is exiting with

```
root@skyspeaker:~# cat /var/log/squeezelite.log
[19:12:42.540908] test_open:281 playback open error: No such file or directory
[19:12:42.541984] output_init_common:391 unable to open output device: default
```

MichaIng, 2021-03-13:
Strange, `/etc/alsa/conf.d/20-bluealsa.conf` defines that device name 🤔.

Does `alsamixer -D bluealsa` work?
Or `aplay -D bluealsa /path/to/some.wav`?

TASSDevon, 2021-03-14:
The mixer works, when trying to play a wav file the output is

```
root@skyspeaker:~# aplay -D bluealsa test.wav
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5036:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM bluealsa
aplay: main:828: audio open error: No such file or directory
```

MichaIng, 2021-03-14:
It is all right there in /etc/alsa/conf.d/50-bluealsa.conf:
```
pcm.bluealsa {
	@args [ SRV DEV PROFILE DELAY ]
	@args.SRV {
		type string
		default {
			@func refer
			name defaults.bluealsa.service
		}
	}
	@args.DEV {
		type string
		default {
			@func refer
			name defaults.bluealsa.device
		}
	}
```

Okay three things: I think pcm/ctl prefix is required as well to override the default device:
```
echo -e 'pcm.!default pcm.bluealsa\nctl.!default pcm.bluealsa' > /etc/asound.conf
```
When playing audio, it's actually not required, but just to assure:
```
aplay -D pcm.bluealsa test.wav
```

I see PulseAudio is installed. Not sure if it might interfere, so, if it's not a package dependency for something else, could you try to remove it?
```
apt purge pulseaudio
apt autopurge
```

You compiled BlueZ-ALSA manually that time: https://github.com/MichaIng/DietPi/issues/3611#issuecomment-647008737
Could you try to re-compile, or otherwise try to use/reinstall the package from the RPi repository: `apt install --reinstall bluealsa`

TASSDevon, 2021-03-14:
Updated the asound.conf file, purged pulseaudio and reinstalled bluealsa, seemed to have no effect:

```
root@skyspeaker:~# aplay -D pcm.bluealsa test.wav
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5036:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM pcm.bluealsa
aplay: main:828: audio open error: No such file or directory
```

MichaIng, 2021-03-14:
`ls -l /run/bluealsa/` contains some files? Generally if the `bluealsa.service` does not do it's job, this error seems to be the result: https://github.com/Arkq/bluez-alsa/issues/26

Did you try to recompile the newest version from sources? Probably it contains some required changes to work with Linux 5.10.

Otherwise it could be tested to downgrade the kernel:
```
apt install rpi-update
rpi-update b7c8ef64ea24435519f05c38a2238658908c038e
```

TASSDevon, 2021-03-15:
I pulled the latest code from the repo and rebuilt bluealsa, then also added another asoundrc file, this was key actually. I think somehow the rc file got removed and that was causing the issues. No harm done in having the latest greatest bluealsa in the process I guess :)

Thanks @MichaIng for your help, again 👍

MichaIng, 2021-03-15:
Great, many thanks for the feedback. How long does compiling take? When we add bluealsa support to our scripts, maybe we should either compile it from sources or ship own updated packages.

TASSDevon, 2021-03-16:
It took a couple of minutes, but then again it was running on a pi1B

MichaIng, 2021-03-16:
Okay, sounds acceptable 👍.

closed 2021-03-29 by TASSDevon