MichaIng/DietPi #4187: Offtopic CPU_USAGE_THROTTLE_UP

up2smoke, 2021-03-10:
is it possible to use the CPU_USAGE_THROTTLE_UP function outside of Dietpi?
I already have a running system (nextcloudPi & pi-hole on RaspiOS Lite) and have seen DietPi on Facebook and YouTube.
I am satisfied with my system, but being able to upgrade the CPU_USAGE_THROTTLE_UP function would be nice.

MichaIng, 2021-03-10:
Many thanks for your question.

Hmm, I'm not quite sure which function you mean. There are a few DietPi-only console functions, like:
```
G_OBTAIN_CPU_TEMP
G_OBTAIN_CPU_USAGE
```
But none that is related to the throttle up break point, if that is what you mean, i.e. the CPU usage at which the CPU frequency is raised when using `ondemand` or `conservative` CPU governor.

In `dietpi.txt`, we have the `CONFIG_CPU_USAGE_THROTTLE_UP` setting, which allows to define that break point which is then applied automatically on boot by a DietPi-only script. But it's pretty simple to apply those settings manually on every Linux distro:
- `/sys/devices/system/cpu/cpufreq/policy0/scaling_governor` allows to read and set the CPU governor. A throttle up break point is only used by `ondemand` and `conservative`.
- `/sys/devices/system/cpu/cpufreq/ondemand/up_threshold` allows to apply the break point (in % CPU usage) for ondemand governor.
- `/sys/devices/system/cpu/cpufreq/conservative/up_threshold` allows to do that for the conservative governor.

However, actually on modern Linux systems, I recommend to use the `schedutil` governor which works closer with the CPU scheduler and does frequency adjustments quicker. On systems with Linux 5.X at least it makes the system more responsive, applying higher frequencies faster, but throttles down much faster as well to reduce overall power usage and heat emission. That scheduler does not take/need any settings but works pretty fine OOTB.

up2smoke, 2021-03-10:
yes, that answers my question and thank you very much for that!  I will test how I can use this for myself

closed 2021-03-10 by up2smoke