MichaIng/DietPi #4188: Hard lockup on USB drive insert/remove

TheOriginalMrWolf, 2021-03-11:
Hi,

Running on Odroid N2, I experience hard lockups (console unresponsive, no response to ping on LAN nor any other network traffic, SSH connections dropped) when using USB drives.

This can occur either randomly on install/remove of a drive, or after the drive has simply been plugged in for a few hours/days.  Occurs if drive is either plugged directly into SBC, or via a powered Pluggable USB3-Hub7C.

Was connected via SSH & running `journalctl -f` when this happened and saw:
```
Mar 11 18:12:25 NewPi kernel: sd 0:0:0:1: [sdb] Attached SCSI removable disk
Mar 11 18:12:29 NewPi kernel: usb 2-1.2.3: usb_reset_and_verify_device Failed to disable LTM
                              .
Mar 11 18:12:29 NewPi kernel: usb 1-1.2.3: new high-speed USB device number 5 using xhci-hcd
Mar 11 18:12:30 NewPi kernel: usb 2-1.2.3: USB disconnect, device number 17
Mar 11 18:12:30 NewPi kernel: amlogic-new-usb2-v2 ffe09000.usb2phy: ---Recovery port(2) tuning for host cf(hub_event)--
Mar 11 18:12:30 NewPi kernel: Unable to handle kernel NULL pointer dereference at virtual address 00000038

Message from syslogd@NewPi at Mar 11 18:12:31 ...
 kernel:[ 6740.706919@0] Internal error: Oops: 96000047 [#1] PREEMPT SMP
```

**_Is there any way to better capture debug information?_**

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=0
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
Buster / 10.8

- Kernel version | `uname -a`
Linux NewPi 4.9.241-arm64 #1 SMP PREEMPT Thu Feb 25 18:56:07 CET 2021 aarch64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
Odroid N2 (aarch64)

- Power supply used | (EG: 5V 1A RAVpower)
Odroid supplied PSU

- SDcard used | (EG: SanDisk ultra)
Sandisk Ultra in Kingston MobileLite G4 card reader
Samsung T5 Portable SSD

#### Additional Information (if applicable)

#### Steps to reproduce

1. Power on SBC
2. Plug in USB drive
3. Wait for a number of hours / days
2. Observe hard lockup

or

1. Power on SBC
2. Repeatedly plug / unplug USB drive
3. Wait an hour or two
4. Goto step 2 until hard lockup observed

#### Expected behaviour

- SBC with DietPi should continue running & being available

#### Actual behaviour

- Hard lockup.  Console unresponsive (just blinking cursor), no messages.  Stops responding to ping or any other network traffic, drops SSH sessions.

#### Extra details

- N/A

MichaIng, 2021-03-11:
Many thanks for your report.

Are there other kernel errors, before the actual crash happens?
```
dmesg -l emerg,alert,crit,err
```

Just in case some kernel files are broken, please try to reinstall it:
```
apt install --reinstall linux-image-4.9.241-arm64 linux-firmware-image-4.9.241-arm64
```

TheOriginalMrWolf, 2021-03-12:
Hi,

Thanks for the quick response!

I reinstalled the kernel as requested.

Rebooted & plugged in the USB drive and waited about six hours before the unit locked up again.

I was running `journalctl -t` to a file on SD card, but didn't capture anything of interest.  Can you possibly suggest a better way to try to capture errors?

Kernel errors:
```
dmesg -l emerg,alert,crit,err
[    0.000000]  7f800000 - 80000000,     8192 KB, linux,meson-fb
[    0.000000]  e5800000 - ed800000,   131072 KB, linux,ion-dev
[    0.000000]  e3000000 - e5800000,    40960 KB, linux,di_cma
[    0.000000]  e3000000 - e3000000,        0 KB, linux,ppmgr
[    0.000000]  cfc00000 - e3000000,   315392 KB, linux,codec_mm_cma
[    0.000000]  cfc00000 - cfc00000,        0 KB, linux,codec_mm_reserved
[    0.000000]  05000000 - 05400000,     4096 KB, linux,secmon
[    0.320770] codec_mm_module_init
[    0.328534] clkmsr ffd18004.meson_clk_msr: failed to get msr ring reg0
[    0.344480] cvbs_out: chrdev devno 264241152 for disp
[    0.468456] dmi: Firmware registration failed.
[    0.897850] meson-pwm ff802000.pwm: pwm pinmux : can't get pinctrl
[    0.898065] meson-pwm ffd1b000.pwm: pwm pinmux : can't get pinctrl
[    0.902554] mtdoops: mtd device (mtddev=name/number) must be supplied
[    0.920725] meson_cpufreq_init:don't find the node <dynamic_gp1_clk>
[    0.922760] meson_cpufreq_init:don't find the node <dynamic_gp1_clk>
[    0.924779] ff803000.serial: clock gate not found
[    3.345880] meson-fb meson-fb: create ion_client ffffffc0c9ff4840, handle=ffffffc0ca333ec0
[    3.345883] meson-fb meson-fb: ion memory(0): created fb at 0x00000000e5800000, size 75 MiB
[    3.686541] di_get_vpu_clkb: get clk vpu error.
[    3.891983] Reserved memory: failed to init DMA memory pool at 0x00000000e3000000, size 0 MiB
[    4.116466] meson-mmc: >>>>>>>>hostbase ffffff800858d000, dmode
[    4.179973] meson-mmc: >>>>>>>>hostbase ffffff8008596000, dmode
[    4.180106] meson-mmc: emmc: resp_timeout,vstat:0x9dff0800,virqc:3fff
[    4.180109] meson-mmc: emmc: err: wait for desc write back, bus_fsm:0x7
[    4.180119] meson-mmc: meson_mmc_irq_thread_v3() 653: set 1st retry!
[    4.180120] meson-mmc: retry cmd 1 the 3-th time(s)
[    4.189809] meson-mmc: emmc: resp_timeout,vstat:0x9dff0800,virqc:3fff
[    4.189810] meson-mmc: emmc: err: wait for desc write back, bus_fsm:0x7
[    4.189818] meson-mmc: retry cmd 1 the 2-th time(s)
[    4.196571] meson-mmc: emmc: resp_timeout,vstat:0x9dff0800,virqc:3fff
[    4.196573] meson-mmc: emmc: err: wait for desc write back, bus_fsm:0x7
[    4.196580] meson-mmc: retry cmd 1 the 1-th time(s)
[    4.203066] meson-mmc: emmc: resp_timeout,vstat:0x9dff0800,virqc:3fff
[    4.203068] meson-mmc: emmc: err: wait for desc write back, bus_fsm:0x7
[    4.203076] meson-mmc: Command retried failed line:675, cmd:1
[    4.304669] cectx ff80023c.aocec: cec driver date:2019/10/22: finetune ARB rising time

[    4.313493] cectx ff80023c.aocec: compatible:amlogic, aocec-g12a
[    4.319521] cectx ff80023c.aocec: cecb_ver:0x1
[    4.324101] cectx ff80023c.aocec: line_reg:0x1
[    4.328701] cectx ff80023c.aocec: line_bit:0x3
[    4.333287] cectx ff80023c.aocec: ee_to_ao:0x1
[    4.343835] cectx ff80023c.aocec: not find 'port_num'
[    4.348979] cectx ff80023c.aocec: using cec:1
[    4.353491] cectx ff80023c.aocec: no hdmirx regs
[    4.358251] cectx ff80023c.aocec: no hhi regs
[    4.362757] cectx ff80023c.aocec: not find 'output'
[    4.371315] cectx ff80023c.aocec: wakeup_reason:0x0
[    4.376230] cectx ff80023c.aocec: cev val1: 0x0;val2: 0x0
[    4.381736] cectx ff80023c.aocec: aml_cec_probe success end
[    4.511085] meson-mmc: card IN
[    4.524195] defendkey ff630218.defendkey: Reserved memory is not enough!
[    4.549845] Error: Driver 'spdif-dit' is already registered, aborting...
[    4.891243] aml_codec_T9015 ff632000.t9015: ASoC: no source widget found for Left DAC
[    4.891245] aml_codec_T9015 ff632000.t9015: ASoC: Failed to add route Left DAC -> LOLP_SEL_DACL -> Lineout left P switch
[    4.891247] aml_codec_T9015 ff632000.t9015: ASoC: no source widget found for Left DAC
[    4.891249] aml_codec_T9015 ff632000.t9015: ASoC: Failed to add route Left DAC -> LOLP_SEL_DACL_INV -> Lineout left P switch
[    4.891251] aml_codec_T9015 ff632000.t9015: ASoC: no source widget found for Left DAC
[    4.891253] aml_codec_T9015 ff632000.t9015: ASoC: Failed to add route Left DAC -> LOLN_SEL_DACL_INV -> Lineout left N switch
[    4.891256] aml_codec_T9015 ff632000.t9015: ASoC: no source widget found for Left DAC
[    4.891257] aml_codec_T9015 ff632000.t9015: ASoC: Failed to add route Left DAC -> LOLN_SEL_DACL -> Lineout left N switch
[    4.891259] aml_codec_T9015 ff632000.t9015: ASoC: no source widget found for Right DAC
[    4.891261] aml_codec_T9015 ff632000.t9015: ASoC: Failed to add route Right DAC -> LORP_SEL_DACR -> Lineout right P switch
[    4.891263] aml_codec_T9015 ff632000.t9015: ASoC: no source widget found for Right DAC
[    4.891264] aml_codec_T9015 ff632000.t9015: ASoC: Failed to add route Right DAC -> LORP_SEL_DACR_INV -> Lineout right P switch
[    4.891266] aml_codec_T9015 ff632000.t9015: ASoC: no source widget found for Right DAC
[    4.891268] aml_codec_T9015 ff632000.t9015: ASoC: Failed to add route Right DAC -> LORN_SEL_DACR_INV -> Lineout right N switch
[    4.891270] aml_codec_T9015 ff632000.t9015: ASoC: no source widget found for Right DAC
[... cut ...] right N switch
[    4.892975] aml_codec_T9015 ff632000.t9015: ASoC: mux Lineout left P switch has no paths
[    4.892979] aml_codec_T9015 ff632000.t9015: ASoC: mux Lineout left N switch has no paths
[    4.892985] aml_codec_T9015 ff632000.t9015: ASoC: mux Lineout right P switch has no paths
[    4.892988] aml_codec_T9015 ff632000.t9015: ASoC: mux Lineout right N switch has no paths
[    5.052168] pm-meson pm: Can't get switch_clk81
[    6.496741] cgroup: cgroup2: unknown option "nsdelegate"

```

MichaIng, 2021-03-12:
Hmm, I don't like kernels which print "expected" error messages as now it's very hard to know if there is an actual issue or not... One thing could be muted by disabling CEC support.

I did just create/update the boot.ini for N2: https://github.com/MichaIng/DietPi/blob/dev/boot_n2.ini
You can install it on your system:
```
curl -sSfL https://raw.githubusercontent.com/MichaIng/DietPi/dev/boot_n2.ini -o /boot/boot.ini
```
Then CEC can be disabled:
```
G_CONFIG_INJECT 'setenv cec[[:blank:]]' 'setenv cec "false"' /boot/boot.ini
```

Remembering the initial reason for the boot.ini update: Is it actually an Odroid N2 or N2+? The latter was not really supported but with this new boot.ini it is as it loads a different device tree. Maybe this is even the reason for all those kernel errors.

TheOriginalMrWolf, 2021-03-14:
Hi @MichaIng,

Thanks for that.

The devices are definitely an N2, not N2+.  I bought two of them over a year ago (and both are affected - which sort of rules out faulty hardware/PSU issues).

Followed your instructions above (copy/pasted both & confirmed they worked), unfortunately the system is still crashing.  Had `journalctl -f` running again and this time got:

```
Mar 14 14:25:57 NewPi kernel: sd 0:0:0:1: [sdb] Attached SCSI removable disk
Mar 14 14:26:02 NewPi kernel: usb 2-1.2.1.3: USB disconnect, device number 11
Mar 14 14:26:10 NewPi kernel: amlogic-new-usb2-v2 ffe09000.usb2phy: ---Recovery port(2) tuning for host cf(hub_event)--
Mar 14 14:26:10 NewPi kernel: Unable to handle kernel NULL pointer dereference at virtual address 00000038
Mar 14 14:26:10 NewPi kernel: pgd = ffffffc0c92dd000
Mar 14 14:26:11 NewPi kernel: [0000000000000038] *pgd=0000000000000000, *pud=0000000000000000
Mar 14 14:26:11 NewPi kernel: Internal error: Oops: 96000045 [#1] PREEMPT SMP
Mar 14 14:26:12 NewPi kernel: Modules linked in: xt_nat xt_tcpudp xt_conntrack ipt_MASQUERADE nf_nat_masquerade_ipv4 nf_conntrack_netlink xt_addrtype iptable_filter iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack br_netfilter bridge stp llc overlay i2c_meson_master mali_kbase ip_tables x_tables ipv6
Mar 14 14:26:12 NewPi kernel: CPU: 0 PID: 12176 Comm: kworker/0:0 Not tainted 4.9.241-arm64 #1
Mar 14 14:26:12 NewPi kernel: Hardware name: Hardkernel ODROID-N2 (DT)
Mar 14 14:26:12 NewPi kernel: Workqueue: usb_hub_wq hub_event
Mar 14 14:26:12 NewPi kernel: task: ffffffc0ca2c1c00 task.stack: ffffffc0ae9ac000
Mar 14 14:26:12 NewPi kernel: PC is at set_usb_phy_host_tuning+0xf4/0x100
Mar 14 14:26:12 NewPi kernel: LR is at set_usb_phy_host_tuning+0xec/0x100
Mar 14 14:26:12 NewPi kernel: pc : [<ffffff800977f2a4>] lr : [<ffffff800977f29c>] pstate: 00000145
Mar 14 14:26:12 NewPi kernel: sp : ffffffc0ae9afc20
Mar 14 14:26:12 NewPi kernel: x29: ffffffc0ae9afc20 x28: ffffffc03d3b4400
Mar 14 14:26:12 NewPi kernel: x27: ffffffc03d3b5308 x26: 0000000000000003
Mar 14 14:26:12 NewPi kernel: x25: ffffffc0c80e2000 x24: 0000000000000002
Mar 14 14:26:12 NewPi kernel: x23: 0000000000000000 x22: 0000000000000001
Mar 14 14:26:12 NewPi kernel: x21: 0000000000000002 x20: ffffff80095d463c
Mar 14 14:26:12 NewPi kernel: x19: ffffff800a816000 x18: 0000000000000001
Mar 14 14:26:12 NewPi kernel: x17: 0000007f9fe71208 x16: ffffff800924eb90
Mar 14 14:26:12 NewPi kernel: x15: ffffffffffffffff x14: 676e696e75742029
Mar 14 14:26:12 NewPi kernel: x13: 322874726f702079 x12: 7265766f6365522d
Mar 14 14:26:12 NewPi kernel: x11: 2d2d203a79687032 x10: 00000000000004e7
Mar 14 14:26:12 NewPi kernel: x9 : 306566662032762d x8 : 6275682866632074
Mar 14 14:26:12 NewPi kernel: x7 : 736f6820726f6620 x6 : ffffff800a736042
Mar 14 14:26:12 NewPi kernel: x5 : 0000000000000000 x4 : 0000000000000000
Mar 14 14:26:12 NewPi kernel: x3 : 0000000000000000 x2 : ffffffc0ca2c1c00
Mar 14 14:26:12 NewPi kernel: x1 : ffffffc0ca2c1c00 x0 : 0000000000000038
Mar 14 14:26:12 NewPi kernel:
                              SP: 0xffffffc0ae9afba0:
Mar 14 14:26:12 NewPi kernel: fba0  00000001 00000000 00000000 00000000 00000002 00000000 c80e2000 ffffffc0
Mar 14 14:26:12 NewPi kernel: fbc0  00000003 00000000 3d3b5308 ffffffc0 3d3b4400 ffffffc0 ae9afc20 ffffffc0
Mar 14 14:26:12 NewPi kernel: fbe0  0977f29c ffffff80 ae9afc20 ffffffc0 0977f2a4 ffffff80 00000145 00000000
Mar 14 14:26:12 NewPi kernel: fc00  095d463c ffffff80 ffffffff 00ffffff ffffffff ffffffff 00000000 00000000
Mar 14 14:26:12 NewPi kernel: fc20  ae9afc60 ffffffc0 095d463c ffffff80 00000001 00000000 ffffff95 00000000
Mar 14 14:26:12 NewPi kernel: fc40  ffffff95 00000000 00000000 00000000 c9e64800 ffffffc0 3d3b5008 ffffffc0
Mar 14 14:26:12 NewPi kernel: fc60  ae9afd70 ffffffc0 090c39d4 ffffff80 3d3b4608 ffffffc0 ca356000 ffffffc0
Mar 14 14:26:12 NewPi kernel: fc80  cf3230c0 ffffffc0 cf328b00 ffffffc0 00000000 00000000 0a577000 ffffff80
Mar 14 14:26:12 NewPi kernel:
                              X1: 0xffffffc0ca2c1b80:
Mar 14 14:26:12 NewPi kernel: 1b80  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Mar 14 14:26:12 NewPi kernel: 1ba0  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Mar 14 14:26:12 NewPi kernel: 1bc0  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Mar 14 14:26:12 NewPi kernel: 1be0  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Mar 14 14:26:12 NewPi kernel: 1c00  00000002 00000000 ffffffff ffffffff 00000001 00000000 00000000 00000000
Mar 14 14:26:12 NewPi kernel: 1c20  ae9ac000 ffffffc0 00000002 0420c060 00000000 00000000 00000000 00000000
Mar 14 14:26:12 NewPi kernel: 1c40  00000001 00000000 00000009 00000000 0022ec14 00000001 c9e76200 ffffffc0
Mar 14 14:26:12 NewPi kernel: 1c60  00000000 00000001 00000078 00000078 00000078 00000000 09bd9498 ffffff80
Mar 14 14:26:12 NewPi kernel:
                              X2: 0xffffffc0ca2c1b80:
Mar 14 14:26:12 NewPi kernel: 1b80  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Mar 14 14:26:12 NewPi kernel: 1ba0  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Mar 14 14:26:12 NewPi kernel: 1bc0  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Mar 14 14:26:12 NewPi kernel: 1be0  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[... cut ...]
[    4.875484] aml_codec_T9015 ff632000.t9015: ASoC: Failed to add route Right DAC -> LORN_SEL_DACR -> Lineout right N switch
[    4.877105] aml_codec_T9015 ff632000.t9015: ASoC: mux Lineout left P switch has no paths
[    4.877109] aml_codec_T9015 ff632000.t9015: ASoC: mux Lineout left N switch has no paths
[    4.877112] aml_codec_T9015 ff632000.t9015: ASoC: mux Lineout right P switch has no paths
[    4.877115] aml_codec_T9015 ff632000.t9015: ASoC: mux Lineout right N switch has no paths
[    5.036371] pm-meson pm: Can't get switch_clk81
[    6.961185] cgroup: cgroup2: unknown option "nsdelegate"
```

MichaIng, 2021-03-14:
CEC errors are still there, which makes sense as CEC is not disabled by boot.ini but only not actively enabled. I.e. it was disabled before and I guess the error is coming from a kernel module which cannot find the "hdmirx" option.
Can you paste: `lsmod`

The other kernel errors can all be found in various other cases on Odroid and Armbian in combination with the Hardkernel kernel. Not beautiful (IMO) but likely not related .

Can you try to clock down the CPUs a bit? I don't think it helps, but it does not hurt to try it out:
```
sed -i 's/\${a73_freq}/\${a73_freq\]/' /boot/boot.ini # A typo in my previous boot.ini
G_CONFIG_INJECT 'setenv max_freq_a73[[:blank:]]' 'setenv max_freq_a73 "1608"' /boot/boot.ini
G_CONFIG_INJECT 'setenv max_freq_a53[[:blank:]]' 'setenv max_freq_a53 "1608"' /boot/boot.ini
```

TheOriginalMrWolf, 2021-03-16:
Hi @MichaIng,

Applied the above as requested.

`lsmod`:
```
Module                  Size  Used by
xt_nat                 16384  3
xt_tcpudp              16384  9
xt_conntrack           16384  1
ipt_MASQUERADE         16384  4
nf_nat_masquerade_ipv4    16384  1 ipt_MASQUERADE
nf_conntrack_netlink    45056  0
xt_addrtype            16384  2
iptable_filter         16384  1
iptable_nat            16384  1
nf_conntrack_ipv4      20480  2
nf_defrag_ipv4         16384  1 nf_conntrack_ipv4
nf_nat_ipv4            16384  1 iptable_nat
nf_nat                 32768  3 xt_nat,nf_nat_masquerade_ipv4,nf_nat_ipv4
nf_conntrack          143360  6 nf_conntrack_ipv4,nf_conntrack_netlink,nf_nat_masquerade_ipv4,xt_conntrack,nf_nat_ipv4,nf_nat
br_netfilter           24576  0
bridge                155648  1 br_netfilter
stp                    16384  1 bridge
llc                    16384  2 bridge,stp
overlay                53248  2
mali_kbase            462848  0
i2c_meson_master       20480  0
ip_tables              32768  2 iptable_filter,iptable_nat
x_tables               49152  7 xt_nat,ip_tables,iptable_filter,xt_tcpudp,ipt_MASQUERADE,xt_addrtype,xt_conntrack
ipv6                  462848  95 bridge
```

Noticed that I can trigger the lockup if I insert the USB card reader slowly (presumably not a 'clean' insert).  Not sure if this is related or another issue?  Surely we shouldn't have a crash if a USB device has some issues on insert?!?

`journalctl` output:
```
[ 6876.579899] amlogic-new-usb2-v2 ffe09000.usb2phy: ---Recovery port(2) tuning for host cf(hub_event)--
[ 6876.584736] Unable to handle kernel NULL pointer dereference at virtual address 00000038
[ 6876.598233] pgd = ffffffc0a679c000
[ 6876.598317] [0000000000000038] *pgd=0000000000000000, *pud=0000000000000000
[ 6876.604966] Internal error: Oops: 96000045 [#1] PREEMPT SMP

Message from syslogd@NewPi at Mar 16 12:09:53 ...
 kernel:[ 6876.604966@0] Internal error: Oops: 96000045 [#1] PREEMPT SMP
[ 6876.609048] Modules linked in: xt_nat xt_tcpudp xt_conntrack ipt_MASQUERADE nf_nat_masquerade_ipv4 nf_conntrack_netlink xt_addrtype iptable_filter iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack br_netfilter bridge stp llc overlay mali_kbase i2c_meson_master ip_tables x_tables ipv6
[ 6876.636618] CPU: 0 PID: 8355 Comm: kworker/0:2 Not tainted 4.9.241-arm64 #1
[ 6876.643707] Hardware name: Hardkernel ODROID-N2 (DT)
[ 6876.648863] Workqueue: usb_hub_wq hub_event
[ 6876.653157] task: ffffffc0ca2cd400 task.stack: ffffffc0a658c000
[ 6876.659235] PC is at set_usb_phy_host_tuning+0xf4/0x100
[ 6876.664600] LR is at set_usb_phy_host_tuning+0xec/0x100
[ 6876.669977] pc : [<ffffff800977f2a4>] lr : [<ffffff800977f29c>] pstate: 00000145
[ 6876.677506] sp : ffffffc0a658fc20
[ 6876.680974] x29: ffffffc0a658fc20 x28: ffffffc03d3a5c00
[ 6876.686433] x27: ffffffc03d3a5308 x26: 0000000000000003
[ 6876.691893] x25: ffffffc0b3225800 x24: 0000000000000002
[ 6876.697353] x23: 0000000000000000 x22: 0000000000000001
[ 6876.702812] x21: 0000000000000002 x20: ffffff80095d463c
[ 6876.708273] x19: ffffff800a816000 x18: 0000000000000001
[ 6876.713733] x17: 0000007f919dcbb8 x16: ffffff800913bbf8
[ 6876.719193] x15: ffffffffffffffff x14: 736f6820726f6620
[ 6876.724653] x13: 676e696e75742029 x12: 322874726f702079
[ 6876.730113] x11: 7265766f6365522d x10: 000000000000042a
[ 6876.735573] x9 : 6273752e30303039 x8 : 2d29746e6576655f
[ 6876.741033] x7 : 6275682866632074 x6 : ffffff800a736042
[ 6876.746492] x5 : 0000000000000000 x4 : 0000000000000000
[ 6876.751952] x3 : 0000000000000000 x2 : ffffffc0ca2cd400
[ 6876.757413] x1 : ffffffc0ca2cd400 x0 : 0000000000000038
[ 6876.762878]
[... 132 lines cut ...]
Message from syslogd@NewPi at Mar 16 12:09:53 ...
 kernel:[ 6877.408551@0] fe20: 0000000000000000 ffffff80090836c0 ffffff80090ca840 ffffffc03d2ad0c0
[ 6877.530751] Exception stack(0xffffffc0a658fa50 to 0xffffffc0a658fb80)
[ 6877.537336] fa40:                                   ffffff800a816000 0000007fffffffff
[ 6877.545312] fa60: ffffffc0a658fc20 ffffff800977f2a4 306566663a6d726f 6273752e30303039
[ 6877.553285] fa80: 0000000079687032 0000000080000400 ffffffc0a658fae0 ffffff80095d91d8
[ 6877.561258] faa0: ffffffc0a4113000 ffffffc0a658fb48 0000000000000000 ffffffc0a658fbdc
[ 6877.569231] fac0: 00000000000003e8 0000000080000400 ffffffc0a658fb70 ffffff80095230e0
[ 6877.577205] fae0: ffffffc0ca14b810 ffffff8009ee6938 0000000000000038 ffffffc0ca2cd400
[ 6877.585177] fb00: ffffffc0ca2cd400 0000000000000000 0000000000000000 0000000000000000
[ 6877.593152] fb20: ffffff800a736042 6275682866632074 2d29746e6576655f 6273752e30303039

Message from syslogd@NewPi at Mar 16 12:09:53 ...
 kernel:[ 6877.424497@0] fe60: ffffffc0ca3ff000 0000000000000000 0000000000000000 ffffffc0bb7720c0
[ 6877.601125] fb40: 000000000000042a 7265766f6365522d 322874726f702079 676e696e75742029
[ 6877.609098] fb60: 736f6820726f6620 ffffffffffffffff ffffff800913bbf8 0000007f919dcbb8
[ 6877.617076] [<ffffff800977f2a4>] set_usb_phy_host_tuning+0xf4/0x100
[ 6877.623492] [<ffffff80095d463c>] hub_event+0xc2c/0x1090
[ 6877.628865] [<ffffff80090c39d4>] process_one_work+0x1fc/0x530
[ 6877.634750] [<ffffff80090c3d60>] worker_thread+0x58/0x4e8
[ 6877.640303] [<ffffff80090ca924>] kthread+0xe4/0xf8
[ 6877.645241] [<ffffff80090836c0>] ret_from_fork+0x10/0x50
[ 6877.650703] Code: 91232021 97f6907e d5033e9f 9100e2e0 (b900001f)
[ 6877.656941] ---[ end trace 7440156809e6e474 ]---
```

TheOriginalMrWolf, 2021-03-16:
Also, another interesting observation, I implemented a hardware watchdog as per https://wiki.odroid.com/odroid-n2/application_note/software/watchdog_timer.  This works for the test scenarios however does _not_ work for the above (system dies & remains unavailable).

MichaIng, 2021-03-16:
Dammit, I had a typo in the boot.ini fix 😄. Please try again with:
```
sed -i 's/\${a73_freq\]/\${a73_freq}/' /boot/boot.ini
reboot
cpu
```
The last command, after a reboot, should then show a max CPU frequency on all cores lowered to 1608 MHz. However, I do not expect this to help, just to rule it out.

@meveric
Do you have an idea what the issue might be here, or how to further debug? Looks like something to forward to Hardkernel, either kernel or hardware issue. The kernel didn't get any fix since your last build: https://github.com/hardkernel/linux/commits/odroidg12-4.9.y

meveric, 2021-03-16:
I remember that right at the beginning the N2 had terrible USB support, whit lots of similar issues as you describe.
It took HardKernel a very long time to fix it, although, I'd say with issues like this, you never know if they are "truly" fixed.
The bug for USB 1.1 devices on the ODROID C1 was never fixed.

Since this is very close to the hardware itself, USB controller, Kernel, etc. it's probably best to ask HardKernel directly.

I've seen similar reports on the coreelec forum, so it doesn't seem to be that uncommon.

MichaIng, 2021-03-16:
There are a few other USB-related issues already, but none that matches this case good enough IMO. @TheOriginalMrWolf please report the issue at: https://github.com/hardkernel/linux/issues
I'll leave this one open to keep track.

TheOriginalMrWolf, 2021-03-17:
Hi @MichaIng,

Looks like something still not right with boot.ini - I'm expecting to see Max Freq == 1608 across the board:

`cpu`
```
 ─────────────────────────────────────────────────────
 DietPi CPU Info
 Use dietpi-config to change CPU / performance options
 ─────────────────────────────────────────────────────
 Architecture |     aarch64
 Temperature  |     32'C : 89'F (Cool runnings)
 Governor     |     interactive

                 Current Freq    Min Freq   Max Freq
 CPU0         |      1896 MHz      100 MHz    1896 MHz
 CPU1         |      1896 MHz      100 MHz    1896 MHz
 CPU2         |      1000 MHz      100 MHz    1800 MHz
 CPU3         |      1000 MHz      100 MHz    1800 MHz
 CPU4         |      1000 MHz      100 MHz    1800 MHz
 CPU5         |      1000 MHz      100 MHz    1800 MHz

[ INFO ] DietPi-CPU_info | CPU current frequency, may be affected by this script, due to the processing required to run it.
```

`boot.ini` is:
```
ODROIDN2-UBOOT-CONFIG

setenv board "odroidn2"

# Serial console device: /dev/tty1 is appended automatically
setenv condev "console=ttyS0,115200n8"

# Auto-detection of monitor settings based on your screen information: "false" or "true"
setenv display_autodetect "true"

# Toggle HDMI output: "false" or "true"
setenv monitor_onoff "false"

# Force SDR or HDR mode: "sdr", "hdr" or "auto"
setenv sdrmode "auto"

# Toggle CEC support: "false" or "true"
setenv cec "false"

# Toggle Wake-On-Lan support: 0=disable, 1=enable
setenv enable_wol "0"

# Device tree overlays
# See /boot/overlays/odroidn2/ for available overlays, e.g.:
# - hktft32: 3.2" TFT from HardKernel
# - hktft35: 3.5" TFT from HardKernel
#setenv overlays "spi0 i2c0 i2c1 uart0"

# HDMI mode resolution configuration
#    Symbol             | Resolution
# ----------------------+-------------
#    "480x272p60hz"     | 480x272 Progressive 60Hz
#    "480x320p60hz"     | 480x320 Progressive 60Hz
#    "480p60hz"         | 720x480 Progressive 60Hz
#    "576p50hz"         | 720x576 Progressive 50Hz
#    "720p60hz"         | 1280x720 Progressive 60Hz
#    "720p50hz"         | 1280x720 Progressive 50Hz
#    "1080p60hz"        | 1920x1080 Progressive 60Hz
#    "1080p50hz"        | 1920x1080 Progressive 50Hz
#    "1080p30hz"        | 1920x1080 Progressive 30Hz
#    "1080p24hz"        | 1920x1080 Progressive 24Hz
#    "1080i60hz"        | 1920x1080 Interlaced 60Hz
#    "1080i50hz"        | 1920x1080 Interlaced 50Hz
[... 85 lines cut ...]
setenv k_addr "0x1100000"
setenv loadaddr "0x1B00000"
setenv initrd_loadaddr "0x3700000"

# Load kernel, dtb and initrd
fatload mmc ${devno}:1 ${k_addr} Image.gz
fatload mmc ${devno}:1 ${dtb_loadaddr} meson64_odroid${variant}.dtb
fatload mmc ${devno}:1 ${initrd_loadaddr} uInitrd
fdt addr ${dtb_loadaddr}

# Load device tree overlays
if test "${overlays}"; then
        setenv dtbo_addr_r "0x11000000"
        fdt resize "16384"
        for overlay in ${overlays}; do
                fatload mmc ${devno}:1 ${dtbo_addr_r} overlays/${board}/${overlay}.dtbo && fdt apply ${dtbo_addr_r}
        done
fi

# Unzip the kernel
unzip ${k_addr} ${loadaddr}

# Boot
booti ${loadaddr} ${initrd_loadaddr} ${dtb_loadaddr}
```

MichaIng, 2021-03-17:
Hmm, syntax looks right now. Could you check `cat /proc/cmdline`?
Or if that file does not exist: `dmesg | grep 'Kernel command line'`

Also to assure the frequencies from the [Wiki](https://wiki.odroid.com/odroid-n2/application_note/software/set_cpu_freq#how_to_set_max_cpu_frequency)/[Wiki2](https://wiki.odroid.com/odroid-n2/hardware/overclocking) are still valid:
```
cat /sys/devices/system/cpu/cpufreq/policy[012]/scaling_available_frequencies
```

TheOriginalMrWolf, 2021-03-17:
@MichaIng 

`cat /proc/cmdline`
```
root=UUID=e139ce78-9841-40fe-8823-96a304a09859 rootwait ro console=ttyS0,115200n8  no_console_suspend fsck.repair=yes net.ifnames=0 elevator=noop hdmimode=2160p60hz cvbsmode=576cvbs maxcpus=6 voutmode=hdmi disablehpd=false enable_wol=0  sdrmode=auto consoleblank=0 logo=osd0,loaded monitor_onoff=false   overscan=100 cvbscable=
```

`cat /sys/devices/system/cpu/cpufreq/policy[012]/scaling_available_frequencies`
```
100000 250000 500000 667000 1000000 1200000 1398000 1512000 1608000 1704000 1896000
100000 250000 500000 667000 1000000 1200000 1398000 1512000 1608000 1704000 1800000
```

MichaIng, 2021-03-17:
Okay, frequencies are valid. Two ideas:
1. It does not like when empty values are applied to boot arguments. Can you try:
    ```
    sed -i 's/ cvbscable=${cvbscable}//' /boot/boot.ini
    reboot
    cpu
    ```
2. `if test "${max_freq_a73}"` does not work like it would in a shell script. Let's try ancient syntax that is also used in the Odroid Wiki:
    ```
    sed -i 's/test "${max_freq_a73}";/test "x${max_freq_a73}" != "x";/' /boot/boot.ini
    sed -i 's/test "${max_freq_a53}";/test "x${max_freq_a53}" != "x";/' /boot/boot.ini
    reboot
    cpu
    ```

TheOriginalMrWolf, 2021-03-18:
Hi @MichaIng - the 'ancient' syntax worked.
`cpu`
```
 ─────────────────────────────────────────────────────
 DietPi CPU Info
 Use dietpi-config to change CPU / performance options
 ─────────────────────────────────────────────────────
 Architecture |     aarch64
 Temperature  |     32'C : 89'F (Cool runnings)
 Governor     |     interactive

                 Current Freq    Min Freq   Max Freq
 CPU0         |      1608 MHz      100 MHz    1608 MHz
 CPU1         |      1608 MHz      100 MHz    1608 MHz
 CPU2         |      1512 MHz      100 MHz    1608 MHz
 CPU3         |      1512 MHz      100 MHz    1608 MHz
 CPU4         |      1512 MHz      100 MHz    1608 MHz
 CPU5         |      1512 MHz      100 MHz    1608 MHz

[ INFO ] DietPi-CPU_info | CPU current frequency, may be affected by this script, due to the processing required to run it.
```

Unfortunately this didn't help with the crashing.

meveric, 2021-03-18:
Have you tried switching the CPU governor?
Most of them are not very stable. Either try performance or ondemand rather than interactive.
Not sure if it helps with that particular issue though.

TheOriginalMrWolf, 2021-03-18:
Just using the default so far.

MichaIng, 2021-03-19:
Default is ondemand on DietPi btw.

And @meveric you were right about using `test "x${var}" != "x"`. I found some uboot docs stating that `test "${var}"` works just the same, like it does in shells, but obviously that is not true for at least this version of uboot.

Joulinar, 2021-12-27:
@MichaIng 
is this still an open issue? We have a user on the forum reporting issues if he connect a HDD via USB. https://dietpi.com/phpbb/viewtopic.php?p=41366#p41366

MichaIng, 2021-12-27:
I guess it is a persistent issue with some USB drives and N2, though I personally didn't run into any with my N2+ so far. Experimenting with mainline kernel is on my ToDo around new years eve when I'm back home (currently family visits).

MichaIng, 2022-01-29:
New image ready for testing based on mainline kernel: https://dietpi.com/downloads/images/testing/
Hopefully it makes USB drive usage more robust.

MichaIng, 2022-02-01:
Referring to the thread for testing the new image and closing this one: https://github.com/MichaIng/DietPi/issues/5039

closed 2022-02-01 by MichaIng