MichaIng/DietPi #4189: Plex server Rpi0

symbios24, 2021-03-11:
Hello
Im trying to install plex server on a Rpi0 with the:
Dietpi-software install 42
But it says its not supported by my system
Is there a way to override this and install plex?
I remember i have it installed in an older dietpi version before.
Thanks

Joulinar, 2021-03-11:
Hi,

can you share Debian version your are running. `echo $G_DISTRO_NAME`. Unfortunately you did not fill out the questioner for opening a new issue 😉

symbios24, 2021-03-11:
Yes sorry its Raspbian buster debian 10

MichaIng, 2021-03-11:
Plex does not provide binaries for ARMv6, so it cannot work on Raspberry Pi 1 and Zero models. That was never different for five years 😉: https://github.com/MichaIng/DietPi/issues/648

closed 2021-05-07 by MichaIng